espressif/esp-aliyun #133: 使用esp-aliyun/examples/solutions/smart_light  demo  烧写后无限重启

PeterSun01, 2019-09-11:
----------------------------- 以下请删除 -----------------------------

提交问题时，请先仔细阅读该段提示信息！！

为了更高效的解决 issue 中提交的问题，请按照以下 issue 模板进行提交，请注意以下信息：
1. 提交时请删除这段提示文字；
2. 提交时删除以下由 `//` 开头的文字，并对此做详细补充，如没有特殊信息，可写 `无`。

感谢配合。

----------------------------- 以上请删除 -----------------------------

## 1. 开发环境

- 模组或开发板：ESP32-WROOM
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：b2b5fb2
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：-----
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：7dd492319

## 2. 问题描述

使用esp-aliyun/examples/solutions/smart_light  demo  烧写后无限重启

### 2.1 复现步骤

1. make chip=esp32 defconfig
2. make menuconfig
3. make erase_flash
4. make flash monitor

### 2.2 复现问题的代码

esp-aliyun/examples/solutions/smart_light

## 3. 调试 Logs

Toolchain path: /root/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /root/esp32/esp-idf/requirements.txt are satisfied.
Erasing entire flash...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: a4:cf:12:73:eb:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.8s
Hard resetting via RTS pin...
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: a4:cf:12:73:eb:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 31...
[... 147 lines cut ...]
I (88) boot:  4 ota_1            OTA app          00 11 00110000 00100000
I (96) boot:  5 fctry            WiFi data        01 02 00210000 00004000
I (103) boot: End of partition table
I (108) boot: No factory image, trying OTA 0
I (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x26610 (157200) map
I (177) esp_image: segment 1: paddr=0x00036638 vaddr=0x3ffb0000 size=0x031a4 ( 12708) load
I (182) esp_image: segment 2: paddr=0x000397e4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /root/esp32/esp-idf/components/freertos/xtensa_vectors.S:1779

I (184) esp_image: segment 3: paddr=0x00039bec vaddr=0x40080400 size=0x06424 ( 25636) load
I (203) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0xa4d94 (675220) map
0x400d0018: _flash_cache_start at ??:?

I (440) esp_image: segment 5: paddr=0x000e4db4 vaddr=0x40086824 size=0x09b24 ( 39716) load
0x40086824: prvCopyDataToQueue at /root/esp32/esp-idf/components/freertos/queue.c:2038

I (467) boot: Loaded app from partition at offset 0x10000
I (467) boot: Disabling RNG early entropy source...
I (467) cpu_start: Pro cpu up.
I (471) cpu_start: Starting app cpu, entry point is 0x400811f0
0x400811f0: call_start_cpu1 at /root/esp32/esp-idf/components/esp32/cpu_start.c:249

## 4. 其他项

无

27-301, 2019-09-11:
当前 v3.2.2 是最新的 release 版本，建议使用该版本进行测试。使用命令 git checkout v3.2.2 切到 v3.2.2。产生 crash 的原因是 v3.2.2 之后的某个 comit 中 wifi 设置 channel api 的行为发生了变化，导致该问题，后面会修改 esp-aliyun。

PeterSun01, 2019-09-12:
你好 应该是V3.2.2 上电log为：I (29) boot: ESP-IDF v3.2.2-132-g7dd492319-dirty 2nd stage bootloader

27-301, 2019-09-12:
这个是 v3.2.2 之后的某一个 commit，`v3.2.2-132-g7dd492319`，这个代表是 v3.2.2 tag 之后的第 132 个 commit，commit id 是 7dd492319。但 v3.2.2 tag 的 commit 是 055943e2。你切到 v3.2.2 上去做测试。

PeterSun01, 2019-09-15:
$git checkout v3.2.2
$git submodule update

目前看暂时没有重启

PeterSun01, 2019-09-15:
使用云智能扫码配网，无法配网成功，log如下
ESP32-WROOM

[inf] zconfig_init(260): zconfig_init
I (885) wifi: ic_enable_sniffer
I (885) awss: Open monitor mode
[crt] awss_save_apinfo(271): [1] ssid:ZZF, mac:7ca177e6ef40, chn:6, rssi:-87, adha:0
[crt] awss_save_apinfo(271): [2] ssid:S1-2201, mac:80891741f476, chn:6, rssi:-65, adha:0
[crt] awss_save_apinfo(271): [3] ssid:, mac:7ca177e6ef45, chn:6, rssi:-89, adha:0
[crt] awss_save_apinfo(271): [4] ssid:Satyr, mac:e4d33245d6a2, chn:6, rssi:-79, adha:0
E (1385) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (1385) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (1385) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (1395) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (1405) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
[crt] aws_switch_channel(166): chan 1
[crt] awss_save_apinfo(271): [5] ssid:sunhao-wifi, mac:50bd5f709100, chn:1, rssi:-42, adha:0
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 1
[crt] awss_save_apinfo(271): [6] ssid:TP-LINK_3551, mac:b0958ecb3551, chn:1, rssi:-94, adha:0
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 4
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 7
[crt] aws_switch_channel(166): chan 8
[crt] aws_switch_channel(166): chan 9
[crt] awss_save_apinfo(271): [7] ssid:deng-PC_Network, mac:54e6fc184ef8, chn:9, rssi:-89, adha:0
[crt] aws_switch_channel(166): chan 10
[crt] awss_save_apinfo(271): [8] ssid:HUAWEI-SHUO, mac:d4a1488da4bc, chn:10, rssi:-93, adha:0
[crt] aws_switch_channel(166): chan 11
[crt] awss_save_apinfo(271): [9] ssid:Luckydog, mac:9ca6151dec0c, chn:11, rssi:-95, adha:0
[crt] awss_save_apinfo(271): [10] ssid:1102, mac:c06118999708, chn:11, rssi:-95, adha:0
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] awss_save_apinfo(271): [11] ssid:CMCC-a7LS, mac:9cfea1333da7, chn:13, rssi:-63, adha:0
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
I (6255) factory_rst: Quick reboot timeout, clear reboot times
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 4
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] awss_save_apinfo(271): [12] ssid:, mac:7ca177e6ef41, chn:6, rssi:-89, adha:0
[crt] aws_switch_channel(166): chan 7
[crt] aws_switch_channel(166): chan 8
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[... 1016 lines cut ...]
[crt] awss_recv_callback_smartconfig(1074): 0+8 [98] 637 F 4c8
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [98] -> [99]
[crt] awss_recv_callback_smartconfig(1090): 8+0 [100] 638 F 3e1
I (72765) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 8+0 [100] 639 F 3e1
I (72775) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 8+1 [100] 63a F 141
[crt] awss_recv_callback_smartconfig(1074): 8+2 [100] 63b F 1cf
[crt] awss_recv_callback_smartconfig(1074): 8+3 [100] 63c F 20d
[crt] awss_recv_callback_smartconfig(1074): 8+6 [84] 640 F 3c6
[crt] awss_recv_callback_smartconfig(1074): 8+7 [84] 641 F 449
[crt] awss_recv_callback_smartconfig(1074): 8+8 [82] 643 F 486
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 645 F 3e2
I (72825) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 647 F 3e2
I (72835) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 16+1 [92] 64a F 125
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [92] -> [95]
[crt] awss_recv_callback_smartconfig(1074): 16+2 [93] 64c F 1a3
[crt] awss_recv_callback_smartconfig(1074): 16+3 [91] 64e F 238
[crt] awss_recv_callback_smartconfig(1074): 16+4 [89] 650 F 290
[crt] awss_recv_callback_smartconfig(1074): 16+5 [87] 652 F 321
[crt] awss_recv_callback_smartconfig(1074): 16+6 [79] 655 F 3a0
[crt] awss_recv_callback_smartconfig(1074): 16+7 [77] 657 F 422
[crt] zconfig_get_ssid_passwd(252):

PeterSun01, 2019-09-16:
以通过量产烧写四元组解决

closed 2019-09-16 by PeterSun01

yanzi418, 2019-09-17:
我这边也出现类似问题
[1;31m[crt] zconfig_get_ssid_passwd(263): 

[0m[1;31m[crt] zconfig_get_ssid_passwd(326): SSID1: [YoSmart]

[0m[0;31mE (61512) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
[1;31m[crt] zconfig_get_ssid_passwd(362): passwd err

[0m[0;31mE (61522) app main: IOTX_AWSS_PASSWD_ERR[0m
[1;31m[crt] awss_recv_callback_smartconfig(960): hint frame: offset:52, T, sn:e49

[0m[1;31m[crt] awss_recv_callback_smartconfig(964): src:dc90881c97a2, bssid:88df9e8f6871

[0m[0;32mI (61542) app main: IOTX_AWSS_LOCK_CHAN[0m
[1;31m[crt] awss_recv_callback_smartconfig(960): hint frame: offset:52, F, sn:260

[0m[1;31m[crt] awss_recv_callback_smartconfig(964): src:dc90881c97a2, bssid:88df9e8f6871

[0m[0;32mI (61582) app main: IOTX_AWSS_LOCK_CHAN[0m
[1;31m[crt] awss_recv_callback_smartconfig(1090): 24+0 [100] e4a T 3e3

yanzi418, 2019-09-18:
已经解决：
参照https://github.com/espressif/esp-aliyun/blob/master/config/mass_mfg/README.md#%E5%8D%95%E4%B8%AA-bin-%E7%94%9F%E6%88%90

yanzi418, 2019-09-18:
和这个问题一样，只需要烧录四元组就可以解决