espressif/esp-aliyun #136: 请问量产在windows下如何操作？

PeterSun01, 2019-09-17:
## 1. 开发环境

- 模组或开发板：ESP32-WROOM
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

针对smartdemo中的NVS量产和烧录，在windows下如何进行？
以下是我在linux下的操作指令，在windows下如何进行？
2、$IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input mfg_config.csv --output mfg_config.bin --size 0x4000
3、$IDF_PATH/components/esptool_py/esptool/esptool.py write_flash 0x210000 mfg_config.bin

### 2.1 复现步骤

无

### 2.2 复现问题的代码

无

## 3. 调试 Logs

无

## 4. 其他项
无

27-301, 2019-09-19:
在 windows 下安装 python 一样可以用这个命令。

PeterSun01, 2019-09-20:
nvs_partition_gen.py 这个python程序 在win和linux下是一个程序吗？可以跨平台使用吗？

27-301, 2019-09-20:
可以。

closed 2019-11-06 by Eavo