espressif/esp-aliyun #138: master下的mqtt例子运行时崩溃

hantown, 2019-09-19:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：master
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：V3.2
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：d7960baa

## 2. 问题描述
编译master  examples/mqtt例子，把WIFI_SSID，WIFI_PASSWORD和ProductKey，DeviceName，DeviceSecret都已经设置好后，编译通过，但运行时崩溃，log如下：
[0;32mI (52) boot: compile time 15:35:58[0m
[0;32mI (53) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (57) boot: SPI Speed      : 40MHz[0m
[0;32mI (62) boot: SPI Mode       : QIO[0m
[0;32mI (66) boot: SPI Flash Size : 2MB[0m
[0;32mI (69) boot: Partition Table:[0m
[0;32mI (73) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (80) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (88) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (95) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (103) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (110) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (118) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (125) boot: End of partition table[0m
[0;32mI (130) boot: No factory image, trying OTA 0[0m
[0;32mI (135) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7feb4 (523956) map[0m

[16:40:23.925]收←◆[0;32mI (320) esp_image: segment 1: paddr=0x0008fecc vaddr=0x4028fec4 size=0x179a0 ( 96672) map[0m
[0;32mI (353) esp_image: segment 2: paddr=0x000a7874 vaddr=0x3ffe8000 size=0x00860 (  2144) load[0m
[0;32mI (354) esp_image: segment 3: paddr=0x000a80dc vaddr=0x40100000 size=0x00a40 (  2624) load[0m
[0;32mI (361) esp_image: segment 4: paddr=0x000a8b24 vaddr=0x40100a40 size=0x04ec4 ( 20164) load[0m
[0;32mI (375) boot: Loaded app from partition at offset 0x10000[0m

[16:40:24.018]收←◆[0;32mI (390) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (401) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m6顼熹鋖o跙狫偂鍈スパ?failed to load RF calibration data (0x1102), falling back to full calibration[0m

[16:40:24.226]收←◆phy_version: 1149.0, c6543a8, Aug 21 2019, 15:51:08
[0;32mI (592) phy_init: phy ver: 1149_0[0m
[0;33mW (596) phy_init: saving new calibration data because of checksum failure, mode(2)[0m
[0;32mI (610) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1[0m
[0;32mI (616) tcpip_adapter: tcpip_adapter_init 
[0m

[16:40:24.313]收←◆[prt] log level set as: [ 4 ]
[0;32mI (678) conn_mgr: Found ssid RFTEST[0m

 LsAp: vTaskDelete: user_init_entry 
 

[16:40:27.785]收←◆ [0;32mI (6268) wifi: state: 0 -> 2 (b0)[0m
[0;32mI (6272) wifi: state: 2 -> 3 (0)[0m
[0;32mI (6280) wifi: state: 3 -> 5 (10)[0m

[16:40:28.825]收←◆[0;32mI (7302) event: sta ip: 1
[16:40:28.853]收←◆92.168.137.10, mask: 255.255.255.0, gw: 192.168.137.1[0m

[16:40:28.932]收←◆[0;32mI (7412) conn_mgr: SNTP get time success
[0m
[0;31mE (7419) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
[0;31mE (7426) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
[0;31mE (7432) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
mqtt_main|124 :: mqtt example
[0;31mE (7440) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
[0;31mE (7446) wrapper_product: HAL_GetProductParam nvs_open failed with 1102[0m
[1;31m[err] IOT_MQTT_Construct(395): Invalid product key, abort!
[0mmqtt_main|227 :: MQTT construct failed
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00000030  A0      : 0x00000000  A1      : 0x3fff7350  
A2      : 0xffffffff  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000000  
A6      : 0x4023a720  A7      : 0x000001f4  A8      : 0x00000000  A9      : 0x00000310  
[... 66 lines cut ...]

Backtrace: 0x4010148e:0x3ffeb030 0x40223cd0:0x3ffeb030 0x402241c3:0x3ffeb040 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010148e  PS      : 0x00000035  A0      : 0x40223cd0  A1      : 0x3ffeb030  
A2      : 0x3ffecacc  A3      : 0x00000000  A4      : 0x3ffeac0c  A5      : 0x3ffeac0c  
A6      : 0x0000000a  A7      : 0x00000017  A8      : 0x000000f0  A9      : 0x00000000  
A10     : 0x000000c1  A11     : 0x402241c3  A12     : 0x00000007  A13     : 0x00000000  
A14     : 0x00005000  A15     : 0x3ffeb040  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  

### 2.1 复现步骤

上电多次都是一样

### 2.2 复现问题的代码

无

## 3. 调试 Logs

看上面

## 4. 其他项

无

hantown, 2019-09-19:
[image]
我留意到log里面提示说product key无效，但我已经配置，这些key是今天刚从阿里云申请的。
[image]

ljy770, 2019-09-19:
您好, 不是在 make menuconfig 里面设置三元组,现在已经改成烧录模式,具体请参考量产说明文档: esp-aliyun/config/mass_mfg/README.md

运行示例前,请您仔细阅读示例下的readme, examples/mqtt/mqtt_example/README.md, 比较清楚地说明了每个步骤, 可以避免一些操作上的错误.

hantown, 2019-09-20:
谢谢！既然已经改了，那就麻烦你们在make menuconfig这里也改掉，否则误导鑫粉们，体验就不好了！

27-301, 2019-09-20:
> 谢谢！既然已经改了，那就麻烦你们在make menuconfig这里也改掉，否则误导鑫粉们，体验就不好了！

感谢支持，会尽快改掉。

27-301, 2019-09-20:
已修改如 master 分支，重新 pull 一下。

closed 2019-09-20 by 27-301