espressif/esp-aliyun #139: master版本在menuconfig中设置三元组后配网提示密码错误

hqchen118, 2019-09-20:
----------------------------- 以下请删除 -----------------------------

提交问题时，请先仔细阅读该段提示信息！！

为了更高效的解决 issue 中提交的问题，请按照以下 issue 模板进行提交，请注意以下信息：
1. 提交时请删除这段提示文字；
2. 提交时删除以下由 `//` 开头的文字，并对此做详细补充，如没有特殊信息，可写 `无`。

感谢配合。

----------------------------- 以上请删除 -----------------------------

## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：Master
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：V3.2

## 2. 问题描述

在examples/solutions/smart_light，make menuconfig出来的菜单里面配置三元组

[image]

编译下载到ESP-32D.
用测试版APP扫二维码配网，log提示密码错误。

### 2.2 复现问题的代码

## 3. 调试 Logs

#
[image]

[image]
# 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

27-301, 2019-09-20:
同 https://github.com/espressif/esp-aliyun/issues/138#issuecomment-533044585，menuconfig 中的是无效的配置，会删除，请使用量产方式烧入三元组信息。

hqchen118, 2019-09-23:
用ali提供的公版app配网（一键配网），模组可以配网了，但是app提示失败，不能进入设备页面。
设备log如下：

[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 8
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[inf] zconfig_callback_channel_locked(118): channel lock @ 6
I (159629) app main: IOTX_AWSS_LOCK_CHAN
[inf] zconfig_callback_over(136): zconfig done. ssid:p10, mac:000000000000
I (159639) app main: IOTX_AWSS_GOT_SSID_PASSWD
I (159639) wifi: ic_disable_sniffer
I (159649) awss: Close monitor mode
[crt] aws_switch_dst_chan(214): adjust chan 1
I (159749) app main: IOTX_AWSS_CONNECT_ROUTER
I (159749) awss: ssid: p10, password: zhang123, channel: 1
I (159749) awss: Connecting AP
I (160259) awss: Connecting AP
I (160479) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (160759) awss: Connecting AP
I (161259) awss: Connecting AP
I (161459) wifi: state: init -> auth (b0)
I (161489) wifi: state: auth -> assoc (0)
I (161499) wifi: state: assoc -> run (10)
I (161529) wifi: connected with p10, channel 6
I (161529) wifi: pm start, type: 1

I (161759) awss: Connecting AP
I (162259) awss: Connecting AP
I (162759) awss: Connecting AP
I (163259) awss: Connecting AP
I (163759) awss: Connecting AP
I (164259) awss: Connecting AP
I (164759) awss: Connecting AP
I (164829) event: sta ip: 192.168.43.137, mask: 255.255.255.0, gw: 192.168.43.9
[inf] dm_client_open(37): CM Fd: 0
I (164839) app main: IOTX_CONN_CLOUD
[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1k4OLALV51.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[wrn] IOT_MQTT_Construct(481): Using default client_id: a1k4OLALV51.sdk_3_ubuntu|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[wrn] IOT_MQTT_Construct(488): Using default username: sdk_3_ubuntu&a1k4OLALV51
[wrn] IOT_MQTT_Construct(496): Using default password: ******
[inf] iotx_mc_init(230): MQTT init success!
I (165259) awss: AP connected
I (165259) app main: IOTX_AWSS_GOT_IP
user_awss_status_event_handler.68: Awss Status 4109
I (165259) udp: success to establish udp, fd=55
I (165299) app main: IOTX_AWSS_SUC_NOTIFY
user_awss_status_event_handler.68: Awss Status 4110
[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
[inf] awss_notify_dev_info(209): send notify success
[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1k4OLALV51.sdk_3_ubuntu|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=sdk_3_ubuntu&a1k4OLALV51
[inf] _mqtt_connect(821): mqtt connect success!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/request/a1k4OLALV51/sdk_3_ubuntu!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/upgrade/a1k4OLALV51/sdk_3_ubuntu!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/config/get_reply!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/config/push!
user_mqtt_connect_succ_event_handler.325: ITE_MQTT_CONNECT_SUCC
[inf] _dm_client_event_cloud_connected_handle(164): IOTX_CM_EVENT_CLOUD_CONNECTED
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/awss/enrollee/match_reply!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/awss/enrollee/checkin!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/awss/enrollee/found_reply!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/cipher/get_reply!
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/awss/device/switchap!
W (168069) wrapper_kv: nvs get blob awss.rst failed with 1102
[inf] awss_report_token_to_cloud(351): report token res:12
[... 32 lines cut ...]
[inf] dm_client_subscribe(86): Subscribe Result: 21
[inf] dm_client_subscribe_all(124): index: 9
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/rrpc/request/+!
[inf] dm_client_subscribe(86): Subscribe Result: 22
[inf] dm_client_subscribe_all(124): index: 10
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ext/ntp/a1k4OLALV51/sdk_3_ubuntu/response!
[inf] dm_client_subscribe(86): Subscribe Result: 23
[inf] dm_client_subscribe_all(124): index: 11
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ext/error/a1k4OLALV51/sdk_3_ubuntu!
[inf] dm_client_subscribe(86): Subscribe Result: 24
[inf] dm_client_subscribe_all(124): index: 12
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/model/down_raw!
[inf] dm_client_subscribe(86): Subscribe Result: 25
[inf] dm_client_subscribe_all(124): index: 13
[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1k4OLALV51/sdk_3_ubuntu/thing/model/up_raw_reply!
[inf] dm_client_subscribe(86): Subscribe Result: 26
[inf] iotx_dm_subscribe(208): Devid 0 Sub Completed
[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 10
[inf] _iotx_linkkit_event_callback(245): Receive Message: {"devid":0}
user_initialized.110: Device Initialized
[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 0
user_connected_event_handler.76: Cloud Connected
[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[inf] iotx_mc_cycle(1570): receive ping response!
[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...

ljy770, 2019-09-25:
ssid: p10, password: zhang123 是您希望设备连接的 AP 吗？
手机是不是也连的这个 AP？
环境中是否存在其他已联网设备？
和您再确认一下，配网 log 前面没有 awss_recv_callback_smartconfig 的打印信息？
如果没有，比较怀疑设备不是通过 smartconfig 配网方式连接的，可能是通过其他已联网设备进行配网。

hqchen118, 2019-09-25:
重新配网了一次，现象还是不能拉取到设备控制界面。
[crt] aws_switch_channel(166): chan 8
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 5
[crt] awss_recv_callback_smartconfig(960): hint frame: offset:52, T, sn:1e7
[crt] awss_recv_callback_smartconfig(964): src:38e60a75d256, bssid:5cc30763438c
[inf] zconfig_callback_channel_locked(118): channel lock @ 5
I (45996) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 8+0 [100] 1e8 T 3e1
I (46006) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 8+1 [100] 1e9 T 155
[crt] awss_recv_callback_smartconfig(1074): 8+2 [100] 1ea T 1cf
[crt] awss_recv_callback_smartconfig(1074): 8+3 [100] 1eb T 207
[crt] awss_recv_callback_smartconfig(1074): 8+4 [100] 1ec T 283
[crt] awss_recv_callback_smartconfig(1074): 8+5 [100] 1ed T 30c
[crt] awss_recv_callback_smartconfig(1074): 8+6 [100] 1ee T 3b3
[crt] awss_recv_callback_smartconfig(1074): 8+7 [100] 1ef T 429
[dbg] awss_recv_callback_smartconfig(1002): drop: 1ef == 1ef
[crt] awss_recv_callback_smartconfig(1074): 8+8 [100] 1f0 T 483
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 1fe T 4e0
I (46076) app main: IOTX_AWSS_LOCK_CHAN
[dbg] awss_recv_callback_smartconfig(1002): drop: 1fe == 1fe
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 1ff T 4e0
I (46096) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 200 T 4e0
I (46106) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 202 T 4e0
I (46116) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 203 T 4e0
I (46126) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 0+1 [100] 204 T 118
[crt] awss_recv_callback_smartconfig(1074): 0+2 [100] 205 T 18f
[crt] awss_recv_callback_smartconfig(1074): 0+3 [100] 206 T 206
[dbg] aws_main_thread_func(317): [channel scanning] 45570 ms
[crt] zconfig_recv_completed(151): ssid auto-complete: kw.luo
[dbg] aws_main_thread_func(326): [crt] awss_recv_callback_smartconfig(1074): 
0+4 [100] 207 T 28c
[crt] awss_recv_callback_smartconfig(1074): 0+5 [70] 212 T 30c
[crt] awss_recv_callback_smartconfig(1162): 	enter try_to_sync_pos: data mismatch
[crt] awss_recv_callback_smartconfig(1058): 	X+5 [1] 212 T 30c
[crt] awss_recv_callback_smartconfig(1058): 	X+6 [1] 213 T 3b3
[crt] awss_recv_callback_smartconfig(960): hint frame: offset:52, F, sn:479
[crt] awss_recv_callback_smartconfig(964): src:38e60a75d256, bssid:5cc30763438c
I (46216) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1081): 	exit try_to_sync_pos: group frame
[crt] try_to_sync_pos(664): [5]=34b, [5]=30c
[crt] try_to_sync_pos(675): match=2, match_group=8, match_end=8
[crt] try_to_sync_pos(738): 	X = 8, score=22, match=2, reason=1
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 217 T 3e2
I (46256) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 8+1 [100] 47a F 155
[crt] zconfig_recv_completed(151): ssid auto-complete: kw.luo
[crt] awss_recv_callback_smartconfig(1074): 8+2 [100] 47b F 1cf
[crt] awss_recv_callback_smartconfig(1074): 8+3 [100] 47c F 207
[crt] awss_recv_callback_smartconfig(1074): 8+4 [100] 47d F 283
[crt] awss_recv_callback_smartconfig(1074): 16+1 [100] 218 T 10f
[crt] awss_recv_callback_smartconfig(1074): 8+5 [100] 47e F 30c
[crt] awss_recv_callback_smartconfig(1074): 8+6 [100] 47f F 3b3
[crt] awss_recv_callback_smartconfig(1074): 8+8 [92] 481 F 483
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [92] -> [96]
[crt] awss_recv_callback_smartconfig(1074): 16+2 [100] 219 T 19e
[crt] awss_recv_callback_smartconfig(1074): 16+3 [100] 21a T 233
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 482 F 3e2
I (46346) app main: IOTX_AWSS_LOCK_CHAN
[... 623 lines cut ...]
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /dev/core/service/dev
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] CoAPRequestMessage_handle(471): Request path is /sys/device/info/get
[dbg] CoAPResourceByPath_get(210): Found the resource: /sys/device/info/get
[dbg] online_get_device_info(335): no req
[dbg] MQTTKeepalive(1645): len = MQTTSerialize_pingreq() = 2
[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[inf] iotx_mc_cycle(1570): receive ping response!

ljy770, 2019-09-25:
在esp-aliyun 目录下 git rev-parse --short HEAD 看一下 commit id ? 
如果不是最新，更新到最新试试。
因为中间有修改过 timer ，可能您用了有问题的版本，9-17号修复的。

hqchen118, 2019-09-26:
esp-aliyun 更新到c5a1728。

烧写到esp-32后，设备出现不断重启。

app端偶尔能进去设备控制界面了

[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 5
[crt] awss_recv_callback_smartconfig(960): hint frame: offset:52, T, sn:238
[crt] awss_recv_callback_smartconfig(964): src:38e60a75d256, bssid:5cc30763438d
[inf] zconfig_callback_channel_locked(118): channel lock @ 5
I (28146) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 239 T 3e2
I (28156) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 16+1 [100] 23a T 116
[crt] awss_recv_callback_smartconfig(1074): 16+2 [100] 23b T 18a
[crt] awss_recv_callback_smartconfig(1074): 16+3 [100] 23c T 209
[crt] awss_recv_callback_smartconfig(1074): 16+4 [100] 23d T 2b4
[crt] awss_recv_callback_smartconfig(1074): 16+5 [100] 23e T 317
[crt] awss_recv_callback_smartconfig(1074): 16+6 [100] 23f T 38e
[crt] awss_recv_callback_smartconfig(1074): 16+7 [100] 240 T 40d
[crt] awss_recv_callback_smartconfig(1074): 16+8 [100] 241 T 4a6
[crt] awss_recv_callback_smartconfig(1090): 24+0 [100] 242 T 3e3
I (28216) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 16+0 [100] 239 T 3e2
I (28236) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 24+0 [100] 243 T 3e3
I (28506) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 244 T 4e0
I (28506) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 245 T 4e0
I (28526) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 246 T 4e0
I (28536) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 247 T 4e0
I (28546) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 248 T 4e0
I (28556) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 249 T 4e0
I (28566) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 0+1 [100] 24a T 118
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 244 T 4e0
I (28586) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 245 T 4e0
I (28596) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 246 T 4e0
I (28606) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1090): 0+0 [100] 247 T 4e0
I (28616) app main: IOTX_AWSS_LOCK_CHAN
[crt] awss_recv_callback_smartconfig(1074): 0+2 [92] 24b T 18f
[crt] awss_recv_callback_smartconfig(1074): 0+3 [92] 24c T 207
[crt] zconfig_recv_completed(151): ssid auto-complete: samyou2
[crt] awss_recv_callback_smartconfig(1074): 0+4 [92] 24d T 28b
[crt] awss_recv_callback_smartconfig(1074): 0+5 [92] 24e T 353
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [92] -> [96]
[crt] awss_recv_callback_smartconfig(1066): 	 drop: group_sn:24e, sn:24c-24b=ffe, pos:3-5, len:207
[crt] awss_recv_callback_smartconfig(1066): 	 drop: group_sn:24e, sn:24d-24c=fff, pos:4-5, len:28b
[crt] awss_recv_callback_smartconfig(1066): 	 drop: group_sn:24e, sn:24e-24d=0, pos:5-5, len:353
[crt] awss_recv_callback_smartconfig(1074): 0+6 [96] 24f T 3c1
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [96] -> [98]
[crt] awss_recv_callback_smartconfig(1074): 0+7 [98] 250 T 44d
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [98] -> [99]
[crt] awss_recv_callback_smartconfig(1074): 0+8 [99] 251 T 4d9
[crt] awss_recv_callback_smartconfig(1168): 			 uplimit [99] -> [100]
[... 104 lines cut ...]
[inf] awss_notify_dev_info(209): send notify success
[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
***ERROR*** A stack overflow in task Tmr Svc has been detected.
abort() was called at PC 0x4008f7c0 on core 0
0x4008f7c0: vApplicationStackOverflowHook at /home/hankchen/esp/esp-idf/components/esp32/panic.c:707

Backtrace: 0x4008f57c:0x3ffbce30 0x4008f7a9:0x3ffbce50 0x4008f7c0:0x3ffbce70 0x4008c834:0x3ffbce90 0x4008e5b8:0x3ffbceb0 0x4008e56e:0x00000000
0x4008f57c: invoke_abort at /home/hankchen/esp/esp-idf/components/esp32/panic.c:707

0x4008f7a9: abort at /home/hankchen/esp/esp-idf/components/esp32/panic.c:707

0x4008f7c0: vApplicationStackOverflowHook at /home/hankchen/esp/esp-idf/components/esp32/panic.c:707

0x4008c834: vTaskSwitchContext at /home/hankchen/esp/esp-idf/components/freertos/tasks.c:4721

0x4008e5b8: _frxt_dispatch at /home/hankchen/esp/esp-idf/components/freertos/portasm.S:406

0x4008e56e: _frxt_int_exit at /home/hankchen/esp/esp-idf/components/freertos/portasm.S:206

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

ljy770, 2019-09-26:
您好，刚提交一些修改，您更新一下 。
重新执行：
make chip=esp32 defconfig
make -j8

hqchen118, 2019-09-26:
可以了，非常感谢！！！

closed 2019-09-26 by hqchen118