espressif/esp-aliyun #140: 配置3元组，从阿里云获取的数据名称与CSV上面的不一致，咋整？

hantown, 2019-09-20:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：c5a1728
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：d7960baa
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：d7960baa

## 2. 问题描述

配置3元组，从阿里云获取的数据名称与CSV上面的对应不上，要怎么弄？
[image]
[image]

### 2.1 复现步骤

无
### 2.2 复现问题的代码

无

## 3. 调试 Logs
无

## 4. 其他项

无

ljy770, 2019-09-20:
/home/lijingyi/esp/esp-aliyun-master/esp-aliyun/examples/solutions/smart_light/_static/p3.png

ljy770, 2019-09-20:
[image]

hantown, 2019-09-20:
好的，谢谢

closed 2019-09-20 by 27-301