espressif/esp-aliyun #141: 使用 nvs_partition_gen.py生成的包含秘钥的 NVS错误！

hantown, 2019-09-20:
## 1. 开发环境

- 模组或开发板：
esp-aliyun 版本（git rev-parse --short HEAD 来获取 commit id）：c5a1728
ESP8266_RTOS_SDK 版本（git rev-parse --short HEAD 来获取 commit id）：d7960baa
ESP-IDF 版本（git rev-parse --short HEAD 来获取 commit id）：d7960ba

## 2. 问题描述
虚拟机运行nvs_partition_gen.py --input my_single_mfg_config.csv --output my_single_mfg.bin --size 0x4000时出现如下错误，我可以确认my_single_mfg_config.csv这个文件已经使用文本编辑器另存为UTF8编码格式。

root@user-virtual-machine:/mnt/hgfs/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/components/nvs_flash/nvs_partition_generator# ./nvs_partition_gen.py --input my_single_mfg_config.csv --output my_single_mfg.bin --size 0x4000
Traceback (most recent call last):
  File "./nvs_partition_gen.py", line 805, in <module>
    main()
  File "./nvs_partition_gen.py", line 800, in main
    nvs_part_gen(input_filename, output_filename, part_size, is_key_gen, is_encrypt_data, key_file, version_no)
  File "./nvs_partition_gen.py", line 739, in nvs_part_gen
    for row in reader:
  File "/usr/lib/python2.7/csv.py", line 107, in next
    self.fieldnames
  File "/usr/lib/python2.7/csv.py", line 90, in fieldnames
    self._fieldnames = self.reader.next()
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufeff' in position 0: ordinal not in range(128)
root@user-virtual-machine:/mnt/hgfs/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/components/nvs_flash/nvs_partition_generator# 

### 2.1 复现步骤

无
### 2.2 复现问题的代码

无

## 3. 调试 Logs

无
## 4. 其他项

无

27-301, 2019-09-20:
你先用这个目录下自带的 single_mfg_config.csv 试试是否可以，先排除是文件格式问题还是其他。如果可以，那是你文件格式的问题，可以基于这个文件修改，建议直接在 linux 下改，不要 mount 到 windows 下改。如果不行，那把你的 csv 发上来看下。

hantown, 2019-09-20:
在Linux里面去做可以了

closed 2019-09-23 by 27-301