espressif/esp-aliyun #143: solutions\smart_light如何触发进入配网？

angle600, 2019-09-27:
solutions\smart_light如何触发进入配网？芯片上电后不断打印如下信息，对天猫精灵音箱说“天猫精灵找队友”，没有配到网络

[1;31m[crt] aws_switch_channel(166): chan 13

[0m[1;31m[crt] aws_switch_channel(166): chan 1

[0m[1;31m[crt] aws_switch_channel(166): chan 2

[0m[1;31m[crt] awss_save_apinfo(271): [33] ssid:ChinaNet-eq8y, mac:e09796e0cb3c, chn:2, rssi:-95, adha:0

[0m[1;31m[crt] aws_switch_channel(166): chan 3

[0m[1;31m[crt] aws_switch_channel(166): chan 6

[0m[1;31m[crt] aws_switch_channel(166): chan 7

[0m[1;31m[crt] aws_switch_channel(166): chan 8

[0m[1;31m[crt] aws_switch_channel(166): chan 9

[0m[1;31m[crt] aws_switch_channel(166): chan 10

[0m[1;31m[crt] aws_switch_channel(166): chan 11

[0m[1;31m[crt] aws_switch_channel(166): chan 12

[0m[1;31m[crt] aws_switch_channel(166): chan 13

[0m[1;31m[crt] aws_switch_channel(166): chan 1

[0m[1;31m[crt] aws_switch_channel(166): chan 6

[0m[1;31m[crt] aws_switch_channel(166): chan 11

[0m[1;31m[crt] aws_switch_channel(166): chan 1

[0m[1;31m[crt] aws_switch_channel(166): chan 2

[0m[1;31m[crt] aws_switch_channel(166): chan 3

[0m[1;31m[crt] aws_switch_channel(166): chan 4

[0m[1;31m[crt] aws_switch_channel(166): chan 5

[0m[1;31m[crt] aws_switch_channel(166): chan 6

[0m[1;31m[crt] aws_switch_channel(166): chan 7

[0m[1;31m[crt] aws_switch_channel(166): chan 8

[0m[1;31m[crt] aws_switch_channel(166): chan 9

[0m[1;31m[crt] aws_switch_channel(166): chan 10

[0m[1;31m[crt] aws_switch_channel(166): chan 11

ljy770, 2019-09-27:
天猫精灵配网，可以对天猫精灵说“发现设备”
一键配网，可以通过云智能 APP，国内用户版，下载链接：https://living.aliyun.com/doc#muti-app.html

angle600, 2019-09-27:
这个设备上电后不断的打印.[0m.[1;31m[crt] aws_switch_channel(166):，用天猫精灵配置不了网络，用手机软件IOT_Espressif_EspTouch也配置不了，是什么原因呢？

.[0;33mW (771) wrapper_kv: nvs get blob ssid failed with 1102.[0m
.[1;31m[crt] awss_config_press(180): enable awss

.[0m.[0;32mI (781) app main: IOTX_AWSS_ENABLE.[0m
.[0;32mI (781) app main: IOTX_AWSS_START.[0m
.[1;33m[inf] zconfig_init(260): zconfig_init

.[0m.[0;32mI (801) awss: Open monitor mode.[0m
.[1;31m[crt] aws_switch_channel(166): chan 1

.[0m.[1;31m[crt] awss_save_apinfo(271): [1] ssid:newifi_5BB8, mac:207693245bb8, chn:1, rssi:-61, adha:0

.[0m.[1;31m[crt] aws_switch_channel(166): chan 6

.[0m.[1;31m[crt] awss_save_apinfo(271): [2] ssid:zheng, mac:c061187f704f, chn:11, rssi:-62, adha:0

.[0m.[1;31m[crt] awss_save_apinfo(271): [3] ssid:TP-LINK_F466, mac:3496727ff466, chn:6, rssi:-76, adha:0

.[0m.[1;31m[crt] aws_switch_channel(166): chan 11

.[0m.[1;31m[crt] awss_save_apinfo(271): [4] ssid:TP-LINK_C9FA, mac:9ca6155ac9fa, chn:11, rssi:-84, adha:0

.[0m.[1;31m[crt] aws_switch_channel(166): chan 1

.[0m.[1;31m[crt] aws_switch_channel(166): chan 2

.[0m.[1;31m[crt] aws_switch_channel(166): chan 3

.[0m.[1;31m[crt] awss_save_apinfo(271): [5] ssid:ChinaNet-yMW4, mac:fc372b8e4d49, chn:3, rssi:-90, adha:0

.[0m.[1;31m[crt] aws_switch_channel(166): chan 4

.[0m.[1;31m[crt] aws_switch_channel(166): chan 5

.[0m.[1;31m[crt] aws_switch_channel(166): chan 6

.[0m.[1;31m[crt] aws_switch_channel(166): chan 7

.[0m.[1;31m[crt] aws_switch_channel(166): chan 8

.[0m.[1;31m[crt] aws_switch_channel(166): chan 9

.[0m.[1;31m[crt] aws_switch_channel(166): chan 10

.[0m.[1;31m[crt] aws_switch_channel(166): chan 11

.[0m.[1;31m[crt] aws_switch_channel(166): chan 12

ljy770, 2019-09-27:
您好， 设备上电后不断的打印.[0m.[1;31m[crt] aws_switch_channel(166): 这是设备进入了配网状态，是没有问题的。
这里是阿里云的配网方式，您用 IOT_Espressif_EspTouch 是不匹配的，需要用阿里云提供的云智能APP，从上一回复里面提供的链接下载。您可以先试一下用这个 APP 配网。
三元组信息是否已经烧录？ 不然配网会不成功。

angle600, 2019-09-28:
> 您好， 设备上电后不断的打印.[0m.[1;31m[crt] aws_switch_channel(166): 这是设备进入了配网状态，是没有问题的。
> 这里是阿里云的配网方式，您用 IOT_Espressif_EspTouch 是不匹配的，需要用阿里云提供的云智能APP，从上一回复里面提供的链接下载。您可以先试一下用这个 APP 配网。
> 三元组信息是否已经烧录？ 不然配网会不成功。

---------------------------------------------------------------------------------------------------------------
我烧了https://www.aligenie.com提供的三元组啊，难道不能直连精灵云？

xuhongv, 2019-10-07:
阿里云生活物联网架构师专题 ①】esp32 sdk 直连接入阿里云物联网平台，实现天猫精灵语音控制； - 徐宏的博客。 - CSDN博客 https://blog.csdn.net/xh870189248/article/details/101363357

@angle600

closed 2019-11-06 by Eavo