espressif/esp-aliyun #144: ESP32零配不成功，是否经过验证？

PeterSun01, 2019-09-28:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：master
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：3.2.2

## 2. 问题描述

ESP32零配不成功，是否经过验证？

### 2.1 复现步骤

smart_light demo

### 2.2 复现问题的代码

smart_light demo

## 3. 调试 Logs

// 由于 log 可能比较长，请通过附件的形式提供，否则该 issue 页面会非常长，不便于其他开发者查阅。

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

xuhongv, 2019-10-07:
@PeterSun01 
阿里云生活物联网架构师专题 ①】esp32 sdk 直连接入阿里云物联网平台，实现天猫精灵语音控制； - 徐宏的博客。 - CSDN博客 https://blog.csdn.net/xh870189248/article/details/101363357

27-301, 2019-10-08:
@xuhongv 👍 可以再出一篇基于 esp-aliyun 接入 “天猫精灵 IoT 开放平台” 的。

xuhongv, 2019-10-08:
@27-301 这个“天猫精灵 IoT 开放平台” 是要和你们签协议再可以拿到工程，我写出来没多大意义。虽然我弄过了

27-301, 2019-10-08:
@xuhongv 不是之前那一套了，现在 “天猫精灵 IoT 开放平台” 底层和 aliyun 打通了，和 "飞燕" 类似，https://www.aligenie.com/doc/357554/pbbggx。

xuhongv, 2019-10-08:
aliosthings吗？这个我去年就做过了，写过博文，貌似现在没啥花样了

27-301, 2019-10-08:
> aliosthings吗？这个我去年就做过了，写过博文，貌似现在没啥花样了

不是，我是说用 esp-aliyun 直接接入 “天猫精灵 IoT 开放平台” ，就像用 esp-aliyun 直接接入 "飞燕" 一样。

xuhongv, 2019-10-08:
噢噢。这样啊，应该不难。。你们可以有啥鼓励么 比如吧博文推到公众号，可以么 哈哈

27-301, 2019-10-08:
没难度，烧个天猫的三元组就 OK 了。

没问题的，等 Claire 小姐姐回来上班，找她清算，记着记着。 😄

xuhongv, 2019-10-08:
那我等着，能否留个联系方式？我有QQ群，里面一大堆乐鑫忠实粉丝，联系方式发我邮箱，我加您！
我的email：870189248@qq.com

xuhongv, 2019-10-10:
- @27-301 弄好了。。
乐鑫esp32直连天猫精灵IOT开放平台，实现天猫精灵找队友零配网功能和语音控制https://mp.weixin.qq.com/s/pxxmoVtLd4G5LXnpkmOyZA

PeterSun01, 2019-10-11:
@xuhongv 可以加我qq吗 一起学习讨论 117013107

PeterSun01, 2019-10-11:
$IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input single_mfg_config.csv --output my_single_mfg.bin --size 0x4000

$IDF_PATH/components/esptool_py/esptool/esptool.py write_flash 0x210000 my_single_mfg.bin
@xuhongv 请问这两个量产指令 能在win下运行吗 尤其是第一个生成bin 第二个可以用烧写软件代替

xuhongv, 2019-10-11:
第一个指令 是把四元组生成bin文件，那里可以把阿里云提供四元组 .xml 文件编译成多个bin文件；
第二个你用烧录软件烧录就好了，或者自己写个 py脚本，每成功烧录一个bin文件就删除它，这样就不用人工去一个一个选择了！

xuhongv, 2019-10-11:
@PeterSun01

closed 2019-11-06 by Eavo