espressif/esp-aliyun #145: esp-aliyun怎么编译?

hscxg, 2019-09-29:
请问一下, 下载的esp-aliyun该放在esp-idf下面的什么目录位置啊?
我试了3个位置:
1. esp-aliyun放在 esp-idf\ 目录外面
2. esp-aliyun放在 esp-idf\ 目录下面
3. esp-aliyun放在 esp-idf\examples\ 目录下面
这三个位置都无法编译 ..\esp-aliyun\examples\solutions\smart_light这个项目.

另外需说明的是: 编译器的环境我都已经搭建好, ...\esp-idf\examples\get-started\hello_world这个项目可以正确编译, 并且可以下载到ESP32开发板中去, 程序也可以正确运行.

27-301, 2019-09-29:
和位置没关系，只需要指定 esp-idf 路径就可以了，export IDF_PATH= ....，esp32 的编译，先执行 make chip=esp32 defconfig，再 menuconfig 配置

xuhongv, 2019-10-07:
-【阿里云生活物联网架构师专题 ①】esp32 sdk 直连接入阿里云物联网平台，实现天猫精灵语音控制； - 徐宏的博客。 - CSDN博客 https://blog.csdn.net/xh870189248/article/details/101363357

@hscxg

hscxg, 2019-10-08:
编译时, 出现下面的错误信息, 不知怎么解决?

KBS_CXG@KBS_CXG-PC MSYS /e/msys32/esp-aliyun/examples/solutions/smart_light
$ make -j8
GENCONFIG
/e/msys32/esp-aliyun/examples/solutions/smart_light/sdkconfig:90:warning: symbol value '' invalid for PHY_DATA_OFFSET
#
# configuration written to /e/msys32/esp-aliyun/examples/solutions/smart_light/sdkconfig
#
Building partitions from /e/msys32/esp-aliyun/examples/solutions/smart_light/partitions_esp32.csv...
CC src/bootloader_random.o
CC rtc_module.o
CXX cxx_guards.o
CC esp_ota_ops.o
CC bt.o
CC libcoap/src/address.o
CC int_wdt.o
CC libcoap/src/async.o
CC src/flash_encrypt.o
AR libbt.a
AR libapp_update.a
CC i2c.o
CC brownout.o
CC emac_main.o
CC src/esp_image_format.o
CC libcoap/src/block.o
CC system_api.o
CC factory_restore.o
CC src/flash_partitions.o
CC libcoap/src/coap_time.o
CC sdmmc_host.o
E:/msys32/esp-aliyun/examples/solutions/smart_light/components/factory_restore/factory_restore.c:31:24: fatal error: dm_wrapper.h: No such file or directory
compilation terminated.
make[1]: *** [/esp32-dir/esp-idf/make/component_wrapper.mk:177：factory_restore.o] 错误 1
make: *** [e:/msys32/esp32-dir/esp-idf/make/project.mk:386：factory_restore-build] 错误 2
make: *** 正在等待未完成的任务....
CC hw_random.o
CC emac_dev.o
CC src/secure_boot_signatures.o
CC cpu_freq.o
CC libcoap/src/debug.o
AR libethernet.a
CC timer.o
CC libcoap/src/encode.o
CC event_loop.o
CC src/secure_boot.o
CC sdmmc_transaction.o
CC rmt.o
CC libcoap/src/mem.o
CC src/bootloader_flash.o
CC core_dump.o
AR libbootloader_support.a
CC libcoap/src/net.o
CC pcnt.o
CC gpio.o
CC intr_alloc.o
CC libcoap/src/option.o
CC src/bootloader_random.o
CC libcoap/src/pdu.o
CC ipc.o
CC ledc.o
CC src/flash_encrypt.o
CC src/esp_image_format.o
CC periph_ctrl.o
CC heap_alloc_caps.o
CC src/flash_partitions.o
CC libcoap/src/resource.o
CC i2s.o
CC src/secure_boot_signatures.o
CC deep_sleep.o
CC uart.o
CC src/secure_boot.o
CC src/bootloader_flash.o
CC phy_init.o
CC crosscore_int.o
AR libbootloader_support.a
CC libcoap/src/str.o
CC cpu_util.o
CC log.o
CC sigmadelta.o
CC libcoap/src/subscribe.o
AR liblog.a
AR libcxx.a
CC lib_printf.o
CC libcoap/src/uri.o
CC spi_master.o
CC port/coap_io_socket.o
CC cpu_start.o
CC flash_qio_mode.o
CC bootloader_start.o
CC task_wdt.o
CC event_default_handlers.o
CC panic.o
AR libcoap.a
AR libmain.a
AR libdriver.a
LD bootloader.elf
AR libesp32.a
esptool.py v2.0-beta2

KBS_CXG@KBS_CXG-PC MSYS /e/msys32/esp-aliyun/examples/solutions/smart_light
$

xuhongv, 2019-10-08:
安装我那博文来，准没错

ljy770, 2019-10-16:
@hscxg  尝试一下命令
rm sdkconfig
make clean
make chip=esp32 defconfig
make -j8

hscxg, 2019-10-17:
哎, 捣腾了很多次, 编译都不成功, 太麻烦了.  算了, 这个ESP32还是直接用阿里云的SDK包, 它的编译说明也很详细, 编译很容易通过.

xuhongv, 2019-10-17:
这么难的吗  等我出视频可好？ 100%过！

hscxg, 2019-10-18:
好的, 期待中......., 记得到时通知我哦 , 谢谢!

closed 2019-11-06 by Eavo