espressif/esp-aliyun #147: (near initialization for 'wifi_config.ap') [-Werror=missing-braces]怎么处理的

Rocklarry, 2019-10-10:
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:59:5: error: missing braces around initializer [-Werror=missing-braces]
     wifi_config_t wifi_config = {0};
     ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:59:5: error: (near initialization for 'wifi_config.ap') [-Werror=missing-braces]
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:59:5: warning: missing initializer for field 'password' of 'wifi_ap_config_t' [-Wmissing-field-initializers]
In file included from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event.h:22:0,
                 from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event_loop.h:22,
                 from /home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:37:
/home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_wifi_types.h:206:13: note: 'password' declared here
     uint8_t password[64];       /**< Password of ESP8266 soft-AP */
             ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c: In function 'conn_mgr_save_wifi_config':
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:87:5: error: missing braces around initializer [-Werror=missing-braces]
     wifi_config_t wifi_config = {0};
     ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:87:5: error: (near initialization for 'wifi_config.ap') [-Werror=missing-braces]
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:87:5: warning: missing initializer for field 'password' of 'wifi_ap_config_t' [-Wmissing-field-initializers]
In file included from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event.h:22:0,
                 from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event_loop.h:22,
                 from /home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:37:
/home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_wifi_types.h:206:13: note: 'password' declared here
     uint8_t password[64];       /**< Password of ESP8266 soft-AP */
             ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c: In function 'conn_mgr_set_wifi_config_ext':
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:234:5: error: missing braces around initializer [-Werror=missing-braces]
     wifi_config_t wifi_config = {0};
     ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:234:5: error: (near initialization for 'wifi_config.ap') [-Werror=missing-braces]
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:234:5: warning: missing initializer for field 'password' of 'wifi_ap_config_t' [-Wmissing-field-initializers]
In file included from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event.h:22:0,
                 from /home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_event_loop.h:22,
                 from /home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c:37:
/home/esp8266/work/ESP8266_RTOS_SDK/components/esp8266/include/esp_wifi_types.h:206:13: note: 'password' declared here
     uint8_t password[64];       /**< Password of ESP8266 soft-AP */
             ^
/home/esp8266/work/esp-aliyun/conn_mgr/conn_mgr.c: At top level:
cc1: warning: unrecognized command line option "-Wno-frame-address" [enabled by default]
cc1: some warnings being treated as errors
make[1]: *** [conn_mgr/conn_mgr.o] Error 1
make: *** [component-esp-aliyun-build] Error 2

ljy770, 2019-10-10:
@Rocklarry 是结构体中有多维数组初始化的问题. 
在出现
`wifi_config_t wifi_config = {0};`
的地方改成
```
wifi_config_t wifi_config;
memset(&wifi_config, 0, sizeof(wifi_config_t));
```
试试还会不会有编译错误

simonliu009, 2019-11-02:
可能是编译工具链版本问题，4.8.5就会出这个问题。
改为5.2.0版本试试。

closed 2019-11-06 by Eavo