espressif/esp-aliyun #154: OTA时出现Failed to verify peer certificate 28!

szd117, 2019-10-31:
1.进行了 make chip=esp8266 defconfig
2.make menuconfig 修改了串口为COM7
3.make -j8 成功
4.make flash成功，正常运行，配网后成功连接。
（阿里云物联网平台查得固件版本为0.0.1）
接下来准备新的固件
1.make menuconfig修改了Device version为0.0.2
2.make -j8成功
3.阿里云上传smart_light.bin，验证固件
串口显示如下：
[0;32mI (3397937) ota: esp_ota_begin succeeded[0m
[0;32mI (3397937) ota: Please Wait. This may take time[0m
[0;31mE (3398497) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (3398507) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (3398517) esp-tls: Failed to open new connection[0m
[1;31m[err] httpclient_connect(422): TCP or TLS connect failed, rc = -1

[0m[0;31mE (3400037) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (3400037) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (3400047) esp-tls: Failed to open new connection[0m
[1;31m[err] httpclient_connect(422): TCP or TLS connect failed, rc = -1

ljy770, 2019-10-31:
esp-aliyun 版本是否更新到 master 最新? 如果没有更新到最新试试.
eps8266 在 OTA 升级时再建立 TLS  连接会内存不够, 已经改成了 tcp 连接.

szd117, 2019-10-31:
已经更新到最新，可以正常升级了，万分感谢

closed 2019-11-06 by Eavo