espressif/esp-aliyun #156: 如何TCP连接阿里IOT平台，编译选项选MQTT中现在了CONFIG_NO_TLS=y，但是没有任何效果。log输出显示还是调用了TLS连接

huozhe0930, 2019-11-03:
log打印信息如下：
[0m[0;32mI (7938) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;32mI (8008) TCP_CLIENT: Successfully connected[0m
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering EVP_get_cipherbyname
wolfSSL Entering TLSv1_2_client_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
wolfSSL Leaving WOLFSSL_CTX_new, return 0
wolfSSL Entering wolfSSL_CTX_load_verify_buffer
Processing CA PEM file
wolfSSL Entering PemToDer
Adding a CA
308203753082025da003020102020b040000000001154b5ac394300d06092a864886f70d01010505003057310b300906035504061302424531193017060355040a1310476c6f62616c5369676e206e762d73613110300e060355040b1307526f6f74204341311b301906035504031312476c6f62616c5369676e20526f6f74204341301e170d3938303930313132303030305a170d3238303132383132303030305a3057310b300906035504061302424531193017060355040a1310476c6f62616c5369676e206e762d73613110300e060355040b1307526f6f74204341311b301906035504031312476c6f62616c5369676e20526f6f7420434130820122300d06092a864886f70d01010105000382010f003082010a0282010100da0ee6998dcea3e34f8a7efbf18b83256bea481ff12ab0b9951104bdf063d1e26766cf1cddcf1b482bee8d898e9aaf298065abe9c72d12cbab1c4c7007a13d0a30cd158d4ff8ddd48c50[0;32mI (8238) i2c_task: ======measurement result======[0m
[0;32mI (8248) i2c_task: port:[0] type:[TMP112]  temp[2938][0m
[0;32mI (8258) i2c_task: port:[0] type:[TMP112]  temp[2856][0m
151cef50eec42ef7fce952f2917de06dd535308e5e4373f241e9d56ae3b2893a5639386f063c88695b2a4dc5a754b86c89cc9bf93ccae5fd89f5123c927896d6dc746e934461d18dc746b2750e86e8198ad5[0;32mI (8288) i2c_task: port:[0] type:[TMP112]  temp[2900][0m
[0;32mI (8298) i2c_task: port:[0] type:[TMP112]  temp[2863][0m
[0;32mI (8298) i2c_task: port:[1] type:[TMP117]  temp[2880][0m
6d6cd5781695a2e9c80a38ebf224134f73549313853a1bbc1e34b58b058cb9778bb1db1f2091ab09536e90ce7b3774b97047912251631679aeb1ae412608c8192bd146aa48d6642ad78334ff2c2ac16c19434a0785e7d37cf62168efeaf2529f7f9390cf0203010001a3423040300e0603551d0f0101ff040403020106300f0603551d130101ff040530030101ff301d0603551d0e04160414607b661a450d97ca89502f7d04cd34a8fffcfd4b300d06092a864886f70d01010505000382010100d673e77c4f76d08dbfecbaa2be34c52832b57cfc6c9c2c2bbd099e53bf6b5eaa1148b6e508a3b3ca3d614dd34609b33ec3a0e363551bf2baefad39e143b938a3e62f8a263befa05056f9c60afd38cdc40b705194979804dfc35f94d515c914419cc45d7564150dff5530ec868fff0def2cb96346f6aafcdfbc69fd2e1248649ae095f0a6ef298f01b115b50c1da5fe692c6924781eb3a71c7162eecac897ac175d8ac2f847866e2ac4563195d06789852bf96ca65d469d0caa82e49951dd70b7db563d61e46ae15cd6f6fe3dde41cc07ae6352bf5353f42be9c7fdb6f7825f85d24118db81b3041cc51fa4806f1520c9de0c880a1dd66655e2fc48c9292669e0
wolfSSL Entering GetExplicitVersion
wolfSSL Entering GetSerialNumber
Got Cert Header
wolfSSL Entering GetAlgoId
wolfSSL Entering GetObjectId()
Got Algo ID
Getting Cert Name
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
Date BEFORE check failed
Getting Cert Name
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
wolfSSL Entering OBJ_osn2nid
Got Subject Name
wolfSSL Entering GetAlgoId
wolfSSL Entering GetObjectId()
Got Key
Parsed Past Key
wolfSSL Entering DecodeCertExtensions
wolfSSL Entering GetObjectId()
wolfSSL Entering DecodeKeyUsage
wolfSSL Entering GetObjectId()
wolfSSL Entering DecodeBasicCaConstraint
wolfSSL Entering GetObjectId()
wolfSSL Entering DecodeSubjKeyId
wolfSSL Entering GetAlgoId
wolfSSL Entering GetObjectId()
	Parsed new CA
	Freeing Parsed CA
	Freeing der CA
		OK Freeing der CA
wolfSSL Leaving AddCA, return -150
wolfSSL error occurred, error = 150 line:5221 file:/home/zhangjunhao/esp/ESP826
CA Parse failed, with progress in file.
Search for other certs in file
wolfSSL Entering wolfSSL_CTX_set_verify
wolfSSL Entering SSL_new
wolfSSL Leaving SSL_new, return 0
wolfSSL Entering SSL_set_fd
wolfSSL Entering SSL_set_read_fd
wolfSSL Leaving SSL_set_read_fd, return 1
wolfSSL Entering SSL_set_write_fd
wolfSSL Leaving SSL_set_write_fd, return 1
wolfSSL Entering SSL_connect()
wolfSSL Entering SendClientHello
growing output buffer

Data to send
	16 03 03 00 4f 01 00 00 4b 03 03 a7 30 ae 1f 29 | ....O...K...0..)
	6f 89 85 ac 97 77 c1 61 b4 a1 c2 0c 09 f5 5b 74 | o....w.a......[t
	db 15 e6 06 54 9d eb 13 b8 5e 63 00 00 14 c0 27 | ....T....^c....'
	c0 23 c0 0a c0 09 c0 14 c0 13 00 3d 00 3c 00 35 | .#.........=.<.5
	00 2f 01 00 00 0e 00 0d 00 0a 00 08 04 03 02 03 | ./..............
	04 01 02 01                                     | ....
Shrinking output buffer

wolfSSL Leaving SendClientHello, return 0
connect state: CLIENT_HELLO_SENT
received 28 bytes from tcp server[122.114.156.102]
a5a513032010000100010c22384e2019110304052001001c00010025
[0;32mI (8718) TCP_CLIENT: decodeFrame[0m
[0;32mI (8728) TCP_CLIENT: sensor data response from server[0m
Data received
	16 03 03 0d d0                                  | .....
growing input buffer
Data received
	02 00 00 46 03 03 5d be 98 50 cf ee 5f 55 a2 14 | ...F..]..P.._U..
	8a cd f6 eb 1b d8 7b 3c 12 40 6e 1c 4a 10 54 a2 | ......{<.@n.J.T.
	99 35 db 2b 44 04 20 5d be 98 50 c8 f1 b5 48 d8 | .5.+D. ]..P...H.
	1f 8a 53 41 d2 0a 50 28 86 61 cd fb f2 1e c3 60 | ..SA..P(.a.....`
	cb 96 d2 90 85 79 0c c0 27 00 0b 00 0c 31 00 0c | .....y..'....1..
	2e 00 07 bb 30 82 07 b7 30 82 06 9f a0 03 02 01 | ....0...0.......
	02 02 0c 2c 75 bc 1a 39 13 80 ea 12 a3 27 5a 30 | ...,u..9.....'Z0
	0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 66 | ...*.H........0f
	31 0b 30 09 06 03 55 04 06 13 02 42 45 31 19 30 | 1.0...U....BE1.0
	17 06 03 55 04 0a 13 10 47 6c 6f 62 61 6c 53 69 | ...U....GlobalSi
	67 6e 20 6e 76 2d 73 61 31 3c 30 3a 06 03 55 04 | gn nv-sa1<0:..U.
	03 13 33 47 6c 6f 62 61 6c 53 69 67 6e 20 4f 72 | ..3GlobalSign Or
	67 61 6e 69 7a 61 74 69 6f 6e 20 56 61 6c 69 64 | ganization Valid
	61 74 69 6f 6e 20 43 41 20 2d 20 53 48 41 32 35 | ation CA - SHA25
	36 20 2d 20 47 32 30 1e 17 0d 31 39 30 38 30 31 | 6 - G20...190801
	30 32 32 36 30 37 5a 17 0d 32 30 30 38 30 31 30 | 022607Z..2008010
received record layer msg
wolfSSL Entering DoHandShakeMsg()
wolfSSL Entering DoHandShakeMsgType
processing server hello
wolfSSL Entering DoServerHello
wolfSSL Entering VerifyClientSuite
wolfSSL Leaving DoServerHello, return 0
wolfSSL Leaving DoHandShakeMsgType(), return 0
wolfSSL Leaving DoHandShakeMsg(), return 0
More messages in record
received record layer msg
wolfSSL Entering DoHandShakeMsg()
wolfSSL Entering DoHandShakeMsgType
processing certificate
wolfSSL Entering DoCertificate
wolfSSL Entering ProcessPeerCerts
Loading peer's cert chain
[image]

ljy770, 2019-11-04:
您好,
CONFIG_NO_TLS=y 是为了OTA升级时再建立一条 TLS 连接, 8266 会内存不够设置的, 只针对 OTA 升级时 http 连接设为 NO TLS, MQTT 连接 阿里云仍然为 TLS 连接.

- 请确认 esp-aliyun 使用的是 master 最新版?
- 是否对原始版本进行了修改和添加其他功能,导致 heap 不够用而 MQTT TLS 连接失败?  TLS 连接时需要 available heap 至少 26 K, 可用 esp_get_free_heap_size(), esp_get_minimum_free_heap_size() 查看.
- "TCP_CLIENT: Successfully connected" 这段打印没有找到对应位置, 是调用的哪里? log中有出现 "i2c_task" 是另外添加的?
- 如果确实因为添加了其他功能导致可用内存不够 TLS 连接失败, 想把 MQTT 连接改成 基于 TCP连接, 在 infra_config.h 中 注释宏定义: //#define SUPPORT_TLS

huozhe0930, 2019-11-04:
你好，谢谢您的回复。
在添加ota前，一切都是OK的。添加ota后，就不行了。
然后我就退回到添加ota前版本，还是不行（之前一直可以正常连接阿里IOT平台的）！
再更新所有SDK，包括8266的SDK和esp-aliyun的SDK，任然不行。
我添加自己的工程文件，没有动过esp-aliyunSDK的任何东西的。esp8266，我多做了一个I2C接口。
除了阿里云，我自己还启动了一个TCP连接。但是，把这些都去掉，仍然不能连接到阿里IOT平台。
我又编译了example 的工程样例smart_light，仍旧不能连接阿里IOT平台。我更换新的开发板，还是不行！才提交的问题单。
您的意思是内存不足导致的TLS连接失败吗？为什么会突然内存不足了呢？

ljy770, 2019-11-04:
您说 "编译了example 的工程样例smart_light，仍旧不能连接阿里IOT平台", 这就比较奇怪了, 请把这时的 log 发一下看看. 
您说的添加 OTA 是指? 能具体描述一下吗?

huozhe0930, 2019-11-04:
正在编译smart_light工程（更改配网为airkiss），我说的添加OTA，是指把样例中的OTA，添加到我自己的工程中。

huozhe0930, 2019-11-04:
smart light 打印的log如下：

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa91f4 (692724) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b920c vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b99b8 vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9b5c vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (553) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (563) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (773) phy_init: phy ver: 1056_13[0m
[0;32mI (773) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (773) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (783) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3313) app main: found the stored AP
[0m
[0;32mI (3313) app main:                  huodayong    |     -52    
[0m
[0;32mI (3323) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6773) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7573) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7573) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7593) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (7863) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7863) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7873) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (9013) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (9013) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (9023) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (10183) esp-tls: wolfSSL_connect returned -0x1[0m
[... 112 lines cut ...]
[0m[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[0;31mE (28363) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send 騈鹡ect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (29313) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (29313) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (29323) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (30473) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (30473) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (30483) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (31633) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (31633) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (31633) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

huozhe0930, 2019-11-04:
sdk中，我把i2c端口改成了2个，这个我用了很长时间，对任何东西都没影响。所有，SDK版本显示会有一个dirty。

huozhe0930, 2019-11-04:
所有SDK都是最新版本，example也是从SDK中重新拷贝出来的。

ljy770, 2019-11-04:
您好, 如果您使用的 smart_light example,  连接飞燕平台, 已经包含了 OTA 功能, 不需要再添加 OTA示例.   不然就会如 log 所示出现两个 MQTT 连接, 导致连接失败.

huozhe0930, 2019-11-04:
但是，这个smart_light example，我什么都没动，没添加任何东西，但是还是连接不到阿里IOT平台啊。

huozhe0930, 2019-11-04:
是这样的：我把smart_light从SDK中拷贝出来，修改了配网方式为airkiss，其他什么都没有改动，也没有添加。我发的log，就是这种情况下打印的log，连接不上阿里IOT平台。

huozhe0930, 2019-11-04:
再发一次新的log

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa91f4 (692724) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b920c vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b99b8 vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9b5c vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (553) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (553) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (773) phy_init: phy ver: 1056_13[0m
[0;32mI (773) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (773) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (783) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3313) app main: found the stored AP
[0m
[0;32mI (3313) app main:                  huodayong    |     -59    
[0m
[0;32mI (3323) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6773) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7573) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7573) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7593) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (7863) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7863) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7873) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (9023) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (9023) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (9033) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (10183) esp-tls: wolfSSL_connect returned -0x1[0m
[... 77 lines cut ...]
[0;31mE (20773) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (21933) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (21933) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (21933) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (23093) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (23093) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (23093) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[0;31mE (24113) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (25063) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (250

huozhe0930, 2019-11-04:
下面是app_main的代码片段：
我添加的都注释掉了
case SYSTEM_EVENT_STA_GOT_IP:
		ESP_LOGI(TAG, "!!connect is established!\n");
		
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);

        //if(xTcpClientHandle == NULL)
        //    xTaskCreate(tcp_client_task, "tcp_client", 4096, NULL, 5, &xTcpClientHandle);
        if (linkkit_started == false) 
	    {
            wifi_config_t wifi_config = {0};
            if (conn_mgr_get_wifi_config(&wifi_config) == ESP_OK &&
                strcmp((char *)(wifi_config.sta.ssid), HOTSPOT_AP) &&
                strcmp((char *)(wifi_config.sta.ssid), ROUTER_AP)) 
                {
                    xTaskCreate((void (*)(void *))linkkit_main, "lightbulb", 10240, NULL, 5, NULL);
                    linkkit_started = true;
                }
        }
		//if (ota_started == false) 
		//{
        //    xTaskCreate((void (*)(void *))ota_main, "ota_example", 20480, NULL, 5, NULL);
        //    ota_started = true;
        //}
        
        break;

ljy770, 2019-11-04:
三元组 烧写正确的吧?
在 HAL_SSL_Establish 里添加打印:
printf("heap %d min %d", esp_get_free_heap_size(), esp_get_minimum_free_heap_size()); 
    struct esp_tls *tls = esp_tls_conn_new(host, strlen(host), port, &cfg);
printf("heap %d min %d", esp_get_free_heap_size(), esp_get_minimum_free_heap_size()); 
头文件添加:
#include "esp_system.h"
查看一下 打印的 heap.

huozhe0930, 2019-11-04:
三元组是正确的。好的。

huozhe0930, 2019-11-04:
打印了很多出来，
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa9278 (692856) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b9290 vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b9a3c vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9be0 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (761) phy_init: phy ver: 1056_13[0m
[0;32mI (771) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (771) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (781) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3311) app main: found the stored AP
[0m
[0;32mI (3311) app main:                  huodayong    |     -50    
[0m
[0;32mI (3321) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6771) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7581) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7581) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7601) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (7901) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7901) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7901) esp-tls: Failed to open new connection[0m
heap 54224 min 40600heap 54224 min 40600[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (9071) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (9071) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (9071) esp-tls: Failed to open new connection[0m
heap 54340 min 40588heap 54340 min 40588[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (10251) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (10251) esp-tls: Failed to verify peer certificate 28![0m
[... 79 lines cut ...]
[0;31mE (22371) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (22381) esp-tls: Failed to open new connection[0m
heap 54340 min 40472heap 54340 min 40472[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[0;31mE (23561) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (23561) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (23571) esp-tls: Failed to open new connection[0m
heap 54340 min 40472heap 54340 min 40472[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[0;31mE (24591) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[... cut ...]
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;31mE (25551) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (25551) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (25551) esp-tls: Failed to open new connection[0m
heap 54340 min 40472heap 54340 min 40472[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m

huozhe0930, 2019-11-04:
我知道你的意思了，要再申请前和申请后打印。我重新编译一下。

huozhe0930, 2019-11-04:
上面打印有误，放错了位置。

huozhe0930, 2019-11-04:
新的log：

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa927c (692860) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b9294 vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b9a40 vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9be4 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (771) phy_init: phy ver: 1056_13[0m
[0;32mI (771) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (771) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (781) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3311) app main: found the stored AP
[0m
[0;32mI (3311) app main:                  huodayong    |     -56    
[0m
[0;32mI (3321) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6771) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7571) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7571) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7591) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mheap 54500 min 54408[0;31mE (7901) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7901) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7911) esp-tls: Failed to open new connection[0m
heap 54340 min 40600[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40600[0;31mE (9081) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (9081) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (9091) esp-tls: Failed to open new connection[0m
heap 54340 min 40588[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40588[0;31mE (10251) esp-tls: wolfSSL_connect returned -0x1[0m
[... 89 lines cut ...]
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]

[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mheap 54340 min 40588[0;31mE (25361) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (25361) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (25361) esp-tls: Failed to open new connection[0m
heap 54224 min 40588[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54364 min 40588[0;31mE (26521) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (26521) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (26531) esp-tls: Failed to open new connection[0m
heap 54340 min 40588[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40588[0;31mE (27691) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (27691) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (27701) esp-tls: Failed to open new connection[0m
heap 54340 min 40588[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m

huozhe0930, 2019-11-04:
这个更清晰：
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa927c (692860) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b9294 vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b9a40 vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9be4 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (551) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (761) phy_init: phy ver: 1056_13[0m
[0;32mI (771) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (771) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (781) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3311) app main: found the stored AP
[0m
[0;32mI (3311) app main:                  huodayong    |     -53    
[0m
[0;32mI (3321) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6771) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7591) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7591) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7611) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mheap 54500 min 54408[0;31mE (7871) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7871) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7881) esp-tls: Failed to open new connection[0m
heap 54340 min 40496[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40496[0;31mE (9051) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (9061) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (9061) esp-tls: Failed to open new connection[0m
heap 54340 min 40484[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40484[0;31mE (10221) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (10221) esp-tls: Failed to verify peer certificate 28![0m
[... 57 lines cut ...]
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;31m[err] _mqtt_connect(276): mqtt connect failed
[0m[0;31mE (20611) app main: IOTX_CONN_CLOUD_FAIL[0m
[1;31m[err] _iotx_linkkit_master_connect(959): DM Start Failed
[0m[1;32;40mlinkkit_thread.441: IOT_Linkkit_Connect Failed
[0m
[1;31m[err] IOT_MQTT_Destroy(610): handler is null
[0m[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (20641) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mheap 54364 min 40484[0;31mE (20861) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (20871) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (20871) esp-tls: Failed to open new connection[0m
heap 54340 min 40484[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mheap 54340 min 40484[0;31mE (22021) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (22031) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (22031) esp-tls: Failed to open new connection[0m
heap 54340 min 40484[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m

huozhe0930, 2019-11-04:
我做了标注：before after：
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa9298 (692888) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b92b0 vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b9a5c vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9c00 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (552) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (552) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,[0;32mI (772) phy_init: phy ver: 1056_13[0m
[0;32mI (772) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (772) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (782) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3312) app main: found the stored AP
[0m
[0;32mI (3312) app main:                  huodayong    |     -62    
[0m
[0;32mI (3322) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6772) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7592) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7592) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7612) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]

[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mbefore heap 54500 min 54408

[0;31mE (7912) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7912) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7912) esp-tls: Failed to open new connection[0m
after heap 54340 min 40368
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mbefore heap 54340 min 40368

[... 33 lines cut ...]
[0mbefore heap 54364 min 40356

拔0*Еj14082) wifi: err2:1250466[0m
214222) wifi: err2:1049834[0m
214422) wifi: err2:976739[0m
614622) wifi: err2:950326[0m
214822) wifi: err2:1214878[0m
15032) wifi: err2:1199233[0m
215232) wifi: err2:920433[0m
215432) wifi: err2:893734[0m
615632) wifi: err2:879714[0m
215732) wifi: err2:891808[0m
216102) wifi: err2:871791[0m
16412) wifi: err2:1188256[0m
216552) wifi: err2:1177817[0m
16652) wifi: err2:1186932[0m
16852) wifi: err2:868886[0m
17122) wifi: err2:1296309[0m
617372) wifi: err2:844043[0m
17472) wifi: err2:1210889[0m
6*17672) wifi: err2:900883[0m
217772) wifi: err2:1187069[0m
617972) wifi: err2:900705[0m
218182) wifi: err2:1357022[0m


huozhe0930, 2019-11-04:
这次没有出现 wifi err2
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa9298 (692888) map[0m
[0;32mI (472) esp_image: segment 1: paddr=0x000b92b0 vaddr=0x3ffe8000 size=0x007a4 (  1956) load[0m
[0;32mI (473) esp_image: segment 2: paddr=0x000b9a5c vaddr=0x3ffe87a4 size=0x0019c (   412) load[0m
[0;32mI (483) esp_image: segment 3: paddr=0x000b9c00 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (552) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (552) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (772) phy_init: phy ver: 1056_13[0m
[0;32mI (772) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (772) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (782) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3312) app main: found the stored AP
[0m
[0;32mI (3312) app main:                  huodayong    |     -54    
[0m
[0;32mI (3322) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6772) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7572) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7572) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7592) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mbefore heap 54500 min 54408

[0;31mE (7822) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (7822) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (7832) esp-tls: Failed to open new connection[0m
after heap 54216 min 40580
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0mbefore heap 54332 min 40580

[0;31mE (8992) esp-tls: wolfSSL_connect returned -0x1[0m
[... 157 lines cut ...]
[0;31mE (28452) esp-tls: Failed to open new connection[0m
after heap 54340 min 40348
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[0;31mE (29472) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0mbefore heap 54364 min 40348

癗0*Еj30482) wifi: err2:1134168[0m
[0;31mE (30762) esp-tls: wolfSSL_connect returned -0x1[0m
[0;31mE (30762) esp-tls: Failed to verify peer certificate 28![0m
[0;31mE (30762) esp-tls: Failed to open new connection[0m
after heap 54340 min 40348
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m

ljy770, 2019-11-04:
heap 是足够的,不是这个原因.
`static esp_err_t conn_mgr_wifi_event_loop_handler(void *ctx, system_event_t *event)
{
    system_event_info_t *info = &event->event_info;
    switch (event->event_id) {
        case SYSTEM_EVENT_STA_GOT_IP:
            conn_mgr_save_wifi_config();
            #if CONFIG_SSL_USING_WOLFSSL
            **conn_mgr_obtain_time();**
            #endif
            break;`

您的代码里是不是少了这一句: conn_mgr_obtain_time() ?

huozhe0930, 2019-11-04:
我加上试下。

huozhe0930, 2019-11-04:
我找不到这个函数啊：conn_mgr_wifi_event_loop_handler()
我加到wifi event hander中就可以吗？

huozhe0930, 2019-11-04:
conn_mgr_obtain_time();---这个添加进去编译错误。整个工程中，没招到这个函数。

ljy770, 2019-11-04:
在 conn_mgr/conn_mgr.c 中: https://github.com/espressif/esp-aliyun/blob/master/conn_mgr/conn_mgr.c

huozhe0930, 2019-11-04:
找到了，在conn_mgr.c中
有这个函数的
static esp_err_t conn_mgr_wifi_event_loop_handler(void *ctx, system_event_t *event)
{
    system_event_info_t *info = &event->event_info;
    switch (event->event_id) {
        case SYSTEM_EVENT_STA_GOT_IP:
            conn_mgr_save_wifi_config();
            #if CONFIG_SSL_USING_WOLFSSL
            conn_mgr_obtain_time();
            #endif
            break;

        case SYSTEM_EVENT_STA_DISCONNECTED:
            ESP_LOGE(TAG, "Disconnect reason : %d", info->disconnected.reason);
#ifdef CONFIG_IDF_TARGET_ESP8266
            if (info->disconnected.reason == WIFI_REASON_BASIC_RATE_NOT_SUPPORT) {
                /*Switch to 802.11 bgn mode */
                esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCAL_11B | WIFI_PROTOCAL_11G | WIFI_PROTOCAL_11N);
            }
#endif
            esp_wifi_connect();
            break;

        default:
            break;
    }

    /** The application loop event handle */
    if (hal_wifi_system_cb) {
        hal_wifi_system_cb(ctx, event);
    }

    return ESP_OK;
}

huozhe0930, 2019-11-04:
我我自己的wifi hander里还需要自己显示的调用一下？

ljy770, 2019-11-04:
您给出的 log 里没有出现打印" SNTP get time success", 检查一下是否没有调用到,
app_main -> conn_mgr_init -> esp_event_loop_init(conn_mgr_wifi_event_loop_handler, NULL)

还有编译之前是否执行了 make chip=esp8266 defconfig, 如果执行了, sdkconfig 里会出现 CONFIG_SSL_USING_WOLFSSL=y 的定义.

huozhe0930, 2019-11-06:
环境清空，重新搭建后，都好了。。。

closed 2019-11-06 by huozhe0930