espressif/esp-aliyun #158: TCP（不是TLS）连接阿里IOT后，出现_mqtt_cycle(1597): error occur rc=-27

huozhe0930, 2019-11-05:
用的是 smart_light 未做其他改动（改了连接方式为TCP）。
打印log如下：
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0xbb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x52
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0x52
csum 0x52
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 15:51:26[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : DIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (131) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (143) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (155) boot: End of partition table[0m
[0;32mI (161) boot: No factory image, trying OTA 0[0m
[0;32mI (169) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x80330 (525104) map[0m
[0;32mI (401) esp_image: segment 1: paddr=0x00090348 vaddr=0x3ffe8000 size=0x0079c (  1948) load[0m
[0;32mI (403) esp_image: segment 2: paddr=0x00090aec vaddr=0x3ffe879c size=0x0019c (   412) load[0m
[0;32mI (413) esp_image: segment 3: paddr=0x00090c90 vaddr=0x40100000 size=0x06958 ( 26968) load[0m
[0;32mI (437) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (477) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (477) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
,债[0;32mI (677) phy_init: phy ver: 1056_13[0m
[0;32mI (687) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;33mW (687) wrapper_kv: nvs get blob q_rt failed with 1102[0m
[0;32mI (697) factory_rst: quick reboot times 1, don't need to restore[0m
[prt] log level set as: [ 4 ]
Number of access points found: 1
[0;32mI (3227) app main: found the stored AP
[0m
[0;32mI (3227) app main:                  huodayong    |     -47    
[0m
[0;32mI (3237) app main: connect to ap SSID:huodayong password:Hvoday0ng[0m
[0;32mI (6687) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (7487) event: sta ip: 192.168.1.8, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7487) app main: !!connect is established!
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (7507) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [1883]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test|timestamp=2524608000000,securemode=3,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m[0;32mI (7577) iot_import_tcp: establish tcp connection with server(host=a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com port=1883)[0m
[0;32mI (7647) iot_import_tcp: fail to establish tcp[0m
[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=3,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[1;33m[inf] _mqtt_connect(821): mqtt connect success!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/request/a1aGB5BurNF/test!

[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/upgrade/a1aGB5BurNF/test!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test/thing/config/get_reply!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test/thing/config/push!
[... 88 lines cut ...]
[0m[1;35m[wrn] wrapper_mqtt_subscribe_sync(3027): sync subscribe time out!!
[0m[1;31m[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 4
[0m[1;31m[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed
[0m[1;33m[inf] awss_report_token_to_cloud(351): report token res:-1
[0m[0;32mI (11287) udp: success to establish udp, fd=0[0m
[1;31m[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 4
[0m[1;31m[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed
[0m[1;33m[inf] awss_bind_report_statis(70): bind report statis fail
[0m[0;32mI (11327) app main: IOTX_CONN_CLOUD_SUC[0m
[1;31m[err] wrapper_mqtt_subscribe(2859): mqtt client state is error,state = 4
[0m[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...
[0m[1;33m[inf] iotx_mc_handle_reconnect(1749): start to reconnect
[0m[1;31m[err] disconnect_tcp(206): Network->handle = -1
[0m[1;33m[inf] iotx_mc_attempt_reconnect(1722): reconnect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=3,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[0;32mI (11407) iot_import_tcp: establish tcp connection with server(host=a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com port=1883)[0m
[0;32mI (11467) iot_import_tcp: success to establish tcp, fd=1[0m
[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test|timestamp=2524608000000,securemode=3,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test&a1aGB5BurNF
[0m[1;33m[inf] _mqtt_connect(821): mqtt connect success!
[0m[1;33m[inf] iotx_mc_keepalive(1845): network is reconnected!
[0m[1;33m[inf] iotx_cloud_conn_mqtt_event_handle(91): connected,fd = 0
[0m[1;33m[inf] _dm_client_event_cloud_connected_handle(164): IOTX_CM_EVENT_CLOUD_CONNECTED
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test/thing/reset_reply!
[0m[1;33m[inf] awss_report_token_to_cloud(351): report token res:20
[0m[1;35m[wrn] MQTTSubscribe(2185): dup sub,topic = /sys/a1aGB5BurNF/test/thing/reset_reply
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test/thing/reset_reply!

lhespress, 2019-12-05:
从提供的信息看，创建 TCP 连接时出现异常的频率较高，建议切换干净的网络环境测试。如依然存在问题，请反馈。

qiufeng81888, 2019-12-12:
[err] _mqtt_cycle(1597): error occur rc=-27
8266上 发现 相同问题。

lhespress, 2019-12-18:
@qiufeng81888 
ESP8266: 请切换至 V3.2 版本
ESP32: 请切换至 V3.3.1 版本

qiufeng81888, 2019-12-18:
ESP8266: 请切换至 V3.2 版本，在哪里下载？

lhespress, 2019-12-18:
https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.2

huozhe0930, 2019-12-18:
我用的就是V3.2; log打印的是
[0;32mI (49) boot: ESP-IDF v3.2-26-gefd0a545-dirty 2nd stage bootloader
你的意思是要下载新版本？

lhespress, 2019-12-18:
git rev-parse --short HEAD 来获取 commit id: 看是否与 3ad3fd7 相一致

huozhe0930, 2019-12-19:
我更新到了最新的SDK
git clone https://github.com/espressif/ESP8266_RTOS_SDK.git
git checkout release/v3.2
我用命令 git rev-parse --short HEAD，查看的是e67c2c5a；要怎样来获取你说的版本呢？

lhespress, 2019-12-19:
git checkout v3.2

huozhe0930, 2019-12-19:
获取到了正确版本，谢谢。再测试下通信是否正常

huozhe0930, 2019-12-19:
windows 下编译无法通过，提示如下错误信息：
$ make
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-100-ge567ec7b
Expected to see version: crosstool-ng-1.22.0-92-g8facf4c
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-100-ge567ec7b
Expected to see version: crosstool-ng-1.22.0-92-g8facf4c
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/coap/port/coap_io_socket.o
D:/msys32/home/huoda/esp/ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c: In function 'coap_new_endpoint':
D:/msys32/home/huoda/esp/ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c:162:5: error: implicit declaration of function 'close' [-Werror=implicit-function-declaration]
     close (sockfd);
     ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/huoda/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:286：port/coap_io_socket.o] 错误 1
make: *** [D:/msys32/home/huoda/esp/ESP8266_RTOS_SDK/make/project.mk:517：component-coap-build] 错误 2

huozhe0930, 2019-12-19:
v3.2的检出信息如下：
$ git checkout v3.2
正在检出文件: 100% (1276/1276), 完成.
注意：正在检出 'v3.2'。

您正处于分离头指针状态。您可以查看、做试验性的修改及提交，并且您可以通过另外
的检出分支操作丢弃在这个状态下所做的任何提交。

如果您想要通过创建分支来保留在此状态下所做的提交，您可以通过在检出命令添加
参数 -b 来实现（现在或稍后）。例如：

  git checkout -b <新分支名>

HEAD 目前位于 3ad3fd78 Merge branch 'docs/fix_script_python3_data_type_error' into 'master'

huozhe0930, 2019-12-20:
编译错误原因找到了。这个版本只能用来对接阿里sdk，其他工程出错。
编译后运行，错误依旧。连接上会马上断掉，再重联后正常。

jiangzhi18, 2019-12-31:
请问这是什么原因导致的？

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo

jony-xia, 2020-05-08:
mqtt_cycle(1597): error occur rc=-27    的问题还未解决