espressif/esp-aliyun #159: OTA升级，接收到很多空包，升级不成功，死在接收流程。

huozhe0930, 2019-11-06:
信息打印如下：
[1;33m[inf] dm_msg_request(205): DM Send Message, URI: /sys/a1aGB5BurNF/test3/thing/event/property/post, Payload: {"id":"534","version":"1.0","params":{"Sensordata": "tmp1:38.0;tmp2:37.0;tmp3:36,7,tmp4:37.1;tmp5:28.5:hum1:75;acc_x:322.05;acc_y:55332,acc_z:6327;"},"method":"thing.event.pr ...
[0m[1;33m[inf] dm_client_publish(131): Publish Result: 0
[0m[1;32;40muser_post_property.354: Post Property Message ID: 534[0m
[1;33m[inf] dm_msg_proc_thing_event_post_reply(318): Event Id: property
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 32
[0m[1;33m[inf] _iotx_linkkit_event_callback(245): Receive Message: {"id":534,"code":200,"devid":0,"payload":{}}
[0m[1;32;40muser_report_reply_event_handler.135: Message Post Reply Received, Message ID: 534, Code: 200, Reply: {}[0m
[0;32mI (1081171) TCP_CLIENT: decodeFrame[0m
[0;32mI (1081171) TCP_CLIENT: sensor data response from server[0m
[1;33m[inf] dm_msg_request(205): DM Send Message, URI: /sys/a1aGB5BurNF/test3/thing/event/property/post, Payload: {"id":"535","version":"1.0","params":{"Sensordata": "tmp1:38.0;tmp2:37.0;tmp3:36,7,tmp4:37.1;tmp5:28.5:hum1:75;acc_x:322.05;acc_y:55332,acc_z:6327;"},"method":"thing.event.pr ...
[0m[1;33m[inf] dm_client_publish(131): Publish Result: 0
[0m[1;32;40muser_post_property.354: Post Property Message ID: 535[0m
[1;33m[inf] dm_msg_proc_thing_event_post_reply(318): Event Id: property
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 32
[0m[1;33m[inf] _iotx_linkkit_event_callback(245): Receive Message: {"id":535,"code":200,"devid":0,"payload":{}}
[0m[1;32;40muser_report_reply_event_handler.135: Message Post Reply Received, Message ID: 535, Code: 200, Reply: {}[0m
[0;32mI (1083171) TCP_CLIENT: decodeFrame[0m
[0;32mI (1083171) TCP_CLIENT: sensor data response from server[0m
[1;33m[inf] dm_msg_request(205): DM Send Message, URI: /sys/a1aGB5BurNF/test3/thing/event/property/post, Payload: {"id":"536","version":"1.0","params":{"Sensordata": "tmp1:38.0;tmp2:37.0;tmp3:36,7,tmp4:37.1;tmp5:28.5:hum1:75;acc_x:322.05;acc_y:55332,acc_z:6327;"},"method":"thing.event.pr ...
[0m[1;33m[inf] dm_client_publish(131): Publish Result: 0
[0m[1;32;40muser_post_property.354: Post Property Message ID: 536[0m
[1;33m[inf] dm_msg_proc_thing_event_post_reply(318): Event Id: property
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 32
[0m[1;33m[inf] _iotx_linkkit_event_callback(245): Receive Message: {"id":536,"code":200,"devid":0,"payload":{}}
[0m[1;32;40muser_report_reply_event_handler.135: Message Post Reply Received, Message ID: 536, Code: 200, Reply: {}[0m
[0;32mI (1085171) TCP_CLIENT: decodeFrame[0m
[0;32mI (1085171) TCP_CLIENT: sensor data response from server[0m
[1;33m[inf] _dm_fota_send_new_config_to_user(51): Send To User: {"version":"0.0.3"}
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 42
[0m[1;33m[inf] _iotx_linkkit_event_callback(245): Receive Message: {"version":"0.0.3"}
[0m[1;32;40muser_fota_event_handler.306: New Firmware Version: 0.0.3[0m

[0;32mI (1086021) ota: Starting OTA...[0m
[0;32mI (1086031) ota: Writing to partition subtype 17 at offset 0x110000[0m
[0;32mI (1088441) ota: esp_ota_begin succeeded[0m
[0;32mI (1088441) ota: Please Wait. This may take time[0m
[0;32mI (1088451) iot_import_tcp: establish tcp connection with server(host=iotx-ota.oss-cn-shanghai.aliyuncs.com port=80)[0m
[0;32mI (1088511) iot_import_tcp: success to establish tcp, fd=3[0m
[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32

[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32

[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 992
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[... 220 lines cut ...]
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 0
[0m

simonliu009, 2019-11-16:
我也遇到了这个问题，收到大量空包，然后直到超时才停止接收

simonliu009, 2019-11-16:
更新一下：我的另外一个项目就没问题，我看看是不是和freeheap太小有关系....

simonliu009, 2019-11-16:
确认是freeheap引起的问题，freeheap在12k的时候有问题，后来缩减任务内存，将freeheap增加到18k，OTA成功。

huozhe0930, 2019-11-16:
我的好像也是free heap的问题，我是tls改成tcp后解决的。

closed 2019-11-16 by huozhe0930

simonliu009, 2019-12-05:
更新一下，我的另外一个项目也遇到了这个问题，但是freeheap有30k+。有时候可以有时候不行，代码改动只有一个参数而已。

temp4gh, 2020-02-15:
Same issue, but not happened everytime. With 20k+ freeheap.

temp4gh, 2020-02-16:
Serious issue！OTA would cause firmware damaged,with offical example ,ota_example_mqtt,no matter with or without tls:
The logs:

[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[err] _utils_check_deadloop(254): extend timer for too many times, exit
[err] httpclient_common(562): httpclient_recv_response is error,ret = -8
[inf] httpclient_close(502): client disconnected
[err] ofc_Fetch(70): fetch firmware failed
[err] IOT_OTA_FetchYield(698): Fetch firmware failed

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16
0x40100000: _stext at ??:?

tail 12
chksum 0xde
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0x57
ho 0 tail 12 room 4
load 0x3ffe8420, len 3524, room 12
tail 8
chksum 0xfa
I (50) boot: ESP-IDF  2nd stage bootloader
I (50) boot: compile time 20:56:52
I (50) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DOUT
I (60) boot: SPI Flash Size : 2MB
I (66) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (94) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (106) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (117) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (129) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (141) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (152) boot: End of partition table
E (159) esp_image: image at 0x110000 has invalid magic byte
E (168) boot: OTA app partition slot 1 is not bootable
I (177) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7a04c (499788) map
E (410) esp_image: unaligned segment length 0x99008122
E (410) boot: OTA app partition slot 0 is not bootable
E (412) boot: No bootable app partitions in the partition table
user code done

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 7196, room 16
0x40100000: _stext at ??:?

tail 12
chksum 0xde
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0x57
ho 0 tail 12 room 4
load 0x3ffe8420, len 3524, room 12
tail 8
chksum 0xfa
I (51) boot: ESP-IDF  2nd stage bootloader
I (51) boot: compile time 20:56:52
I (51) boot: SPI Speed      : 40MHz
I (55) boot: SPI Mode       : DOUT
I (61) boot: SPI Flash Size : 2MB
I (67) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (96) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (107) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (119) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (130) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (142) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (154) boot: End of partition table
E (160) esp_image: image at 0x110000 has invalid magic byte
E (170) boot: OTA app partition slot 1 is not bootable
I (179) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7a04c (499788) map
E (411) esp_image: unaligned segment length 0x99008122
E (411) boot: OTA app partition slot 0 is not bootable
E (413) boot: No bootable app partitions in the partition table
user code done