espressif/esp-aliyun #160: 启动阶段，连接阿里IOT成功后，随即出现连接错误，重连后有时正常，有时一直错误。

huozhe0930, 2019-11-06:
打印log如下：
[0;32mI (28766) app main: SC_STATUS_LINK[0m
[0;32mI (28766) app main: SSID:huodayong[0m
[0;32mI (28766) app main: PASSWORD:Hvoday0ng[0m
[0;32mI (34646) event: sta ip: 192.168.1.17, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (34646) app main: !!connect is established!
[0m
[0;32mI (34646) app main: WiFi Connected to ap[0m
[0;32mI (36646) sntp: Time is not set yet. Connecting to WiFi and getting time over NTP.[0m
[0;32mI (36746) sntp: SNTP get time success
[0m
[1;32;40mlinkkit_main.412: Enter iot linkkit_thread
[0m
[0;32mI (36756) TCP_CLIENT: Socket created[0m
[1;32;40mlinkkit_main.464: Enter iot linkkit_thread loop 3
[0m
[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[1;32;40mlinkkit_main.471: Enter iot linkkit_thread loop 2
[0m
[0;32mI (36776) app main: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1aGB5BurNF.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1aGB5BurNF.test2|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|

[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: test2&a1aGB5BurNF
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!

[0m[0;32mI (36856) TCP_CLIENT: Successfully connected[0m
[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1aGB5BurNF.test2|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=test2&a1aGB5BurNF
[0m[1;33m[inf] _mqtt_connect(821): mqtt connect success!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/request/a1aGB5BurNF/test2!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /ota/device/upgrade/a1aGB5BurNF/test2!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/config/get_reply!
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/config/push!
[0m[1;32;40muser_mqtt_connect_succ_event_handler.338: ITE_MQTT_CONNECT_SUCC[0m
[1;33m[inf] _dm_client_event_cloud_connected_handle(164): IOTX_CM_EVENT_CLOUD_CONNECTED
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/awss/enrollee/match_reply!
[0m[0;32mI (39796) app main: SC_STATUS_LINK_OVER[0m
[0;32mI (39796) app main: Phone ip: 0.0.0.0
[0m
[0;32mI (39796) app main: smartconfig over[0m
[0;33mW (41686) iot_import_tls: Poll timeout or error, errno=No more processes, fd=2, timeout_ms=2000[0m
[0;33mW (41686) iot_import_tls: Poll timeout or error, errno=No more processes, fd=2, timeout_ms=1[0m
[1;31m[err] MQTTSubscribe(2147): run sendPacket error!
[0m[1;31m[err] wrapper_mqtt_subscribe(2875): run MQTTSubscribe error, rc = -14
[0m[1;31m[err] wrapper_mqtt_subscribe(2859): mqtt client state is error,state = 3
[0m[1;31m[err] wrapper_mqtt_subscribe(2859): mqtt client state is error,state = 3
[0m[1;31m[err] wrapper_mqtt_subscribe(2859): mqtt client state is error,state = 3
[0m[0;33mW (41756) wrapper_kv: nvs get blob awss.rst failed with 1102[0m
[1;31m[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 3
[0m[1;31m[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed
[0m[1;33m[inf] awss_report_token_to_cloud(351): report token res:-1
[0m[0;32mI (41796) udp: success to establish udp, fd=0[0m
[1;31m[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 3
[0m[1;31m[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed
[0m[1;33m[inf] awss_bind_report_statis(70): bind report statis fail
[0m[0;32mI (41836) app main: IOTX_CONN_CLOUD_SUC[0m
[1;31m[err] wrapper_mqtt_subscribe(2859): mqtt client state is error,state = 3
[0m[1;31m[err] iotx_mc_keepalive(1855): network is disconnected!
[0m[1;33m[inf] iotx_cloud_conn_mqtt_event_handle(80): disconnected,fd = 0
[0m[1;33m[inf] _dm_client_event_cloud_disconnect_handle(174): IOTX_CM_EVENT_CLOUD_DISCONNECT
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27

[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[... 97 lines cut ...]
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/property/desired/delete_reply!
[0m[0;32mI (46246) app main: IOTX_AWSS_BIND_NOTIFY[0m
[1;32;40muser_awss_status_event_handler.81: Awss Status 4111[0m
[1;33m[inf] awss_notify_dev_info(205): topic:/sys/device/info/notify
[0m[1;33m[inf] awss_notify_dev_info(209): send notify success
[0m[1;33m[inf] dm_client_subscribe(86): Subscribe Result: 21
[0m[1;33m[inf] dm_client_subscribe_all(124): index: 2
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/property/desired/get_reply!
[0m[1;33m[inf] awss_notify_dev_info(205): topic:/sys/device/info/notify
[0m[1;33m[inf] awss_notify_dev_info(209): send notify success
[0m[1;33m[inf] dm_client_subscribe(86): Subscribe Result: 22
[0m[1;33m[inf] dm_client_subscribe_all(124): index: 3
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/service/property/get!
[0m[1;33m[inf] dm_client_subscribe(86): Subscribe Result: 23
[0m[1;33m[inf] dm_client_subscribe_all(124): index: 4
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/service/property/set!
[0m[1;33m[inf] awss_notify_dev_info(205): topic:/sys/device/info/notify
[0m[1;33m[inf] awss_notify_dev_info(209): send notify success
[0m[1;33m[inf] dm_client_subscribe(86): Subscribe Result: 24
[0m[1;33m[inf] dm_client_subscribe_all(124): index: 5
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/service/+!
[0m[1;33m[inf] dm_client_subscribe(86): Subscribe Result: 25
[0m[1;33m[inf] dm_client_subscribe_all(124): index: 6
[0m[1;33m[inf] wrapper_mqtt_subscribe(2879): mqtt subscribe packet sent,topic = /sys/a1aGB5BurNF/test2/thing/deviceinfo/update_reply!
[0m[1;33m[inf] awss_notify_dev_info(205): topic:/sys/device/info/notify

wangqianjiao, 2019-11-07:
您好，请问有解决这个问题吗，我们也遇到跟您的一样，时不时掉线，然后一直重连报错，尤其是断电重新通电初始化连接很容易出现

huozhe0930, 2019-11-07:
没解决，不知道什么原因，查看阿里后台，还经常离线。

huozhe0930, 2019-11-16:
没有解决，HAL的TCP连接一直输出错误信息。没有时间持续跟踪。

lcnja, 2019-11-23:
我也有这个错误, esp8266, 加电启动后, 连接成功, 在阿里平台看到设备在线, 过一段时间就离线了, 然后报错和上面一样的.

lcnja, 2019-11-28:
esp sdk checkout v3.2, 然后稳定运行了

lhespress, 2019-12-18:
@huozhe0930 
ESP8266：请切换 SDK 至 V3.2 版本 
ESP32：请切换 SDK 至 V3.3.1 版本

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo