espressif/esp-aliyun #161: OTA升级完成后，程序没有从新固件引导系统，还在原来固件里继续运行。

huozhe0930, 2019-11-06:
在阿里IOT平台部署了新固件，然后启动OTA流程进行升级。
升级过程很顺利，但是提示升级完成后；程序没有自动引导从新固件启动。
等了若干分钟后按复位键重启设备，发现OTA升级时成功的，阿里平台也显示升级成功了。
log打印如下：
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 216
[0m[1;33m[inf] httpclient_common(570): close http channel
[0m[1;33m[inf] httpclient_close(502): client disconnected
[0m[0;32mI (719815) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x8ea6c (584300) map[0m
[0;32mI (720005) esp_image: segment 1: paddr=0x0019ea84 vaddr=0x3ffe8000 size=0x008b4 (  2228) [0m
[0;32mI (720015) esp_image: segment 2: paddr=0x0019f340 vaddr=0x3ffe88b4 size=0x0019c (   412) [0m
[0;32mI (720025) esp_image: segment 3: paddr=0x0019f4e4 vaddr=0x40100000 size=0x0696c ( 26988) [0m
[0;32mI (720045) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x8ea6c (584300) map[0m
[0;32mI (720225) esp_image: segment 1: paddr=0x0019ea84 vaddr=0x3ffe8000 size=0x008b4 (  2228) [0m
[0;32mI (720235) esp_image: segment 2: paddr=0x0019f340 vaddr=0x3ffe88b4 size=0x0019c (   412) [0m
[0;32mI (720245) esp_image: segment 3: paddr=0x0019f4e4 vaddr=0x40100000 size=0x0696c ( 26988) [0m
[0;32mI (720275) ota: esp_ota_set_boot_partition succeeded[0m
[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!
[0m[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!
[0m[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...

[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!
[0m[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!
[0m[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!
[0m[1;33m[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[0m[1;33m[inf] iotx_mc_cycle(1570): receive ping response!

ljy770, 2019-11-07:
升级成功后的 log 和手动重启以后的 log 发一下看看. 类似 https://github.com/espressif/esp-aliyun/blob/master/examples/solutions/smart_light/README.md  中 12.OTA 支持

huozhe0930, 2019-11-07:
没有记录，明天再测试一个吧。

huozhe0930, 2019-11-22:
nothing

closed 2019-11-22 by huozhe0930