espressif/esp-aliyun #165: 遇到了经常重连一直失败的问题，E (503890) esp-tls: couldn't get hostname，E (503890) esp-tls: Failed to open new connection

wangqianjiao, 2019-11-19:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

您好，
我们使用ESP32的最新适配了阿里云3.0.1的源码去调试设备联网时，遇到了经常重连一直失败的问题，
打印的错误日志如下：
 
E (503890) esp-tls: couldn't get hostname for :a1Qix1VyXKU.iot-as-mqtt.cn-shanghai.aliyuncs.com:
E (503890) esp-tls: Failed to open new connection

### 2.1 复现步骤

1. 配网完成；
2. 查看串口日志，观察到IP地址已获取成功；
3. 等待连接阿里云，就过一段时间报一次上面的错误日志；
4. 重新通电自动重连又恢复正常；

### 2.2 复现问题的代码

无

## 3. 调试 Logs
11-18 20:38:45[admin:app] current_speed=0 
11-18 20:38:45[admin:app] current_speed=0 
11-18 20:38:45[admin:app] current_speed=0 
11-18 20:38:45I (94199) tcpip_adapter: sta ip: 192.168.0.46, mask: 255.255.255.0, gw: 192.168.0.1
11-18 20:38:45[admin:app] STA Got IP 
11-18 20:38:45[admin:app] got ip:192.168.0.46 
11-18 20:38:45[admin:app] current_speed=0 
11-18 20:38:46[prt] log level set as: [ 2 ]
11-18 20:38:46[iotkit:app] iotkit_cloud_start thread start
11-18 20:38:46W (94339) wifi: alloc eb len=24 type=3 fail, heap:4171248
11-18 20:38:46
11-18 20:38:46W (94340) wifi: m f null
11-18 20:38:46
11-18 20:38:50[admin:app] current_speed=0 
11-18 20:38:51W (99283) wifi: alloc eb len=232 type=2 fail, heap:4169792
11-18 20:38:51
11-18 20:38:51W (99283) wifi: m f probe rsp l=207
11-18 20:38:51[admin:app] current_speed=0 
11-18 20:38:51W (99824) wifi: alloc eb len=232 type=2 fail, heap:4169796
11-18 20:38:51
11-18 20:38:51W (99824) wifi: m f probe rsp l=207
11-18 20:38:51
11-18 20:38:51[admin:app] current_speed=0 
11-18 20:38:51[admin:app] current_speed=0 
11-18 20:38:51W (100079) wifi: alloc eb len=232 type=2 fail, heap:4169796
11-18 20:38:51
11-18 20:38:51W (100079) wifi: m f probe rsp l=207
11-18 20:38:51
11-18 20:38:54[admin:app] current_speed=0 
11-18 20:38:54W (103191) wifi: alloc eb len=232 type=2 fail, heap:4169432
11-18 20:38:54
11-18 20:38:54W (103192) wifi: m f probe rsp l=207
11-18 20:38:54
11-18 20:38:54[admin:app] read_diymode=0 
11-18 20:38:54[admin:app] diymode=3 
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55W (103450) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103450) wifi: m f probe rsp l=207
11-18 20:38:55
11-18 20:38:55W (103469) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103469) wifi: m f probe rsp l=207
11-18 20:38:55
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55W (103521) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103521) wifi: m f probe rsp l=207
11-18 20:38:55
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55[admin:app] current_speed=0 
11-18 20:38:55W (103870) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103871) wifi: m f probe rsp l=207
11-18 20:38:55
11-18 20:38:55W (103894) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103894) wifi: m f probe rsp l=207
11-18 20:38:55
11-18 20:38:55W (103914) wifi: alloc eb len=232 type=2 fail, heap:4169248
11-18 20:38:55
11-18 20:38:55W (103915) wifi: m f probe rsp l=207
11-18 20:38:59[admin:app] current_speed=0 
11-18 20:38:59E (108198) esp-tls: couldn't get hostname for :这里是ProductKey不便贴出.iot-as-mqtt.cn-shanghai.aliyuncs.com:
11-18 20:38:59E (108198) esp-tls: Failed to open new connection
11-18 20:38:59[admin:app] current_speed=0 
11-18 20:39:01W (109546) wifi: alloc eb len=232 type=2 fail, heap:4169472
11-18 20:39:01
11-18 20:39:01W (109546) wifi: m f probe rsp l=207
11-18 20:39:01
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01[admin:app] read_diymode=0 
11-18 20:39:01[admin:app] diymode=3 
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01W (109829) wifi: alloc eb len=232 type=2 fail, heap:4169472
11-18 20:39:01
11-18 20:39:01W (109829) wifi: m f probe rsp l=207
11-18 20:39:01
11-18 20:39:01W (109865) wifi: alloc eb len=232 type=2 fail, heap:4169472
11-18 20:39:01
11-18 20:39:01W (109865) wifi: m f probe rsp l=207
11-18 20:39:01
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01W (109891) wifi: alloc eb len=232 type=2 fail, heap:4169472
11-18 20:39:01
11-18 20:39:01W (109891) wifi: m f probe rsp l=207
11-18 20:39:01
11-18 20:39:01W (109921) wifi: alloc eb len=232 type=2 fail, heap:4169472
11-18 20:39:01
11-18 20:39:01W (109921) wifi: m f probe rsp l=207
11-18 20:39:01
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01[admin:app] current_speed=0 
11-18 20:39:01W (110202) iot_import_tls: Poll timeout or error, errno=Success, fd=-1, timeout_ms=2000
11-18 20:39:01[err] MQTTConnect(460): send connect packet failed
11-18 20:39:01[err] _mqtt_connect(786): send connect packet failed, rc = -14
11-18 20:39:01[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed[admin:app] current_speed=0 
11-18 20:39:02W (111057) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:02
11-18 20:39:02W (111057) wifi: m f probe rsp l=207
11-18 20:39:02
11-18 20:39:02W (111058) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:02
11-18 20:39:02W (111060) wifi: m f probe rsp l=207
11-18 20:39:02
11-18 20:39:02[admin:app] current_speed=0 
11-18 20:39:02W (111158) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:02
11-18 20:39:02W (111158) wifi: m f probe rsp l=207
11-18 20:39:02
11-18 20:39:02W (111158) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:02
11-18 20:39:02W (111161) wifi: m f probe rsp l=207
11-18 20:39:02
11-18 20:39:02[admin:app] read_diymode=0 
11-18 20:39:02[admin:app] diymode=3 
11-18 20:39:02[admin:app] current_speed=0 
11-18 20:39:02W (111264) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:02
11-18 20:39:02W (111264) wifi: m f probe rsp l=207
11-18 20:39:02
11-18 20:39:03W (111264) wifi: alloc eb len=232 type=2 fail, heap:4171796
11-18 20:39:03
11-18 20:39:03W (111267) wifi: m f probe rsp l=207
11-18 20:39:03
11-18 20:39:03[admin:app] current_speed=0 
11-18 20:39:30[admin:app] current_speed=0 
11-18 20:39:30E (139198) esp-tls: couldn't get hostname for :这里是ProductKey不便贴出.iot-as-mqtt.cn-shanghai.aliyuncs.com:
11-18 20:39:30E (139198) esp-tls: Failed to open new connection
11-18 20:39:30[admin:app] current_speed=0 
11-18 20:39:31[admin:app] current_speed=0 
11-18 20:39:31[admin:app] read_diymode=0 
11-18 20:40:01[admin:app] current_speed=0 
11-18 20:40:01[admin:app] current_speed=0 
11-18 20:40:01E (170198) esp-tls: couldn't get hostname for :这里是ProductKey不便贴出.iot-as-mqtt.cn-shanghai.aliyuncs.com:
11-18 20:40:01E (170198) esp-tls: Failed to open new connection
11-18 20:40:01[admin:app] current_speed=0 
11-18 20:40:03[admin:app] diymode=3 
11-18 20:40:03[admin:app] current_speed=0 
11-18 20:40:03W (172202) iot_import_tls: Poll timeout or error, errno=Success, fd=-1, timeout_ms=2000
11-18 20:40:03[err] MQTTConnect(460): send connect packet failed
11-18 20:40:03[err] _mqtt_connect(786): send connect packet failed, rc = -14
11-18 20:40:04[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
11-18 20:40:04[admin:app] current_speed=0 
11-18 20:40:04[admin:app] current_speed=0 
11-18 20:40:04[admin:app] current_speed=0 

## 4. 其他项

无

ljy770, 2019-11-19:
11-18 20:38:55W (103870) wifi: alloc eb len=232 type=2 fail, heap:4169248
这是哪里打印出来的?

wangqianjiao, 2019-11-19:
> 11-18 20:38:55W (103870) wifi: alloc eb len=232 type=2 fail, heap:4169248
> 这是哪里打印出来的?

您好，我们找不到这个是哪里打印的

wangqianjiao, 2019-11-19:
> > 11-18 20:38:55W (103870) wifi: alloc eb len=232 type=2 fail, heap:4169248
> > 这是哪里打印出来的?
> 
> 您好，我们找不到这个是哪里打印的

SDk4.0源码

ljy770, 2019-11-19:
您说的是 esp-idf  SDk 4.0 吗?  需要基于 esp-idf release/v3.2

wjj-chn, 2019-11-19:
为了给您更好的支持，请帮忙完善以下信息，谢谢！  	 	项目应用场景介绍   	ESP 芯片  	ESP 模组名称  	ESP SDK 版本号   	您参考的文档链接   	问题复现步骤  Log

wjj-chn, 2019-11-19:
为了给您更好的支持，请帮忙完善以下信息，谢谢！ 
	
项目应用场景介绍  
ESP 芯片 ESP 模组名称 
	硬件是否定制,PSRAM 大小
	esP-idf 版本号
	您参考的文档链接  
	问题复现步骤

448386711dong, 2019-11-19:
> esp-idf  SDk 4.0

是的，基于 esp-idf release/v3.2版本在热点配网出现问题，ep32厂商回复说使用esp-idf  SDk 4.0之后转到4.0版本热点配网就没这个问题

wangqianjiao, 2019-11-19:
> 为了给您更好的支持，请帮忙完善以下信息，谢谢！
> 
> 项目应用场景介绍
> ESP 芯片 ESP 模组名称
> 硬件是否定制,PSRAM 大小
> esP-idf 版本号
> 您参考的文档链接
> 问题复现步骤

您好，已更新内容

wjj-chn, 2019-11-19:
> > 为了给您更好的支持，请帮忙完善以下信息，谢谢！
> > 项目应用场景介绍
> > ESP 芯片 ESP 模组名称
> > 硬件是否定制,PSRAM 大小
> > esP-idf 版本号
> > 您参考的文档链接
> > 问题复现步骤
> 
> 您好，已更新内容

你好，请提供工程使用的 sdkconfig 到我司， 进行问题复现。代码应用层还有其它改动吗？

wangqianjiao, 2019-11-19:
> > > 为了给您更好的支持，请帮忙完善以下信息，谢谢！
> > > 项目应用场景介绍
> > > ESP 芯片 ESP 模组名称
> > > 硬件是否定制,PSRAM 大小
> > > esP-idf 版本号
> > > 您参考的文档链接
> > > 问题复现步骤
> > 
> > 
> > 您好，已更新内容
> 
> 你好，请提供工程使用的 sdkconfig 到我司， 进行问题复现。代码应用层还有其它改动吗？

请问是发送邮箱吗

wjj-chn, 2019-11-19:
> > > > 为了给您更好的支持，请帮忙完善以下信息，谢谢！
> > > > 项目应用场景介绍
> > > > ESP 芯片 ESP 模组名称
> > > > 硬件是否定制,PSRAM 大小
> > > > esP-idf 版本号
> > > > 您参考的文档链接
> > > > 问题复现步骤
> > > 
> > > 
> > > 您好，已更新内容
> > 
> > 
> > 你好，请提供工程使用的 sdkconfig 到我司， 进行问题复现。代码应用层还有其它改动吗？
> 
> 请问是发送邮箱吗

好的，wangjingjing@espressif.com

closed 2019-12-05 by wangqianjiao

RobsXiao, 2021-03-12:
最后结论是什么呢？我也是遇到这个问题