espressif/esp-aliyun #168: 这个esp-aliyun支持连接阿里云的国际站吗

hscxg, 2019-11-28:
经测试, 这个esp-aliyun连接阿里云的中国站没有问题,  由于我们的产品是需要连接阿里云的国际站, 但我试着连接国际站时, 没有成功.

请问一下,  这个esp-aliyun支持连接阿里云的国际站吗, 如果支持的话, 是否需要一些其它配置?

谢谢!

ljy770, 2019-11-28:
您好, 还没有试过连接国际站, 应该是支持的,可能需要一些特殊的配置. 
(1) 按照阿里云文档:https://g.alicdn.com/aic/ilop-docs/2018.10.35/international.html
需要在国际站创建产品.
(2) esp-aliyun/examples/solutions/smart_light/components/linkkit_handle/linkkit_solo.c :
domain_type = IOTX_CLOUD_REGION_SHANGHAI;
这里改一下地区.
试试看呢.

wanglemao, 2019-12-26:
@ljy770 里面还有一个关于 MQTT 直连和预认证的切换，但是我在esp-aliyun没有找到这个选项，但是官网说这个会导致设备无法上线，请问这个有影响的么？

ljy770, 2020-03-26:
阿里云选项的位置在 iotkit-embedded\infra\infra_config.h, MQTT 直连默认是支持的.

wanglemao, 2020-04-22:
我尝试了，连接美国是不行的，我咨询了阿里的技术支持，说要采用他们生活平台的sdk，否则出问题不管。。。

ljy770, 2020-04-22:
他们生活平台的sdk 是指 aliyun/iotkit-embedded 吗? 具体哪个版本支持呢?
esp-aliyun 是基于 aliyun/iotkit-embedded v3.0.1 版本开发的.

ljy770, 2020-05-14:
@wanglemao @hscxg 

国家版的功能已经添加,使用 https://github.com/espressif/esp-aliyun master 版本更新到最新, 操作说明在  esp-aliyun/examples/solutions/smart_light/README.md (14.国际站设备开发)

closed 2020-05-21 by Eavo