espressif/esp-aliyun #169: ESP8266 烧录不成功

hscxg, 2019-12-06:
我用的开发板为: ESP8266DevKitC-02D-F (2M Flash)

make chip=esp8266 defconfig 配置ESP8266芯片 

make flash 直接编译这个SDK包, 编译通过

但是在烧录芯片时, 却出现错误:
A fatal error occurred: Timed out waiting for packet header
[image]
不知怎么回事, 请指点, 谢谢!

hscxg, 2019-12-06:
[image]

ljy770, 2019-12-06:
检查一下 USB 端口有没有选择正确,
比如,对于 Ubuntu系统, 如果设备连接端口是 USB2
make flash ESPPORT=/dev/ttyUSB2

hscxg, 2019-12-06:
这个板子通过USB芯片转成了COM3口, 我是用COM3口烧录的,  串口号肯定没错,  可以通过COM3监视到这个板子输出的LOG信息

不知是哪里配置问题, 真是头痛!

ljy770, 2019-12-06:
参考 https://github.com/espressif/esptool/issues/226
在界面上出现
Connecting...字样时候按住boot按钮即可，等到出现烧录的时候就可以松开了

hscxg, 2019-12-06:
多谢ljy770!

按这种方法试了, 还是不行!

查看了一下这个项目, 编译完成后, 新生成了4个bin文件:

bootloader.bin, partitions_esp8266.bin, smart_light.bin, ota_data_initial.bin

如果用烧录工具: flash_download_tools_v3.6.7.exe 直接烧录这些bin文件,  是可以烧进去的, 

但我不知道这4个文件在Flash中烧写地址,  网上找了很长时间, 也找不到相关说明. 

不知哪里有?

ljy770, 2019-12-06:
编译结束最后的打印信息有, 看一下

hscxg, 2019-12-06:
KBS_CXG@KBS_CXG-PC MSYS ~
$ cd D:/ESP32/esp-aliyun3/examples/solutions/smart_light

KBS_CXG@KBS_CXG-PC MSYS /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make chip=esp8266 defconfig
D:/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
DEFCONFIG
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:764:warning: override: reassigning to symbol ESPTOOLPY_BAUD_921600B
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:764:warning: override: ESPTOOLPY_BAUD_921600B changes choice state
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:766:warning: override: reassigning to symbol ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:768:warning: override: reassigning to symbol PARTITION_TABLE_CUSTOM
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:768:warning: override: PARTITION_TABLE_CUSTOM changes choice state
/d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:769:warning: override: reassigning to symbol PARTITION_TABLE_CUSTOM_FILENAME
#
# configuration written to /d/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig
#

KBS_CXG@KBS_CXG-PC MSYS /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make flash
D:/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
GENCONFIG
App "smart_light" version: v2.0-20-gb2b5fb2-dirty
/d/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
LD build/bootloader/bootloader.elf
esptool.py v2.8-dev
Python requirements from D:/Temp/esp-idf/requirements.txt are satisfied.
Building partitions from /d/ESP32/esp-aliyun3/examples/solutions/smart_light/partitions_esp8266.csv...

App "smart_light" version: v2.0-20-gb2b5fb2-dirty
CC build/esp-aliyun3/conn_mgr/conn_mgr.o
CC build/esp-aliyun3/iotkit-embedded/certs/root_ca.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_packet.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_report.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_info.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/sha256.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_cmp_mqtt.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_cmp_coap.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/passwd.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_timer.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_notify.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_event.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_bind_statis.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_bind.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/os/os_misc.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_reset/awss_dev_reset.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_coap.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_msg_process.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_cota.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_ntp.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_server.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_gateway.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_localsetup.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_base64.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_client_adapter.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_fota.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_server_adapter.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_tsl_alink.o
[... 136 lines cut ...]
A fatal error occurred: Timed out waiting for packet header
make: *** [/d/Temp/esp-idf/components/esptool_py/Makefile.projbuild:67：flash] 错误 2

KBS_CXG@KBS_CXG-PC MSYS /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make flash
D:/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
/d/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
[... cut ...]
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from D:/Temp/esp-idf/requirements.txt are satisfied.

App "smart_light" version: v2.0-20-gb2b5fb2-dirty
Flashing binaries to serial port /COM3 (app at offset 0x10000)...
esptool.py v2.8-dev
Serial port E:/msys32/COM3
Connecting....

A fatal error occurred: Timed out waiting for packet header
make: *** [/d/Temp/esp-idf/components/esptool_py/Makefile.projbuild:67：flash] 错误 2

hscxg, 2019-12-06:
KBS_CXG@KBS_CXG-PC MSYS /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make flash
D:/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
GENCONFIG
App "smart_light" version: v2.0-20-gb2b5fb2-dirty
/d/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
LD build/bootloader/bootloader.elf
esptool.py v2.8-dev
Python requirements from D:/Temp/esp-idf/requirements.txt are satisfied.
Building partitions from /d/ESP32/esp-aliyun3/examples/solutions/smart_light/partitions_esp8266.csv...

App "smart_light" version: v2.0-20-gb2b5fb2-dirty
CC build/esp-aliyun3/conn_mgr/conn_mgr.o
CC build/esp-aliyun3/iotkit-embedded/certs/root_ca.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_packet.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_report.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_info.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/sha256.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_cmp_mqtt.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_cmp_coap.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/passwd.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_timer.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_notify.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_event.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_bind_statis.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_bind.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/os/os_misc.o
CC build/esp-aliyun3/iotkit-embedded/dev_bind/impl/awss_reset/awss_dev_reset.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_coap.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_msg_process.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_cota.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_ntp.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_server.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_gateway.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_localsetup.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_base64.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_client_adapter.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_fota.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_server_adapter.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_tsl_alink.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_server.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_ota.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/iotx_cm_coap.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/iotx_cm.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_ipc.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_message_cache.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_solo.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_opt.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_client.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_client.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_log_report.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_adapter.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_utils.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/iotx_cm_mqtt.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/impl_linkkit.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_shadow.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_api.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_api.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/alcs_mqtt.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_message.o
CC build/esp-aliyun3/iotkit-embedded/dev_model/dm_manager.o
CC build/esp-aliyun3/iotkit-embedded/dev_reset/dev_reset.o
[... 68 lines cut ...]
CC build/esp-aliyun3/wrappers/wrapper_kv.o
CC build/esp-aliyun3/wrappers/wrapper_tls.o
CC build/esp-aliyun3/wrappers/wrapper_ota.o
CC build/esp-aliyun3/wrappers/wrapper_product.o
CC build/esp-aliyun3/wrappers/wrapper_system.o
CC build/esp-aliyun3/wrappers/wrapper_tcp.o
CC build/esp-aliyun3/wrappers/wrapper_os.o
D:/ESP32/esp-aliyun3/wrappers/wrapper_os.c:36:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "os";
                    ^
CC build/esp-aliyun3/wrappers/wrapper_crypto.o
AR build/esp-aliyun3/libesp-aliyun3.a
AR build/esp32/libesp32.a
Generating libesp-aliyun3.a.sections_info
Generating libesp32.a.sections_info
Generating esp32.project.ld
LD build/smart_light.elf
esptool.py v2.8-dev
Flashing binaries to serial port /COM3 (app at offset 0x10000)...
esptool.py v2.8-dev
Serial port E:/msys32/COM3
Connecting....

A fatal error occurred: Timed out waiting for packet header
make: *** [/d/Temp/esp-idf/components/esptool_py/Makefile.projbuild:67：flash] 错误 2

ljy770, 2019-12-06:
直接执行 make

hscxg, 2019-12-06:
KBS_CXG@KBS_CXG-PC MSYS /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make
D:/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
/d/Temp/esp-idf/make/project.mk:63: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from D:/Temp/esp-idf/requirements.txt are satisfied.

App "smart_light" version: v2.0-20-gb2b5fb2-dirty
To flash all build output, run 'make flash' or:
python /d/Temp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/ota_data_initial.bin 0x1000 /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/bootloader/bootloader.bin 0x10000 /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/smart_light.bin 0x8000 /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/partitions_esp8266.bin

ljy770, 2019-12-06:
**0xd000** /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/ota_data_initial.bin 
**0x1000** /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/bootloader/bootloader.bin **0x10000** /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/smart_light.bin 
**0x8000** /d/ESP32/esp-aliyun3/examples/solutions/smart_light/build/partitions_esp8266.bin
烧写地址和对应的 bin 文件

ljy770, 2019-12-06:
> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
toolchain 没有配置正确,这是 esp32 的

8266 配置 toolchain:
https://docs.espressif.com/projects/esp8266-rtos-sdk/en/release-v3.1/get-started/linux-setup.html

hscxg, 2019-12-06:
哦, 我还以为这工具链, ESP32的兼容ESP8266的呢
那我再研究研究, 
谢谢!

hscxg, 2019-12-07:
我已经按上面给出的网页配置了ESP8266工具链, 编译前也配置了8266芯片: make chip=esp8266 defconfig
怎么编译信息还是: --chip esp32 

编译出的程序烧到8266芯片后, 也无法正常运行, 一直报Fatal exception信息, 无法正常运行.
是不是还是哪里配置不对?

[image]

ljy770, 2019-12-09:
IDF_PATH 有没有选择您下载的 ESP8266_RTOS_SDK 路径?
有没有 make clean?
单独编译下载 ESP8266_RTOS_SDK 的某个 example 是否成功?
不然再对照https://github.com/espressif/esp-aliyun/blob/master/README.md 有没有遗漏的?

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo