espressif/esp-aliyun #170: 断网重连会使设备内存减少，最终永久丢失内存导致设备无法正常运行

wangqianjiao, 2019-12-11:
## 1. 开发环境

项目应用场景介绍：空气净化器家电配网连云
ESP 芯片 ESP 模组名称：ESP32-WROVER-IB
硬件是否定制,PSRAM 大小：PSRAM 8M flash 也是8M
esP-idf 版本号：
```
sdk4.0到 commit ID 99fb9a3f7c28c5fa12b1bd4aa6fb7b622d841326
```
aliyun版本号：
```
esp-aliyun release v2.0
@wujiangang wujiangang released this on 16 Aug · 42 commits to master since this release

重要更新
支持阿里云 iotkit-embedded v3.0.1
支持阿里云 生活物联网平台 接入和阿里云 物联网平台 接入
新增三元组批量生产配置文件 & 使用说明
支持 esp32 和 esp8266
```
参考文档：
```
参考了这个源码Demo ：https://github.com/espressif/esp-aliyun/releases
下的：esp-aliyun/examples/mqtt/mqtt_example
```

## 2. 问题描述

您好，我们发现每次设备断网重连都会导致设备内存减少，当断网重连次数到一定阶段，内存减少到无法支持重连的阶段时，内存将永久无法恢复了。
PS：虽然重连成功后，1小时内不掉线内存可以恢复，比如：机器距离路由器正好信号不稳定，设备就会处于“掉线”和“上线”频繁切换，最终导致内存永久无法恢复，进入假死状态（没有资源进行任何动作了）

### 2.1 复现步骤

1.设备连上云端；
2.手动拔掉网线，路由不断电；
3.等设备打印出云端重连日志几次后，再插上网线。
4.如此不断反复拔插网线内存就会很明显减少了。

### 2.2 复现问题的代码

无

## 3. 调试 Logs
无
## 4. 其他项

无

ljy770, 2019-12-11:
您可以先用 master 最新版本试一下是否问题还存在?

wangqianjiao, 2019-12-12:
> 您可以先用 master 最新版本试一下是否问题还存在? 如依然存在问题，请反馈。

您好，请问是esp-aliyun还是esp-idf呢，master最新版要怎么更新呢有commit ID吗

ljy770, 2019-12-12:
esp-aliyun, 如果是 git 下载, git clone https://github.com/espressif/esp-aliyun.git

wangqianjiao, 2019-12-12:
> esp-aliyun, 如果是 git 下载,
> git clone https://github.com/espressif/esp-aliyun.git
> git checkout master
> git pull

这个代码之前就用过了啊，昨天有更新代码吗，我们重新下载一次再试试啊

ljy770, 2019-12-12:
@wangqianjiao 能否提供一下每次重连内存大概减少多少? 重连多少次内存耗尽?

wangqianjiao, 2019-12-12:
> @wangqianjiao 能否提供一下每次重连内存大概减少多少? 重连多少次内存耗尽?

您好，重新下载了下，和现在正在用的代码是一样的，每次重连大概掉约：2-3kb，日志稍等再抓一次发您啊

ljy770, 2019-12-12:
您使用的 esP-idf 版本号：v4.0, 这里有特殊需要一定要用 4.0 吗? 
我们测试比较稳定的版本 v3.2 v3.3.

wangqianjiao, 2019-12-12:
> 您使用的 esP-idf 版本号：v4.0, 这里有特殊需要一定要用 4.0 吗?
> 我们测试比较稳定的版本 v3.2 v3.3.
之前没发现这个问题，更新了 IDF 前后都会出现。最早使用情况：aliyun 和IDF v3.2.3一起使用，之后热点配网出问题了邮件问的热点配网使用安卓手机配不上，你们邮件回复IDF 切换到v4.0版。我这边切换到IDF v4.0后热点配网没问题了，经过测试之后才发现内存溢出问题

wangqianjiao, 2019-12-12:
> > 您使用的 esP-idf 版本号：v4.0, 这里有特殊需要一定要用 4.0 吗?
> > 我们测试比较稳定的版本 v3.2 v3.3.
> > 之前没发现这个问题，更新了 IDF 前后都会出现。最早使用情况：aliyun 和IDF v3.2.3一起使用，之后热点配网出问题了邮件问的热点配网使用安卓手机配不上，你们邮件回复IDF 切换到v4.0版。我这边切换到IDF v4.0后热点配网没问题了，经过测试之后才发现内存溢出问题

PS：部分安卓手机（比如：小米）发现的

ljy770, 2019-12-12:
这边会在 v4.0 上复现试试.
顺便问下, 热点配网是设备热点还是手机热点呢?

wangqianjiao, 2019-12-12:
> 这边会在 v4.0 上复现试试.
> 顺便问下, 热点配网是设备热点还是手机热点呢?

您好，是设备产生热点，手机去连，小米5S连的时候会莫名其妙多发一个请求导致热点不可用了

wangqianjiao, 2019-12-12:
> > 这边会在 v4.0 上复现试试.
> > 顺便问下, 热点配网是设备热点还是手机热点呢?
> 
> 您好，是设备产生热点，手机去连，小米5S连的时候会莫名其妙多发一个请求导致热点不可用了

D (20507) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (20513) httpd_txrx: httpd_recv_with_opt: pending length = 112
D (20519) httpd_parse: read_block: received HTTP request block size = 112
D (20526) httpd_parse: continue_parsing: skip pre-parsed data of size = 11
D (20533) httpd_parse: continue_parsing: un-paused
D (20538) httpd_parse: cb_header_value: processing value = Dalvik/2.1.0 (Linux; U; Android 8.0.0; MI 5s MIUI/V10.2.1.0.OAGCNXM)
D (20549) httpd_parse: cb_header_field: processing field = Host
D (20555) httpd_parse: cb_header_value: processing value = 192.168.4.1
D (20562) httpd_parse: cb_header_field: processing field = Connection
D (20568) httpd_parse: parse_block: parsed block size = 112
D (20574) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (20580) httpd_txrx: httpd_recv_with_opt: received length = 38
D (20586) httpd_parse: read_block: received HTTP request block size = 38
D (20593) httpd_parse: cb_header_value: processing value = Keep-Alive
D (20599) httpd_parse: cb_header_field: processing field = Accept-Encoding
D (20606) httpd_parse: cb_header_value: processing value = gzip
D (20612) httpd_parse: cb_headers_complete: bytes read     = 166
D (20618) httpd_parse: cb_headers_complete: content length = 0
D (20624) httpd_parse: pause_parsing: paused
D (20629) httpd_parse: cb_no_body: message complete
D (20633) httpd_parse: httpd_parse_req: parsing complete
D (20639) httpd_uri: httpd_uri: request for / with type 1
D (20644) httpd_uri: httpd_find_uri_handler2: [0] = /proto-ver
D (20650) httpd_uri: httpd_find_uri_handler2: [1] = /prov-session
D (20656) httpd_uri: httpd_find_uri_handler2: [2] = /prov-config
W (20662) httpd_uri: httpd_uri: URI '/' not found
W (20668) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist
D (20677) httpd_txrx: httpd_send_all: sent = 69
D (20681) httpd_txrx: httpd_send_all: sent = 2
D (20685) httpd_txrx: httpd_send_all: sent = 22
D (20690) httpd_sess: httpd_sess_process: httpd_req_delete
D (20695) httpd_sess: httpd_sess_process: success

wangqianjiao, 2019-12-12:
W (20662) httpd_uri: httpd_uri: URI '/' not found
W (20668) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist
之后，设备热点不管发什么请求都没反应了

lhespress, 2019-12-18:
如沟通，切换到 V3.3.1 版本之后，上述问题还存在吗？

wangqianjiao, 2019-12-20:
> 如沟通，切换到 V3.3.1 版本之后，上述问题还存在吗？

切换到V3.3.1没有这个问题了，现在遇到新问题，先关闭这个issue

closed 2019-12-20 by wangqianjiao