espressif/esp-aliyun #171: esp-aliyun配置esp8266 芯片出错

hscxg, 2019-12-11:
msys32下, 已经将工具链设置为esp8266的工具链: xtensa-lx106-elf-gcc
这个工具链可以成功编译 ESP8266_RTOS_SDK-3.2 里面的项目程序.

但是在编译esp-aliyun这个项目时, 先需要将芯片配置成esp8266芯片, 但在配置时出错, 错误信息如下:

KBS_CXG@KBS_CXG-PC MINGW32 /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make chip=esp8266 defconfig
make：xtensa-esp32-elf-gcc：命令未找到
expr: 语法错误
/bin/sh: xtensa-esp32-elf-gcc: 未找到命令
which: no xtensa-esp32-elf-gcc in (/mingw32/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/opt/xtensa-lx106-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)
/bin/sh: xtensa-esp32-elf-gcc: 未找到命令
Toolchain path:
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from D:/Tool/msys32/home/KBS_CXG/ESP8266_RTOS_SDK/requirements.txt are satisfied.
DEFCONFIG
D:/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:418 CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT was replaced with CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT
GENCONFIG
make：xtensa-esp32-elf-gcc：命令未找到
expr: 语法错误

hscxg, 2019-12-11:
工具链配置文件(esp32_toolchain.sh)内容如下:
export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
export IDF_PATH="D:/Temp/esp-idf"

不知配置是否对不对?

ljy770, 2019-12-11:
> 工具链配置文件(esp32_toolchain.sh)内容如下:
> export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
> export IDF_PATH="D:/Temp/esp-idf"
> 
> 不知配置是否对不对?
IDF_PATH 这里 esp-idf 是 ESP32 的, 应该指定 ESP8266_RTOS_SDK

hscxg, 2019-12-11:
嗯, 工具链配置文件(esp32_toolchain.sh)内容我改成如下:
export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
export IDF_PATH="D:/Tool/msys32/home/KBS_CXG/ESP8266_RTOS_SDK"

在配置esp8266 芯片时, 怎么还是出现错误:
KBS_CXG@KBS_CXG-PC MINGW32 /d/ESP32/esp-aliyun3/examples/solutions/smart_light
$ make chip=esp8266 defconfig
make：xtensa-esp32-elf-gcc：命令未找到
expr: 语法错误
/bin/sh: xtensa-esp32-elf-gcc: 未找到命令
which: no xtensa-esp32-elf-gcc in (/mingw32/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/opt/xtensa-lx106-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)
/bin/sh: xtensa-esp32-elf-gcc: 未找到命令
Toolchain path:
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from D:/Tool/msys32/home/KBS_CXG/ESP8266_RTOS_SDK/requirements.txt are satisfied.
DEFCONFIG
D:/ESP32/esp-aliyun3/examples/solutions/smart_light/sdkconfig:10 CONFIG_TOOLPREFIX was replaced with CONFIG_SDK_TOOLPREFIX

明明配置的是: /opt/xtensa-lx106-elf/bin
错误信息中怎么还是有esp32
make：xtensa-esp32-elf-gcc：命令未找到

ljy770, 2019-12-12:
如果配置没有错, 建议 make clean,或者下载 一份干净的 esp-aliyun 试一下

27-301, 2019-12-17:
把工程目录下的 sdkconfig 和 build 目录删除，重试。是否之前配置过 esp32？

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo