espressif/esp-aliyun #172: ESP8266进入配网模式后, 扫描不到AP站点信号?

hscxg, 2019-12-12:
我的这个产品准备用ESP8266来连接阿里云, 由于esp-aliyun这个SDK包我始终没能配置好, ESP8266开发板上无法正常跑程序.  因时间关系, 暂时只好放弃, 待以后有时间时再研究.

所以就采用阿里云的Alios-Things SDK包, 编译里面的linkkitapp@esp8266项目, 程序可以正常启动, 但扫描不到周边的AP站点信号, 按下GPIO14按键, 可以进入配网模式, 但扫描不到周边的AP站点信号. 

LOG信息如下:

[012660]<A> awss config press 2
[012660]<A> do_awss_active 0
[012660]<F> enable awss
[012660]<A> IOTX_AWSS_ENABLE
[012850]<F> chan 6
[013120]<F> chan 7
[013390]<F> chan 8
[013660]<F> chan 9
[013930]<F> chan 10
[014200]<F> chan 11
[014470]<F> chan 12
[014740]<F> chan 13
[015010]<F> chan 1
[015280]<F> chan 6
[015550]<F> chan 11
[015820]<F> chan 1
[016090]<F> chan 2
[016360]<F> chan 3
[016630]<F> chan 4
[016900]<F> chan 5
[017170]<F> chan 6
[017440]<F> chan 7
......

上面扫描的各个信道中, 没有一个AP站点信息, 不正常.

是不是代码中哪里要更改WIFI模式配置啊, 请指点, 谢谢!

FireWinter, 2019-12-12:
不需要改wifi模式配置，你可以试一下在linkkit_example_solo.c中增加这一行来启动awss配网，然后确认一下debug log是否配置，如下图：
[image]

hscxg, 2019-12-12:
按上面方式改为直接启动配网模式也是一样,  还是扫不到周边AP站点信息.
LOG级别和上图一样, 也是: AOS_LL_DEBUG.

这个项目我编译成ESP32程序, 烧到ESP32开发板上,  可以扫到很多周边的AP站点,

真是奇怪.

hscxg, 2019-12-12:
我用的是安信可的ESP-12F模块, WIFI硬件线路没有问题, 我用ESP8266_RTOS_SDK参考网上代码, 写了一个扫描周边WIFI热点程序, 可以扫描到周边的WIFI热点, 也可以连上我上网用的路由器.

hscxg, 2019-12-12:
对于阿里云的Alios-Things SDK包, 我仔细对比了ESP32和ESP8266的启动LOG信息, 我发现ESP32有输出一条: ic_enable_sniffer 信息, 而ESP8266没有输出这条信息. 

是不是sniffer没有开启, 造成了ESP8266扫不到周边AP站点?

hscxg, 2019-12-13:
请问一下, 这个阿里云的Alios-Things SDK - ESP8266 程序, 能运行在2M Flash上面吗, 我手上还有一块2M Flash的ESP8266板子,  我想在这块板子上试试.
如果仅更改烧录地址, 将程序烧到芯片上去, 程序跑不起来. 
是不是哪里要更改编译配置?

lhespress, 2019-12-18:
esp-aliyun SDK 运行在 8266 的配置信息见 ```examples/solutions/smart_light/sdkconfig_esp8266.defaults```,
烧录信息见 ```examples/solutions/smart_light/partitions_esp8266.csv```。
建议按照 README 中介绍的步骤操作。

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo