espressif/esp-aliyun #173: 组件调试不成功

AToygun, 2019-12-17:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：a1c2df6
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：3d394bd9
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

公版软件配网不成功

### 2.1 复现步骤

按照WIKI步骤来smart_light  编译

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs
I (43) boot: ESP-IDF v3.2-301-g3d394bd9 2nd stage bootloader
I (44) boot: compile time 23:41:15
I (44) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DOUT
I (56) boot: SPI Flash Size : 2MB
I (62) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (91) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (102) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (125) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (137) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (149) boot: End of partition table
I (155) boot: No factory image, trying OTA 0
I (163) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x8e2c4 (582340) map
0x40210010: _stext at ??:?

I (431) esp_image: segment 1: paddr=0x0009e2dc vaddr=0x4029e2d4 size=0x18478 ( 99448) map
I (476) esp_image: segment 2: paddr=0x000b675c vaddr=0x3ffe8000 size=0x008b0 (  2224) load
I (477) esp_image: segment 3: paddr=0x000b7014 vaddr=0x40100000 size=0x00a50 (  2640) load
I (487) esp_image: segment 4: paddr=0x000b7a6c vaddr=0x40100a50 size=0x059c8 ( 22984) load
I (510) boot: Loaded app from partition at offset 0x10000
I (534) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (541) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1155.0, 6cb3053, Nov 11 2019, 17:31:08, RTOS new
I (595) phy_init: phy ver: 1155_0
I (598) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
W (604) wrapper_kv: nvs get blob q_rt failed with 1102
I (614) factory_rst: quick reboot times 1, don't need to restore
I (622) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (636) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (650) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (664) pwm: --- PWM v3.2

[prt] log level set as: [ 4 ]
W (696) wrapper_kv: nvs get blob ssid failed with 1102
[crt] awss_config_press(180): enable awss
I (702) app main: IOTX_AWSS_ENABLE
I (705) app main: IOTX_AWSS_START
[inf] zconfig_init(260): zconfig_init
I (736) awss: Open monitor mode
[crt] awss_save_apinfo(271): [1] ssid:Akan, mac:480eec7e9748, chn:6, rssi:-66, adha:0
[crt] aws_switch_channel(166): chan 1
[crt] awss_save_apinfo(271): [2] ssid:polygon, mac:94d9b3f42847, chn:1, rssi:-29, adha:0
[crt] awss_save_apinfo(271): [3] ssid:ChinaNet, mac:f483cd3ec9c0, chn:11, rssi:-74, adha:0
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 4
[crt] awss_save_apinfo(271): [4] ssid:Akan, mac:5475957da049, chn:6, rssi:-80, adha:0
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 7
[crt] aws_switch_channel(166): chan 9

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

AToygun, 2019-12-17:
我现在用的node mcu，想请教下4MB那个csv文件应该怎么配置呢

27-301, 2019-12-17:
可以使用 examples/solutions/smart_light/partitions_esp32.csv

AToygun, 2019-12-18:
我这个问题是csv文件没有烧录进去吧，我都是按照官方那个单bin文件操作的。您觉得可能是什么原因呢

xuhongv, 2019-12-26:
@AToygun 看看我写的文章，你就懂了：https://blog.csdn.net/xh870189248/article/details/101448052

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo