espressif/esp-aliyun #177: 国际版-是否实现了region切换功能

wanglemao, 2020-01-20:
请问，当前的版本是否支持region切换功能，我在阿里官网支持上看到，如果要支持国际版的，首先必须支持region切换，我自己试了下，在新加坡可以正常使用，但是在美国就无法使用，是不是底层没有实现region切换导致的，同时MQTT是否是直连模式，这个能过GDPR么？

ljy770, 2020-03-26:
请问您的问题有解决吗?
这是 阿里云关于国际站的说明: https://living.aliyun.com/doc#international.html, 您看一下,里面有提到 国际版和 GDPR.
请问下 region 切换说明您在那里看到的, 方便提供一下连接吗?

ljy770, 2020-05-14:
@wanglemao 国家版的功能已经添加,使用 https://github.com/espressif/esp-aliyun master 版本更新到最新, 操作说明在  esp-aliyun/examples/solutions/smart_light/README.md (14.国际站设备开发)

ljy770, 2020-05-19:
@wanglemao esp-alilyun 已经添加region自动切换功能:自动重启切换到新的服务器.

wanglemao, 2020-05-21:
@ljy770 感谢支持，我一会升级试试，刚好我们有样品在华盛顿，他可以帮我验证。不过我之前咨询过阿里技术人员，他的意思好像是我们的固件不支持GDPR，所以才失败的

wanglemao, 2020-05-21:
@ljy770 刚才看了官方升级公告，说支持全球统一码，即不区分海外或中国，同时他们的sdk会发布新版，当前这个版本会支持么，还是说你们还要更新

ljy770, 2020-05-21:
@wanglemao 麻烦之后把验证结果反馈一下哦.
在哪里看到的官方升级公告? 提供一下官方公告链接,我来看下.

Eavo, 2020-05-21:
如果新版版验证有任何问题，可随时 reopen issue。谢谢。

closed 2020-05-21 by Eavo

wanglemao, 2020-05-26:
@ljy770 今天尝试的时候，才发现我用的版本不是master上面的，用的是feature/update_3.0.1_for_mesh，我用的是esp-mdf，这个应该还没有更新吧，我试着合并了分支，然后发现各种异常。。。能帮忙合并下么

ljy770, 2020-05-27:
@wanglemao 请问使用 for_mesh 版本是因为对 mesh 有需求是吗? 如果没有,建议用 master 版本.esp-mdf 也是使用的 早的版本吧? 最新的版本已经不依赖 esp-aliyun 了.

如果还是使用 feature/update_3.0.1_for_mesh 版本,可以将 master 版本上 2020.5.13~5.19 的提交 merge 或者 cherry-pick 到当前的分支试试看.