espressif/esp-aliyun #178: 天猫精灵四元组不能用量产工具生成批量BIN文件的问题

wangyueyang2, 2020-02-04:
1,我是用ESP8266_RTOS_SDK 3.0这个SDK,用ESP-IDF编译的,其它编译烧写都没问题的,调试时用以下命令来生成四元组也没问题.($IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input my_single_mfg_config.csv --output my_single_mfg.bin --size 0x4000)
2,现在调试完毕,我想用量产工具来生成批量的BIN文件,根据文档命令如下:
$IDF_PATH/tools/mass_mfg/mfg_gen.py --conf multipule_mfg_config.csv --values my_multipule_mfg_values.csv --prefix Fan --size 0x4000 --version v1
出现错误,调试发现SDK的tools下压根就没有mass_mfg文件夹,在你们的网站上看了好几个SDK版本,也找不到这个文件,请问我能在哪找到这个mass_mfg/mfg_gen.py文件,支持以上命令实现量产,谢谢

## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

// 请提供详细的问题描述。

### 2.1 复现步骤

1. ...
2. ...

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs

// 由于 log 可能比较长，请通过附件的形式提供，否则该 issue 页面会非常长，不便于其他开发者查阅。

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

simonliu009, 2020-02-06:
试试这个？
https://github.com/espressif/esp-idf/blob/master/tools/mass_mfg/mfg_gen.py

wangyueyang2, 2020-02-06:
你好：
      我试过了，还是不行，日志如下
admin@DESKTOP-090E0J7 MINGW32 ~/mass_mfg
$ ls -l
总用量 28
drwxr-xr-x 1 admin None     0  2月  6 20:01 docs
-rwxr-xr-x 1 admin None 20493  2月  5 02:09 mfg_gen.py
drwxr-xr-x 1 admin None     0  2月  6 20:04 samples
drwxr-xr-x 1 admin None     0  2月  6 20:01 testdata

admin@DESKTOP-090E0J7 MINGW32 ~/mass_mfg
$ python mfg_gen.py generate samples/sample_config.csv samples/sample_values_singlepage_blob.csv Sample 0x3000

Created CSV file: ===&gt; D:/tianmaoConnect/toolAndSdk/msys32/home/admin/mass_mfg/csv/Sample-1.csv
'module' object has no attribute 'generate'

admin@DESKTOP-090E0J7 MINGW32 ~/mass_mfg

帮我看一下，是怎以回事，这个是编译IDF自带的例程的，我没有改任何东西。
       2020。2。6

 https://github.com/espressif/esp-idf/blob/master/tools/mass_mfg/mfg_gen.py

simonliu009, 2020-02-08:
那我也不知道了，如果是商用的话，找FAE支持一下吧

ljy770, 2020-02-18:
https://github.com/espressif/esp-idf/tree/release/v3.2/tools/mass_mfg/mfg_gen.py
用这版本的试一下

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo