espressif/esp-aliyun #180: 8266在smart_light上启用一型一密动态注册功能

tomssy, 2020-02-10:
- 模组或开发板：
- esp-aliyun 版本 master
- ESP8266_RTOS_SDK 版本 3.2

## 2. 问题描述

已经成功跑通smart_light示例，尝试在移植一型一密功能时遇到问题，
1.已经在linkiit_solo中令dynamic_register = 1;
    IOT_Ioctl(IOTX_IOCTL_SET_DYNAMIC_REGISTER, (void *)&dynamic_register);
2. 在makefile中增加了宏定义CFLAGS += -DDYNAMIC_REGISTER
根据输出的log代码，代码已经跑进了动态注册流程，但是在http建立连接的阶段失败。
LOG如下：
[1;33m[inf] _iotx_dynamic_register(310): DeviceSecret KV not exist, Now We Need Dynamic Register...
[0m[1;33m[inf] _calc_dynreg_sign(127): Random Key: 8Ygb7ULYh53B6OA
[0m[1;33m[inf] _calc_dynreg_sign(143): Sign: EB0BCC6B4D8256F88E3D70B2544591EB9C917275395F43001CE7AB80F1ABDE58
[0m[1;31m[err] iotx_net_tcp_connect(455): no method match!
[0m[1;31m[err] iotx_net_tcp_disconnect(441): no method match!
[0m[1;31m[err] httpclient_connect(421): TCP or TLS connect failed, rc = -1
[0m[1;31m[err] iotx_net_tcp_connect(455): no method match!
[0m[1;31m[err] iotx_net_tcp_disconnect(441): no method match!
[0m[1;31m[err] httpclient_connect(421): TCP or TLS connect failed, rc = -1
[0m[1;31m[err] iotx_net_tcp_connect(455): no method match!
[0m[1;31m[err] iotx_net_tcp_disconnect(441): no method match!
[0m[1;31m[err] httpclient_connect(421): TCP or TLS connect failed, rc = -1
[0m[1;31m[err] _http_send(537): _http_send_request is error, ret = -1
[0m[1;33m[inf] httpclient_close(501): client disconnected
[0m[1;31m[err] _fetch_dynreg_http_resp(195): Http Download Failed
[0m[1;31m[err] IOT_Dynamic_Register(275): Get Device Secret Failed
[0m[1;31m[err] _iotx_dynamic_register(314): Dynamic Register Failed
[0m[1;31m[err] IOT_MQTT_Construct(413): ret = _iotx_dynamic_register() = -1, abort

已检查support tcp与support tls均开启，是由于什么原因造成的呢，是否是遗漏了什么宏或者功能？

ljy770, 2020-02-18:
CONFIG_SUPPORT_TCP 先关闭这个设置试试，只打开　#define SUPPORT_TLS
（即确认看下这里动态加密是否要用　https　才行）

tomssy, 2020-02-18:
> CONFIG_SUPPORT_TCP 先关闭这个设置试试，只打开　#define SUPPORT_TLS
> （即确认看下这里动态加密是否要用　https　才行）

已经确认关闭tcp可以正常工作了。也就是目前不支持https下的动态注册？

ljy770, 2020-02-20:
应该说是不支持 http 下的动态注册. 猜测如果是 HTTP 下注册信息没有加密会不安全, 所以不支持.
说明一下:
1 . 8266 默认打开 tcp 这个设置是专门为 8266 OTA 时候设置的, 因为  OTA 时候两条 TLS 连接( https 和 MQTT) 会内存不够导致升级失败,所以把 https 这条连接改成了 http , mqtt 还是基于 tls.
2 . 关闭tcp, 就是 mqtt 和 http 都是基于 tls, ota 会受到影响.

tomssy, 2020-02-23:
> 应该说是不支持 http 下的动态注册. 猜测如果是 HTTP 下注册信息没有加密会不安全, 所以不支持.
> 说明一下:
> 1 . 8266 默认打开 tcp 这个设置是专门为 8266 OTA 时候设置的, 因为 OTA 时候两条 TLS 连接( https 和 MQTT) 会内存不够导致升级失败,所以把 https 这条连接改成了 http , mqtt 还是基于 tls.
> 2 . 关闭tcp, 就是 mqtt 和 http 都是基于 tls, ota 会受到影响.

了解了，谢谢

closed 2020-02-23 by tomssy