espressif/esp-aliyun #181: ESP8266 aliyun  官方ota样例会造成固件损坏，无法启动

temp4gh, 2020-02-16:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：f5438e1和fed581b（2.0 release）都会出现
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：3ad3fd7（3.2 release）

## 2. 问题描述

使用官方ota样例ota_example_mqtt，会出现ota失败后设备无法正常启动的问题

### 2.1 复现步骤

1. 循环使用生成的bin文件进行阿里云物联网套件云端ota升级
2.使用 tls方式ota，很容易出现，大概10次以内会出现，表现为直接失败，自动重启后就无法继续运行
3.使用tcp方式ota，出现几率小些，大约30次以内必然出现，出现：
[inf] _http_recv(214): ret of _http_recv is 0
后不要人工干预，让设备一直运行，多等待一段时间就会出现

### 2.2 复现问题的代码

官方ota样例 ：ota_example_mqtt

## 3. 调试 Logs

----========正常ota的log：========-
tail 12
chksum 0x57
ho 0 tail 12 room 4
load 0x3ffe8420, len 3524, room 12
tail 8
chksum 0xfa
I (50) boot: ESP-IDF  2nd stage bootloader
I (50) boot: compile time 20:56:52
I (50) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DOUT
I (60) boot: SPI Flash Size : 2MB
I (66) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (94) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (106) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (117) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (129) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (141) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (152) boot: End of partition table
I (159) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7a04c (49
I (391) esp_image: segment 1: paddr=0x0008a064 vaddr=0x3ffe8000 size=0x007f4 (
I (393) esp_image: segment 2: paddr=0x0008a860 vaddr=0x3ffe87f4 size=0x0019c (
I (403) esp_image: segment 3: paddr=0x0008aa04 vaddr=0x40100000 size=0x064b4 ( 2
0x40100000: _stext at ??:?

I (427) boot: Loaded app from partition at offset 0x10000
I (543) system_api: Base MAC address is not set, read default base MAC address f
I (553) system_api: Base MAC address is not set, read default base MAC address f
I (753) phy_init: phy ver: 1055_12
I (753) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2
[prt] log level set as: [ 4 ]
I (833) conn_mgr: Found ssid TP-LINK_MY
I (3203) event: sta ip: 192.168.1.100, mask: 255.255.255.0, gw: 192.168.1.1
[wrn] IOT_MQTT_Construct(467): Using default hostname: 'xxxxxxxxxxxx.iot-as-mqtt.
m'
[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[wrn] IOT_MQTT_Construct(481): Using default client_id: xxxxxxxxxxxx.
608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[wrn] IOT_MQTT_Construct(488): Using default username: xxxxxxxxxxxxXXXXXXXXXXXXX
[wrn] IOT_MQTT_Construct(496): Using default password: ******
[inf] iotx_mc_init(230): MQTT init success!
[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=xxxxxxxxxxxx.zc
stamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1
, username=xxxxxxxxxxxxXXXXXXXXXXXXX&xxxxxxxxxxxx
[... 1536 lines cut ...]
I (55) boot: SPI Mode       : DOUT
I (61) boot: SPI Flash Size : 2MB
I (67) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (96) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (107) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (119) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (130) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (142) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (154) boot: End of partition table
E (160) esp_image: image at 0x110000 has invalid magic byte
E (170) boot: OTA app partition slot 1 is not bootable
I (179) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x7a04c (49
9788) map
E (411) esp_image: unaligned segment length 0x99008122
E (411) boot: OTA app partition slot 0 is not bootable
E (413) boot: No bootable app partitions in the partition table
user code done

// 由于 log 可能比较长，请通过附件的形式提供，否则该 issue 页面会非常长，不便于其他开发者查阅。

## 4. 其他项

该问题较严重，会造成设备变砖无法启动也无法回退到ota前固件，请帮忙尽快确认解决修复，谢谢！

ljy770, 2020-02-18:
[err] IOT_OTA_FetchYield(698): Fetch firmware failed

从 log 上看不出为什么会接着进入 reboot, 这中间是否还有其他 log？从代码上看应该要检查固件有效才会进入 reboot,
ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16
0x40100000: _stext at ??:?

ljy770, 2020-02-18:
examples/ota/ota_example_mqtt/components/ota_handle/ota_solo.c
` 
               } else {
                    IOT_OTA_ReportProgress(h_ota, IOT_OTAP_FETCH_FAILED, NULL);
                    EXAMPLE_TRACE("ota fetch fail");
                    **break;**           // 这里添加 break 退出试试
                }
`

temp4gh, 2020-02-18:
> [err] IOT_OTA_FetchYield(698): Fetch firmware failed
> 从 log 上看不出为什么会接着进入 reboot, 这中间是否还有其他 log？从代码上看应该要检查固件有效才会进入 reboot,
> ets Jan 8 2013,rst cause:2, boot mode:(3,6)
> load 0x40100000, len 7196, room 16
> 0x40100000: _stext at ??:?

是啊，我也很奇怪这里为什么会重启，中间没有其它内容了，截图如下：
[image]

temp4gh, 2020-02-18:
> examples/ota/ota_example_mqtt/components/ota_handle/ota_solo.c， 202~205 行
>                 } else { IOT_OTA_ReportProgress(h_ota, IOT_OTAP_FETCH_FAILED, NULL); EXAMPLE_TRACE("ota fetch fail"); break;           // 这里添加 break 退出试试 }

感谢支持！

按照建议试了下：走到：
[err] _utils_check_deadloop(254): extend timer for too many times, exit
[err] httpclient_common(562): httpclient_recv_response is error,ret = -8
[inf] httpclient_close(502): client disconnected
[err] ofc_Fetch(70): fetch firmware failed
[err] IOT_OTA_FetchYield(698): Fetch firmware failed

原来这里会重启，现在会出现暂停大概10秒左右，然后系统崩溃了，后续的log如下：

Task stack [ppT] stack from [0x3fff3120] to [0x3fff391c], total [2048] size

                   0          4          8          c         10         14
    18         1c         20         24         28         2c         30
 34         38         3c

3fff3840  0x0000c800 0x3ffec91c 0x00000001 0x3fff2f68 0x00000000 0x3fff38d0 0x00
000000 0x00000008 0x3fff38d0 0x00000001 0xffffffff 0x402b97b9 0x3ffe9eac 0x3ffec
f45 0x00019000 0x00000005
0x402b97b9: pm_onBcnRx at ??:?

3fff3880  0x00010000 0x3ffecf45 0x3ffe9eac 0x402b8f30 0xffffffff 0x3ffecf45 0x3f
fe9eac 0x402b990f 0x3ffe9e98 0x3ffe9920 0x00000001 0x00000000 0x3ffea020 0x3ffea
0d0 0x00000000 0x3ffed003
0x402b8f30: pm_send_nullfunc at ??:?

然后开始循环启动-崩溃-启动-崩溃......
手工重新上电后：
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16
0x40100000: _stext at ??:?

tail 12
chksum 0xde
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0x57
ho 0 tail 12 room 4
load 0x3ffe8420, len 3524, room 12
tail 8
chksum 0xfe
I (50) boot: ESP-IDF  2nd stage bootloader
I (50) boot: compile time 00:07:36
I (50) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DOUT
I (60) boot: SPI Flash Size : 2MB
I (66) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (94) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (106) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (117) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (129) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (141) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (152) boot: End of partition table
I (159) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xada54 (71
1252) map
I (484) esp_image: segment 1: paddr=0x000bda6c vaddr=0x3ffe8000 size=0x00850 (
2128) load
I (486) esp_image: segment 2: paddr=0x000be2c4 vaddr=0x3ffe8850 size=0x0019c (
 412) load
I (496) esp_image: segment 3: paddr=0x000be468 vaddr=0x40100000 size=0x066b4 ( 2
6292) load
0x40100000: _stext at ??:?

E (520) esp_image: Checksum failed. Calculated 0xbd read 0x28
E (520) boot: OTA app partition slot 0 is not bootable
I (528) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0xada54 (71
1252) map
I (853) esp_image: segment 1: paddr=0x001bda6c vaddr=0x00000000 size=0x00000 (
   0) load
I (854) esp_image: segment 2: paddr=0x001bda74 vaddr=0x00000000 size=0x00000 (
   0) load
I (865) esp_image: segment 3: paddr=0x001bda7c vaddr=0x00000000 size=0x00000 (
   0) load
E (879) esp_image: Checksum failed. Calculated 0x3f read 0x0
E (888) boot: OTA app partition slot 1 is not bootable
E (897) boot: No bootable app partitions in the partition table
user code done
仍然是损坏了原有启动分区内容
感觉前面出现大量：
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0
[inf] _http_recv(214): ret of _http_recv is 0

的时候已经不正常了，不知道是否可以避免出现这样的接收数据为0的情况。

另外参考过去ota的流程，难道不是应该先写完ota分区确认无误后，才会修改boot标志，然后设备reboot吗？实在很不理解，为何会在写完ota分区前，就损坏了现有boot的分区，而且会把物联网套件的一机一秘的密匙内容（预先写在0x100000的）也损坏，需要重新写入。推测是有什么操作，会误写flash，所以造成这样的问题。

ljy770, 2020-02-19:
} else {
                    // IOT_OTA_ReportProgress(h_ota, IOT_OTAP_FETCH_FAILED, NULL); // 后面的 log 没有打印出来，怀疑这里引起重启，这句话去掉看看？
                    EXAMPLE_TRACE("ota fetch fail");
                    break;
                }

Eavo, 2020-03-26:
为了方便管理，我们将关闭该问题。如果还有遗留问题，欢迎随时 reopen issue，谢谢。

closed 2020-03-26 by Eavo