espressif/esp-aliyun #183: smart_light如何像mqtt_example那样配网？

asdfw-sdf, 2020-04-09:
----------------------------- 以下请删除 -----------------------------

提交问题时，请先仔细阅读该段提示信息！！

为了更高效的解决 issue 中提交的问题，请按照以下 issue 模板进行提交，请注意以下信息：
1. 提交时请删除这段提示文字；
2. 提交时删除以下由 `//` 开头的文字，并对此做详细补充，如没有特殊信息，可写 `无`。

感谢配合。

----------------------------- 以上请删除 -----------------------------

## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

// 请提供详细的问题描述。

### 2.1 复现步骤

1. ...
2. ...

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs

// 由于 log 可能比较长，请通过附件的形式提供，否则该 issue 页面会非常长，不便于其他开发者查阅。

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

ljy770, 2020-04-09:
mqtt_example 没有配网, 是直接在代码里面设置要连的 ssid 和 password, 您是指这个吗?

asdfw-sdf, 2020-04-09:
就是在menuconfig里面填写wifi名称和密码，不用手机app向设备发送连接的wifi的信息，请问您知道吗？

ljy770, 2020-04-09:
仿照 mqtt_example 添加就好了.
1. 拷贝 esp-aliyun/examples/mqtt/mqtt_example/main/Kconfig.projbuild  到  esp-aliyun/examples/solutions/smart_light/main
2. esp-aliyun/examples/solutions/smart_light/main/app_main.c  里面添加

`
#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS CONFIG_WIFI_PASSWORD
`
conn_mgr_init(); 后面添加:

`conn_mgr_set_wifi_config_ext((const uint8_t *)EXAMPLE_WIFI_SSID, strlen(EXAMPLE_WIFI_SSID), (const uint8_t *)EXAMPLE_WIFI_PASS, strlen(EXAMPLE_WIFI_PASS));`

asdfw-sdf, 2020-04-09:
哇~谢谢大佬耐心地解答，我之前也尝试着做了，但是没有做第一步，不知道那个文件是什么意思，现在可以在menuconfig里面配置了，那之前的那个Factory restore我查了好像是恢复出厂设置，会对刚才的操作有影响吗？

ljy770, 2020-04-09:
Factory restore 这里是为了实现快速重启5次后擦除配置信息,重新进入配网状态, 如果不这么操作就没问题, 或者你不需要也可以删除这句.

asdfw-sdf, 2020-04-09:
好的，没仔细看README里面的图，汗~，谢谢老哥~

rillhu, 2020-04-17:
采用阿里“云智能”APP，扫码配网即可

asdfw-sdf, 2020-04-20:
老哥，可以帮我看一下为啥我云端可以下发，但是LightSwitch只运行了一次呀？就是点击发送1后灯亮，之后设置为低就没反应了，您可以帮我看看吗？

asdfw-sdf, 2020-04-21:
找到问题了，要转换类型，LightSwitch-&gt;valueint

yu870502, 2021-03-05:
请问下各位，这个配网需不要按键配合么？设备正在那运行呢，想过想更换wifi不需要通过按键触发么？