espressif/esp-aliyun #185: MQTT断线后无法重连

jony-xia, 2020-05-08:
## 1. 开发环境

- 模组或开发板：STM32 NODEMCU
- esp-aliyun 版本：sdk-c-3.0.1
- ESP-IDF 版本：v4.0

## 2. 问题描述

链接MQTT服务器运行一段时间后，必出现断开链接的问题
开始出现问题时LOG输出如下：

[0;32mI (350300) MQTT: MQTT RUNNING!![0m
[0;32mI (350501) MQTT: MQTT RUNNING!![0m
[0;32mI (350702) MQTT: MQTT RUNNING!![0m
[0;31mE (350783) esp-tls: read error :-76:[0m
[0;31mE (350783) iot_import_tls: esp_tls_conn_read error, errno:Socket is not connected[0m
[1;31m[err] iotx_mc_read_packet(541): mqtt read error, rc=-76

[0m[1;31m[err] iotx_mc_cycle(1508): readPacket error,result = -14

[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-14

[0m[1;36m[dbg] iotx_mc_cycle(1489): state = 3

[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27

[0m[1;36m[dbg] iotx_mc_cycle(1489): state = 3

·
·
·
·
·
[0m[0;32mI (351116) MQTT: MQTT RUNNING!![0m
[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...

[0m[1;31m[err] iotx_mc_keepalive(1841): reconnect network fail, rc = -1

[0m[1;36m[dbg] iotx_mc_cycle(1489): state = 4

[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27

[0m[1;36m[dbg] iotx_mc_cycle(1489): state = 4

[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
· 
·
·
·
·
[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...

[0m[1;33m[inf] iotx_mc_handle_reconnect(1749): start to reconnect

[0m[1;33m[inf] iotx_mc_attempt_reconnect(1722): reconnect params: MQTTVersion=4, clientID=a1ZmDyPmvTN.LED1|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=60, username=LED1&a1ZmDyPmvTN

[0m[1;36m[dbg] wrapper_mqtt_connect(2697): calling TCP or TLS connect HAL for [1/3] iteration

[0m[0;31mE (358626) esp-tls: couldn't get hostname for :a1ZmDyPmvTN.iot-as-mqtt.cn-shanghai.aliyuncs.com:[0m
[0;31mE (358627) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

[0m[1;36m[dbg] wrapper_mqtt_connect(2697): calling TCP or TLS connect HAL for [2/3] iteration

[0m[0;31mE (366626) esp-tls: couldn't get hostname for :a1ZmDyPmvTN.iot-as-mqtt.cn-shanghai.aliyuncs.com:[0m
[0;31mE (366628) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

[0m[1;36m[dbg] wrapper_mqtt_connect(2697): calling TCP or TLS connect HAL for [3/3] iteration

ljy770, 2020-05-08:
esp-aliyun 版本：sdk-c-3.0.1
ESP-IDF 版本：v4.0
请问您是自己移植的 sdk-c-3.0.1,还是使用的 esp/esp-aliyun 上的版本?
这里出现错误mqtt断线时候wifi有没有断线呢?

jony-xia, 2020-05-08:
自己答一波：原程序中的   void  *pclient = NULL;  位于MQTT配置任务函数中。
将其移到开头的声明后问题解决，目前测试联网稳定。
请管理员关闭该反馈，谢谢！~

closed 2020-05-08 by jony-xia