espressif/esp-aliyun #186: 程序在固定运行时间后重启

feverGeek, 2020-05-11:
## 1. 开发环境

- 模组或开发板：esp32-s
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：esp-aliyun release v2.0
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：v4.0

## 2. 问题描述

程序在固定运行时间后重启
以下是报错

[err] wrapper_mqtt_publish(3126): MQTTPublish is error, rc = -1
[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed                                                                           
app_publish|089 :: publish failed, res = -1                                                                                           
[P] Time: 2020-5-11 19:25:24
I (2399660) wifi: bcn_timout,ap_probe_send_start
W (2399660) wifi: alloc eb len=76 type=2 fail, heap:5744

W (2399660) wifi: m f probe req l=0

[#] Getting property...
[^] Light: 0

[#] Getting property...
[^] Rain: 0

[#] Getting property...
[^] Windy: 0

[err] wrapper_mqtt_publish(3126): MQTTPublish is error, rc = -1
[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed                                                                           
app_publish|089 :: publish failed, res = -1                                                                                           
[#] Getting property...
[^] Light: 0

[#] Getting property...
[^] Rain: 0

[#] Getting property...
[^] Windy: 0

[err] wrapper_mqtt_publish(3126): MQTTPublish is error, rc = -1
[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed                                                                           
app_publish|089 :: publish failed, res = -1                                                                                           
[P] Time: 2020-5-11 19:25:25
W (2400660) wifi: alloc eb len=76 type=2 fail, heap:5740

W (2400660) wifi: m f probe req l=0

[P] Time: 2020-5-11 19:25:26
abort() was called at PC 0x400d1157 on core 0
0x400d1157: __cxa_end_catch at /root/Code/esp-idf-v4.0/components/cxx/cxx_exception_stubs.cpp:13
                                                                                                                                      

ELF file SHA256: bd9c2026fea412d2837dbe5acbfa4f971c5baa14d279edcfa11293f0ec9c1d9c

Backtrace: 0x40088dd9:0x3ffaf8a0 0x40089151:0x3ffaf8c0 0x400d1157:0x3ffaf8e0 0x400e91f4:0x3ffaf900 0x400e0e66:0x3ffaf920 0x400e0eea:0x3ffaf950 0x4013c700:0x3ffaf970 0x40138b4e:0x3ffaf9a0 0x400d26a8:0x3ffaf9e0 0x400d270f:0x3ffafa00 0x4008ebe9:0x3ffafa20
0x40088dd9: invoke_abort at /root/Code/esp-idf-v4.0/components/esp32/panic.c:155
                                                                                                                                      
0x40089151: abort at /root/Code/esp-idf-v4.0/components/esp32/panic.c:172
                                                                                                                                      
0x400d1157: __cxa_end_catch at /root/Code/esp-idf-v4.0/components/cxx/cxx_exception_stubs.cpp:13
                                                                                                                                      
0x400e91f4: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
                                                                                                                                      
0x400e0e66: nvs_open_from_partition at /root/Code/esp-idf-v4.0/components/nvs_flash/src/nvs_api.cpp:285
                                                                                                                                      
0x400e0eea: nvs_open at /root/Code/esp-idf-v4.0/components/nvs_flash/src/nvs_api.cpp:299
                                                                                                                                      
0x4013c700: app_get_data_length at /root/Code/esp-aliyun/project/mqtt_post_reply/components/nvs/app_nvs.c:219
                                                                                                                                      
0x40138b4e: periodic_timer_callback at /root/Code/esp-aliyun/project/mqtt_post_reply/components/mqtt_handle/mqtt_solo.c:767
                                                                                                                                      
0x400d26a8: timer_process_alarm at /root/Code/esp-idf-v4.0/components/esp_common/src/esp_timer.c:297
                                                                                                                                      
0x400d270f: timer_task at /root/Code/esp-idf-v4.0/components/esp_common/src/esp_timer.c:318
                                                                                                                                      
0x4008ebe9: vPortTaskWrapper at /root/Code/esp-idf-v4.0/components/freertos/port.c:143
                                                                                                                                      

 
### 2.1 复现问题的代码
在mqtt_main(){
...
    // 开启定时器 每1秒调用一次回调函数
    ESP_ERROR_CHECK(esp_timer_start_periodic(timer_handle, 1000000)); 
...
}
固定39分钟后就重启,如果不开定时器就不会重启,也不会报错

ljy770, 2020-05-12:
您可以使用 esp-aliyun master 版本, 执行 idf.py menuconfig --->Component config--->FreeRTOS->FreeRTOS timer task stack size
改为: 3072, 再试下问题是否存在

ljy770, 2020-05-19:
@feverGeek 请问下解决内存泄露问题以后,这个问题是否已解决?

feverGeek, 2020-05-19:
对就是内存泄露引起的重启，已经解决了

ljy770, 2020-05-19:
好的

closed 2020-05-21 by Eavo