espressif/esp-aliyun #188: 程序运行一段时间后,mqtt会断开连接

feverGeek, 2020-05-12:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：8a93328
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：463a9d8b7

## 2. 问题描述

程序运行一段时间后,mqtt会断开,已经使用了IOT_MQTT_Yield()函数

### 2.1 复现步骤

[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...
[err] iotx_mc_keepalive(1841): reconnect network fail, rc = -1                                                                        
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 4                                                                
[err] IOT_MQTT_Publish_Simple(764): IOT_MQTT_Publish failed                                                                           
app_publish|089 :: publish failed, res = -1

feverGeek, 2020-05-12:
是因为网络原因吗?

ljy770, 2020-05-13:
这里 wifi 没有断线的话, 就是网络原因了. MQTT重连可以连上吗?

feverGeek, 2020-05-13:
没有断网

feverGeek, 2020-05-13:
重连一直连不上,除非重启程序

feverGeek, 2020-05-13:
一直都在发ping包,也可以收到ping包的回应

feverGeek, 2020-05-13:
阿里云那边需要配置什么吗,或者要付费什么的

ljy770, 2020-05-13:
使用的example 下的哪个示例呢?

feverGeek, 2020-05-13:
W (2420421) wifi: alloc eb len=24 type=3 fail, heap:2792
W (2420421) wifi: m f null
这个警告是什么意思呢

ljy770, 2020-05-13:
这里表示 heap 不够了, 您是使用原生的示例,还是有增加一些自己的内容,我看能够本地复现一下?

ljy770, 2020-05-13:
> 阿里云那边需要配置什么吗,或者要付费什么的

应该是不需要的

feverGeek, 2020-05-13:
有增加自己的内容,但是不清楚是那一部引起的,例程的话没有经过长时间测试

ljy770, 2020-05-13:
> 有增加自己的内容,但是不清楚是那一部引起的,例程的话没有经过长时间测试

上面heap不够的问题, 您排查下是否有内存泄露,比如每隔几秒钟打印 free heap size.
mqtt 重连连不上是长时间测试后出现的吗?
您可以对比下原生的示例同样情况下是否会出现?

feverGeek, 2020-05-13:
是长时间测试出现的

feverGeek, 2020-05-13:
内存泄漏具体是用什么函数呢

feverGeek, 2020-05-13:
我程序里面每次用HAL_Malloc()都free过了,我也不清楚是哪里出现内存泄漏的

ljy770, 2020-05-13:
#include "esp_system.h"
uint32_t esp_get_free_heap_size( void );
uint32_t esp_get_minimum_free_heap_size( void );
用这两个函数看下. 起一个 task 每隔5s 打印一下这两个数值. 看最后是不是越来越少.

feverGeek, 2020-05-13:
esp_get_free_heap_size()的返回值是一直在变小

feverGeek, 2020-05-13:
esp_get_minimun_free_heap_size()的没有变

feverGeek, 2020-05-13:
怎么定位我程序里面哪一部分没有释放内存呢

ljy770, 2020-05-13:
1)您首先看一下原生示例是否同样情况,排除 esp-aliyun + esp-idf 4.0 本身的问题, 因为我们之前是在 esp-idf 3.3.x 版本上测试的,没有试过 esp-idf 4.0.
2) heap 减少需要在稳定运行以后查看, 因为开始会有一些初始化等操作会消耗内存. 具体因为看不到你的代码和 log,不好判断. 您可以把 log 先一下看看

feverGeek, 2020-05-13:
我把log保存在文件里面了,怎么发上来呢,好像没有上传文件的地方

ljy770, 2020-05-13:
直接t

> 我把log保存在文件里面了,怎么发上来呢,好像没有上传文件的地方

直接贴在这里

feverGeek, 2020-05-13:
代码大概就是
init() // 一些初始化
esp_subscribe();

while(1)
{
    ...
    publish(); 
    esp_subscribe();
}

这是运行了一段时间后的log(是在while循环运行时的输出)
[P] Time: 2020-5-13 11:14:14
[P] Time: 2020-5-13 11:14:15
[P] Time: 2020-5-13 11:14:16
[P] Time: 2020-5-13 11:14:17
[P] Time: 2020-5-13 11:14:18
temperature:-0.000000
humidity:-0.000000
report_time:1589339660000

free heap: 20796
minimun free heap: 18664

[P] Time: 2020-5-13 11:14:19
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function                                              
[P] Time: 2020-5-13 11:14:19                                                                                                          
[P] Time: 2020-5-13 11:14:20
[P] Time: 2020-5-13 11:14:21
[P] Time: 2020-5-13 11:14:22
[P] Time: 2020-5-13 11:14:23
[P] Time: 2020-5-13 11:14:24
[P] Time: 2020-5-13 11:14:25
[P] Time: 2020-5-13 11:14:26
[P] Time: 2020-5-13 11:14:27
temperature:-0.000000
humidity:-0.000000
report_time:1589339669000

free heap: 19708
minimun free heap: 17940

[P] Time: 2020-5-13 11:14:28
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function                                              
[P] Time: 2020-5-13 11:14:29                                                                                                          
[P] Time: 2020-5-13 11:14:30
[P] Time: 2020-5-13 11:14:31
[P] Time: 2020-5-13 11:14:32
[P] Time: 2020-5-13 11:14:33
[P] Time: 2020-5-13 11:14:34
[P] Time: 2020-5-13 11:14:34
[P] Time: 2020-5-13 11:14:35
[P] Time: 2020-5-13 11:14:36
temperature:-0.000000
humidity:-0.000000
report_time:1589339679000

free heap: 18984
minimun free heap: 17216

[P] Time: 2020-5-13 11:14:37
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function
[inf] iotx_mc_deliver_message(1280): NO matching any topic, call default handle function                                              
[P] Time: 2020-5-13 11:14:38                                                                                                          
[P] Time: 2020-5-13 11:14:39
[P] Time: 2020-5-13 11:14:40
[P] Time: 2020-5-13 11:14:41
[P] Time: 2020-5-13 11:14:42
[... 942 lines cut ...]
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[err] _mqtt_cycle(1597): error occur rc=-27                                                                                           
[P] Time: 2020-5-13 11:19:31                                                                                                          
[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...
[inf] iotx_mc_handle_reconnect(1749): start to reconnect                                                                              
[inf] iotx_mc_attempt_reconnect(1722): reconnect params: MQTTVersion=4, clientID=a1IaLAeerSz.device|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=60, username=device&a1IaLAeerSz                                
I (2388009) wifi: bcn_timout,ap_probe_send_start                                                                                      
E (2389539) esp-tls: mbedtls_ssl_handshake returned -0x2880
I (2389539) esp-tls: Certificate verified.
E (2389549) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
I (2399509) wifi: bcn_timout,ap_probe_send_start                                                                                      
E (2399549) esp-tls: mbedtls_ssl_handshake returned -0x2880
I (2399549) esp-tls: Certificate verified.
E (2399549) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

feverGeek, 2020-05-13:
测试过例程了,heap比较稳定.
free heap: 178352                                                                                                                     
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368
[inf] iotx_mc_keepalive_sub(1699): send MQTT ping...
[inf] iotx_mc_cycle(1570): receive ping response!                                                                                     
minimum free heap: 164368                                                                                                             

free heap: 178352
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 177964
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

[... 407 lines cut ...]
free heap: 178352
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178084
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

free heap: 178352
minimum free heap: 164368

大概是5秒打印一次

ljy770, 2020-05-13:
可以检查一下周期性执行代码里面有没有内存泄露: 比如看到log 中有:
temperature:-0.000000
humidity:-0.000000
report_time:1589339688000

比如在 HAL_Malloc  和 HAL_Free 的里面添加打印检查
或者检查有没有创建 timer 或者 task 但没有释放

ljy770, 2020-05-18:
@feverGeek 请问问题有解决吗?

feverGeek, 2020-05-18:
问题已经解决了,是我打开nvs的时候没有及时关闭

feverGeek, 2020-05-18:
谢谢

closed 2020-05-21 by Eavo