espressif/esp-aliyun #189: esp-aliyun：esp8266ex断电重启后无法进入app,串口显示ets_main.c

nasoul, 2020-05-14:
## 1. 开发环境

- 模组或开发板：esp8266ex
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：8a93328
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）:d364076
## 2. 问题描述

// 请提供详细的问题描述。

### 2.1 复现步骤

1. make chip=esp8266 defconfig make flash make monitor
2. 断电重启

### 2.2 复现问题的代码

awss

## 3. 调试 Logs
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7176, room 16 
0x40100000: _stext at ??:?

tail 8
chksum 0x18
load 0x3ffe8408, len 24, room 0 
tail 8
chksum 0x31
load 0x3ffe8420, len 3544, room 0 
tail 8
chksum 0x0c
I (62) boot: ESP-IDF v3.2-58-gd364076 2nd stage bootloader
I (62) boot: compile time 21:16:41
I (72) qio_mode: Enabling default flash chip QIO
I (72) boot: SPI Speed      : 40MHz
I (76) boot: SPI Mode       : QIO
I (82) boot: SPI Flash Size : 8MB
I (88) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
I (105) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (117) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (128) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (140) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (152) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (163) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (175) boot: End of partition table
I (181) boot: No factory image, trying OTA 0
I (189) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x65b84 (416644) map
I (342) esp_image: segment 1: paddr=0x00075b9c vaddr=0x3ffe8000 size=0x0078c (  1932) load
I (343) esp_image: segment 2: paddr=0x00076330 vaddr=0x3ffe878c size=0x0019c (   412) load
I (354) esp_image: segment 3: paddr=0x000764d4 vaddr=0x40100000 size=0x0684c ( 26700) load
0x40100000: _stext at ??:?

I (376) boot: Loaded app from partition at offset 0x10000
I (421) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (431) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE

iI (501) phy_init: phy ver: 1058_14
I (511) reset_reason: RTC reset 2 wakeup 0 store 3964287681, reason is 2
[prt] log level set as: [ 4 ]
W (531) wrapper_kv: nvs get blob stassid failed with 1102
W (531) wrapper_kv: nvs get blob scmode failed with 1102
[crt] awss_config_press(180): enable awss
I (541) app main: IOTX_AWSS_ENABLE
I (541) app main: IOTX_AWSS_START
[inf] zconfig_init(260): zconfig_init
I (571) awss: Open monitor mode
E (581) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (581) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (591) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (601) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
E (611) wrapper_product: HAL_GetProductParam nvs_open failed with 1102
[crt] aws_switch_channel(166): chan 1
[crt] awss_save_apinfo(271): [1] ssid:Massky-Door, mac:0c8268ca421a, chn:6, rssi:-63, adha:0
[crt] awss_save_apinfo(271): [2] ssid:Massky, mac:c412f5365ce0, chn:6, rssi:-61, adha:0
[crt] awss_save_apinfo(271): [3] ssid:ChinaNet-eaee, mac:681ab2ea0f94, chn:6, rssi:-90, adha:0
[crt] awss_save_apinfo(271): [4] ssid:HiWiFi_102074, mac:d4ee07102074, chn:6, rssi:-17, adha:0
[crt] awss_save_apinfo(271): [5] ssid:AAIET, mac:349672515ade, chn:6, rssi:-75, adha:0
[crt] awss_save_apinfo(271): [6] ssid:xingkong, mac:8cbebe11161e, chn:1, rssi:-86, adha:0
[crt] awss_save_apinfo(271): [7] ssid:HRD_2.4G, mac:480eec6fbecf, chn:1, rssi:-72, adha:0
[crt] awss_save_apinfo(271): [8] ssid:TP-LINK_7206, mac:bc46992886aa, chn:1, rssi:-75, adha:0
[crt] awss_save_apinfo(271): [9] ssid:ChinaNet-QwwV, mac:18d22521586c, chn:2, rssi:-84, adha:0
[crt] awss_save_apinfo(271): [10] ssid:TP-LINK_GJ007, mac:8ca6df906488, chn:1, rssi:-91, adha:0
[crt] awss_save_apinfo(271): [11] ssid:wyg120244, mac:d8150dd9ab96, chn:1, rssi:-80, adha:0
[crt] awss_save_apinfo(271): [12] ssid:Massky_AP, mac:50bd5f25aebe, chn:1, rssi:-52, adha:0
[crt] awss_save_apinfo(271): [13] ssid:Xiaomi_PC-20, mac:6409800b0545, chn:1, rssi:-91, adha:0
[crt] awss_save_apinfo(271): [14] ssid:ChinaNet-ijR4, mac:ca50e9c72e77, chn:1, rssi:-88, adha:0
[crt] awss_save_apinfo(271): [crt] aws_switch_channel(166): chan 6
[15] ssid:iTV-ijR4, mac:da50e9c72e77, chn:1, rssi:-87, adha:0
[crt] awss_save_apinfo(271): [16] ssid:EBTSZ-FOTA, mac:8ca6df3ef4f0, chn:6, rssi:-79, adha:0
[crt] awss_save_apinfo(271): [17] ssid:ackpass, mac:e4f3f51bff20, chn:6, rssi:-74, adha:0
[crt] awss_save_apinfo(271): [18] ssid:aWiFi, mac:7e03c9b879c9, chn:5, rssi:-81, adha:0
[crt] awss_save_apinfo(271): [19] ssid:小天才行政, mac:246968c147f3, chn:6, rssi:-68, adha:0
[crt] awss_save_apinfo(271): [20] ssid:WiFi-9A, mac:008736073b9a, chn:6, rssi:-78, adha:0
[crt] awss_save_apinfo(271): [21] ssid:szyz, mac:147590ef7de0, chn:6, rssi:-78, adha:0
[crt] awss_save_apinfo(271): [22] ssid:KK-STRIP_DCD7, mac:28d98a8bdcd7, chn:6, rssi:-71, adha:0
[crt] aws_switch_channel(166): chan 11
[crt] awss_save_apinfo(271): [23] ssid:, mac:62da83f91443, chn:11, rssi:-89, adha:0
[crt] awss_save_apinfo(271): [24] ssid:TP-LINK_0572, mac:8ca6dfbe0572, chn:11, rssi:-90, adha:0
[crt] awss_save_apinfo(271): [25] ssid:AD, mac:d076e7173e2e, chn:11, rssi:-83, adha:0
[crt] awss_save_apinfo(271): [26] ssid:Massky_AP2, mac:808917614bc8, chn:11, rssi:-67, adha:0
[crt] awss_save_apinfo(271): [27] ssid:NJBX, mac:60da83f91445, chn:11, rssi:-86, adha:0
[crt] awss_save_apinfo(271): [28] ssid:tp7206, mac:b0958e9d0fd6, chn:11, rssi:-69, adha:0
[crt] awss_save_apinfo(271): [29] ssid:TP-LINK_5ED9, mac:f483cd925ed9, chn:11, rssi:-90, adha:0
[crt] awss_save_apinfo(271): [30] ssid:ChinaNet-UXm3, mac:34e0cf2cae9e, chn:11, rssi:-89, adha:0
[crt] awss_save_apinfo(271): [31] ssid:ydsadc, mac:9c216a0eda6a, chn:11, rssi:-92, adha:0
[crt] awss_save_apinfo(271): [32] ssid:ChinaNet-FVQV, mac:88bcc14f6750, chn:11, rssi:-80, adha:0
[crt] awss_save_apinfo(271): [33] ssid:TP-LINK_贤二, mac:ba958e9d0fd6, chn:11, rssi:-71, adha:0
[crt] awss_save_apinfo(271): [34] ssid:iTV-UXm3, mac:34e0cf2cae9f, chn:11, rssi:-90, adha:0
[crt] aws_switch_channel(166): chan 1
[crt] awss_save_apinfo(271): [35] ssid:iTV-QwwV, mac:08d22521586c, chn:2, rssi:-87, adha:0
[crt] awss_save_apinfo(271): [36] ssid:aWiFi-QwwV, mac:28d22521586c, chn:2, rssi:-85, adha:0
[crt] aws_switch_channel(166): chan 2
[crt] awss_save_apinfo(271): [37] ssid:10-203, mac:04d4c43e3278, chn:2, rssi:-87, adha:0
[crt] aws_switch_channel(166): chan 3
[crt] awss_save_apinfo(271): [38] ssid:ChinaNet-ARbJ, mac:e8b54134dfd0, chn:4, rssi:-75, adha:0
[crt] aws_switch_channel(166): chan 4
[crt] awss_save_apinfo(271): [39] ssid:XINHUA, mac:0cb6d24b95b1, chn:3, rssi:-89, adha:0
[crt] awss_save_apinfo(271): [40] ssid:ChinaNet-6203, mac:7c03c9a879c9, chn:5, rssi:-88, adha:0
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 7
[crt] aws_switch_channel(166): chan 8
[crt] awss_save_apinfo(271): [41] ssid:ChinaNet-SaTb, mac:88bcc112bf60, chn:8, rssi:-82, adha:0
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] awss_save_apinfo(271): [42] ssid:ChinaNet-V25z, mac:5c09795422ac, chn:10, rssi:-93, adha:0
[crt] awss_save_apinfo(271): [43] ssid:ChinaNet-tvLQ, mac:30d386c6fbcd, chn:10, rssi:-90, adha:0
[crt] aws_switch_channel(166): chan 11
[crt] awss_save_apinfo(271): [44] ssid:, mac:60da83f91443, chn:11, rssi:-93, adha:0
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c 
## 4. 其他项
flash用的8MB的，把fctry,    data, nvs,     0x100000,0x4000注释掉断电重启正常

ljy770, 2020-05-14:
flash用的8MB 应该不会有问题，只是用了前 2MB . 确认下 flash SIZE 是 8MB 不是 8Mb ?

nasoul, 2020-05-15:
> flash用的8MB 应该不会有问题，只是用了前 2MB . 确认下 flash SIZE 是 8MB 不是 8Mb ?
非常感谢，拆换了flash问题没有了。

closed 2020-05-15 by nasoul