espressif/esp-aliyun #190: 在连接上服务器之后会重启，log打印了栈溢出的错误

ChampOnBon, 2020-05-15:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：506ec36
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：34f6bde8b

## 2. 问题描述

我们开发的是国际站的设备，按照对接指南进行设置，并且在阿里云创建对应的设备。在软AP配网成功之后，设备连接阿里云服务器，从log上可以看出已经连接成功，但在连接成功并且进行短暂的数据收发后设备重启，并且重启后现象不断重复，重启时log打印出栈溢出的错误。APP端可以正常完成配对，但由于设备重启，所以APP无法控制设备。APP使用的是德国账号，从log上也可以看出设备首先连接了新加坡服务器，然后重启切换到德国服务器。将阿里云线程栈大小从3K增加至6K问题依然存在。

### 2.1 复现步骤

1. 根据对接指南取消MQTT DIRECT。
2. 使用smart_light工程，在app_main.c文件的conn_mgr_start()函数调用前增加conn_mgr_set_sc_mode(CONN_SOFTAP_MODE);来指定使用软AP配网。
3.编译工程并且烧录。
4.使用云智能安卓开发版APP进行配对。

### 2.2 复现问题的代码

[smart_light.zip](https://github.com/espressif/esp-aliyun/files/4633263/smart_light.zip)

## 3. 调试 Logs

[log.txt](https://github.com/espressif/esp-aliyun/files/4633271/log.txt)

## 4. 其他项

手机系统是安卓7.1.2，路由器型号华硕AC1900P，APP是云智能安卓开发版V3.1.1

ljy770, 2020-05-15:
编译之前有没有执行 'make  chip=esp32 defconfig' 呢? 其中有设置 timer task stack size = 3072

ChampOnBon, 2020-05-15:
执行 'make chip=esp32 defconfig' 后正常了，谢谢！

closed 2020-05-15 by ChampOnBon