espressif/esp-aliyun #191: 能否让设备自动重启来连接新加坡以外的服务器？

ChampOnBon, 2020-05-18:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：506ec36
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：34f6bde8b

## 2. 问题描述

现有esp-aliyun的代码在首次连接新加坡服务器后，需要手动重启设备来连接至其他服务器。是否有办法让设备自动重启来较少这一手动操作？如果有，需要在哪些地方添加重启代码？

ljy770, 2020-05-18:
@ChampOnBon 你好,需要添加一些内容, 在其他 aliyun 版本上看的操作是: 注册 redirect  topic,收到这个 topic 后断开 mqtt 连接重新 mqtt 连接, 就可以连接到其他服务器. 或者按您说的, 收到这个 topic 重启设备. 您需要哪一种呢? 可能第一种源代码有些 bug, 需要多一些调试.

ChampOnBon, 2020-05-18:
根据我们产品的需求，偏向于使用第二种。具体需要怎么添加代码呢？
另外，我们产品还接入了homekit平台，在IOS设备上使用时为homekit配网，非阿里云提供的配网。那么还能够通过订阅redirect topic来接收切换服务器的消息吗？

ljy770, 2020-05-18:
这里是注册 mqtt topic, 配网方式不影响. 如何添加代码: 下载 飞燕SDK, https://code.aliyun.com/living_platform/ali-smartliving-device-sdk-c/commits/rel_1.3.0, 代码中查找 'redirect '字段, 相关操作移植到 esp-aliyun.

closed 2020-05-18 by ChampOnBon

ljy770, 2020-05-19:
@ChampOnBon esp-alilyun 已经添加自动切换功能:自动重启切换到新的服务器.