espressif/esp-aliyun #192: 连接阿里云物联网超过一定时间（24小时）掉线

ZorkHuang, 2020-06-29:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本 ：v3.3(bb376c5)
- ESP-IDF 版本：v4.0(fe67bedee)

## 2. 问题描述

连接阿里云超过一定时间（大约24小时以上）设备掉线，无法连接阿里云；此时设备在路由器上能够正常获取到IP

### 2.1 复现步骤

### 2.2 复现问题的代码

static int linkkit_thread(void *paras)
{
    int res = 0;
    iotx_linkkit_dev_meta_info_t master_meta_info;
    int domain_type = 0, dynamic_register = 0, post_reply_need = 0;

#ifdef ATM_ENABLED
    if (IOT_ATM_Init() < 0)
    {
        EXAMPLE_TRACE("IOT ATM init failed!\n");
        return -1;
    }
#endif

    memset(&g_user_example_ctx, 0, sizeof(user_example_ctx_t));
    HAL_GetProductKey(PRODUCT_KEY);
    HAL_GetProductSecret(PRODUCT_SECRET);
    HAL_GetDeviceName(DEVICE_NAME);
    HAL_GetDeviceSecret(DEVICE_SECRET);
    memset(&master_meta_info, 0, sizeof(iotx_linkkit_dev_meta_info_t));
    memcpy(master_meta_info.product_key, PRODUCT_KEY, strlen(PRODUCT_KEY));
    memcpy(master_meta_info.product_secret, PRODUCT_SECRET, strlen(PRODUCT_SECRET));
    memcpy(master_meta_info.device_name, DEVICE_NAME, strlen(DEVICE_NAME));
    memcpy(master_meta_info.device_secret, DEVICE_SECRET, strlen(DEVICE_SECRET));
    print_ali_config("load ali config");

    /* Register Callback */
    IOT_RegisterCallback(ITE_AWSS_STATUS, user_awss_status_event_handler);
    IOT_RegisterCallback(ITE_CONNECT_SUCC, user_connected_event_handler);
    IOT_RegisterCallback(ITE_CONNECT_FAIL, user_connect_fail_event_handler);
    IOT_RegisterCallback(ITE_DISCONNECTED, user_disconnected_event_handler);
    IOT_RegisterCallback(ITE_RAWDATA_ARRIVED, user_rawdata_arrived_event_handler);
    IOT_RegisterCallback(ITE_SERVICE_REQUEST, user_service_request_event_handler);
    IOT_RegisterCallback(ITE_PROPERTY_SET, user_property_set_event_handler);
    IOT_RegisterCallback(ITE_PROPERTY_GET, user_property_get_event_handler);
    IOT_RegisterCallback(ITE_PROPERTY_DESIRED_GET_REPLY, user_property_desired_get_reply_event_handler);
    IOT_RegisterCallback(ITE_REPORT_REPLY, user_report_reply_event_handler);
    IOT_RegisterCallback(ITE_TRIGGER_EVENT_REPLY, user_trigger_event_reply_event_handler);
    IOT_RegisterCallback(ITE_TIMESTAMP_REPLY, user_timestamp_reply_event_handler);
    IOT_RegisterCallback(ITE_TOPOLIST_REPLY, user_toplist_reply_event_handler);
    IOT_RegisterCallback(ITE_PERMIT_JOIN, user_permit_join_event_handler);
    IOT_RegisterCallback(ITE_INITIALIZE_COMPLETED, user_initialized);
    IOT_RegisterCallback(ITE_FOTA, user_fota_event_handler);
    IOT_RegisterCallback(ITE_COTA, user_cota_event_handler);
    IOT_RegisterCallback(ITE_MQTT_CONNECT_SUCC, user_mqtt_connect_succ_event_handler);

    domain_type = IOTX_CLOUD_REGION_SHANGHAI;
    IOT_Ioctl(IOTX_IOCTL_SET_DOMAIN, (void *)&domain_type);

    /* Choose Login Method */
    dynamic_register = 0;
    IOT_Ioctl(IOTX_IOCTL_SET_DYNAMIC_REGISTER, (void *)&dynamic_register);

    /* post reply doesn't need */
    /* value(int*): 0 - Disable property post reply by cloud; 1 - Enable property post reply by cloud */
    post_reply_need = 0;
[... 114 lines cut ...]
[0;32mI (297061444) esp-tls: Failed to verify peer certificate![0m
[0;32mI (297061454) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA
[0m
[0;31mE (297061454) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m
[10:18:28.800]收←◆[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1c60FnaYQk.104084286F24|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=104084286F24&a1c60FnaYQk
[0m[0;31mE (297062474) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] iotx_mc_attempt_reconnect(1728): run iotx_mqtt_connect() error!
[0m[1;31m[err] iotx_mc_handle_reconnect(1784): mqtt reconnect failed rc = -14
[0m[1;31m[err] iotx_mc_keepalive(1841): reconnect network fail, rc = -14
[0m
[10:18:28.931]收←◆[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...
[0m[1;31m[err] iotx_mc_keepalive(1841): reconnect network fail, rc = -1
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...
[0m
[10:18:29.100]收←◆[1;31m[err] iotx_mc_keepalive(1841): reconnect network fail, rc = -1
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;33m[inf] iotx_mc_handle_reconnect(1742): Waiting to reconnect...

ljy770, 2020-07-01:
看 log 是 CA 校验失败。

建议使用 esp-idf 加 tag 的版本重新测试，https://github.com/espressif/esp-idf/tags

ZorkHuang, 2020-07-12:
好的，谢谢！
这个问题我们跟阿里云的技术交流过，应该是网络不稳定导致阿里云认为设备掉线（30秒无心跳包）把连接主动关闭了，客户端再重用以前的连接重试，无论如何都不行；我们是手动释放这个链接然后重新连接阿里云解决的。至于用另外的版本：esp-idf/tags来测试，晚点可以尝试下

ZorkHuang, 2020-07-12:
设备运行几天后会出现如下问题，不清楚该如何处理，谢谢
[0m[0;32mI (414956973) LOCK485: IOTX_CONN_CLOUD[0m
[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1c60FnaYQk.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1c60FnaYQk.104084286F24|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: 104084286F24&a1c60FnaYQk
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m
[08:59:05.102]收←◆[0;31mE (414957863) esp-tls: mbedtls_ssl_handshake returned -0x10[0m
[0;32mI (414957873) esp-tls: Certificate verified.[0m
[0;31mE (414957873) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m
[08:59:06.943]收←◆[0;31mE (414959713) esp-tls: mbedtls_ssl_handshake returned -0x10[0m
[0;32mI (414959713) esp-tls: Certificate verified.[0m
[0;31mE (414959713) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m
[08:59:08.787]收←◆[0;31mE (414961553) esp-tls: mbedtls_ssl_handshake returned -0x10[0m
[0;32mI (414961553) esp-tls: Certificate verified.[0m
[0;31mE (414961553) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[0m
[08:59:09.806]收←◆[1;33m[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1c60FnaYQk.104084286F24|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=104084286F24&a1c60FnaYQk
[0m[0;31mE (414962583) iot_import_tls: HAL_SSL_Write, handle == NULL[0m
[1;31m[err] MQTTConnect(460): send connect packet failed
[0m[1;31m[err] _mqtt_connect(786): send connect packet failed, rc = -14
[0m[1;31m[err] IOT_MQTT_Construct(578): wrapper_mqtt_connect failed
[0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m
[08:59:10.549]收←◆[1;35m[wrn] IOT_MQTT_Construct(467): Using default hostname: 'a1c60FnaYQk.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[0m[1;35m[wrn] IOT_MQTT_Construct(474): Using default port: [443]
[0m[1;35m[wrn] IOT_MQTT_Construct(481): Using default client_id: a1c60FnaYQk.104084286F24|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[0m[1;35m[wrn] IOT_MQTT_Construct(488): Using default username: 104084286F24&a1c60FnaYQk
[0m[1;35m[wrn] IOT_MQTT_Construct(496): Using default password: ******
[0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
[0m
[08:59:11.445]收←◆[0;31mE (414964213) esp-tls: mbedtls_ssl_handshake returned -0x10[0m
[0;32mI (414964213) esp-tls: Certificate verified.[0m
[0;31mE (414964213) esp-tls: Failed to open new connection[0m
[1;31m[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

ljy770, 2020-07-12:
谢谢反馈。
对刚提到的问题，请打印下失败时候的的 heap size 或者周期性打印 heap size, 看是否内存不够导致。如果是内存不够，可能是由于内存泄露导致，排查一下您前面提到的手动释放连接过程。
#include "esp_system.h"
uint32_t esp_get_free_heap_size( void );