espressif/esp-aliyun #193: 天猫精灵 零配不成功

yl673091830, 2020-06-30:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（）：
- ESP8266_RTOS_SDK 版本（ crosstool-ng-1.22.0-100-ge567ec7b）：
- ESP-IDF 版本（v3.2）：

## 2. 问题描述

使用公版APP 一键配网可以成功， 但是使用天猫精灵零配不成功，侦测不到天猫精灵的配网信号，一直在扫AP，切换channel

### 2.1 复现步骤

1. ...
2. ...

### 2.2 复现问题的代码

## 3. 调试 Logs

[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 4
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 7
[crt] aws_switch_channel(166): chan 8
[crt] awss_save_apinfo(271): [32] ssid:CMCCxxxxxxxxxxxxxx
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 10
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 3
[crt] aws_switch_channel(166): chan 5
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 8
[crt] aws_switch_channel(166): chan 9
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 12
[crt] aws_switch_channel(166): chan 13
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 6
[crt] aws_switch_channel(166): chan 11
[crt] aws_switch_channel(166): chan 1
[crt] aws_switch_channel(166): chan 2
[crt] aws_switch_channel(166): chan 3

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

Eavo, 2020-07-01:
可以试一下这个工程：https://github.com/espressif/esp-ali-smartliving

yl673091830, 2020-07-01:
邮件已收到，我先按这个工程调试一下，后续有问题或者有了结果我再告诉你，谢谢！

yl673091830, 2020-07-01:
@Eavo 你提供的工程连接，按照步骤操作，提示如下错误，请问该如何解决？
Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
# make reconfig
SELECT A CONFIGURATION:

1) config.esp32.aos     4) config.mk3060.aos    7) config.ubuntu.x86
2) config.esp8266.aos   5) config.mk3080.aos    8) config.win7.mingw32
3) config.macos.x86     6) config.rhino.make    9) config.xboard.make
#? 2

SELECTED CONFIGURATION:

VENDOR :   esp8266
MODEL  :   aos

CONFIGURE .............................. [examples]
CONFIGURE .............................. [src/infra/log]
CONFIGURE .............................. [src/infra/system]
CONFIGURE .............................. [src/infra/utils]
CONFIGURE .............................. [src/protocol/alcs]
CONFIGURE .............................. [src/protocol/coap/cloud]
CONFIGURE .............................. [src/protocol/coap/local]
CONFIGURE .............................. [src/protocol/http]
CONFIGURE .............................. [src/protocol/http2]
CONFIGURE .............................. [src/protocol/mqtt]
CONFIGURE .............................. [src/ref-impl/hal]
CONFIGURE .............................. [src/ref-impl/tls]
CONFIGURE .............................. [src/sdk-impl]
CONFIGURE .............................. [src/services/awss]
CONFIGURE .............................. [src/services/dev_bind]
CONFIGURE .............................. [src/services/dev_diagnosis]
CONFIGURE .............................. [src/services/http2_stream]
CONFIGURE .............................. [src/services/linkkit/cm]
CONFIGURE .............................. [src/services/linkkit/dev_reset]
CONFIGURE .............................. [src/services/linkkit/dm]
CONFIGURE .............................. [src/services/mdal/mal]
CONFIGURE .............................. [src/services/mdal/sal]
CONFIGURE .............................. [src/services/ota]
CONFIGURE .............................. [src/services/shadow]
CONFIGURE .............................. [src/services/subdev]
CONFIGURE .............................. [src/tools/linkkit_tsl_convert]

BUILDING WITH EXISTING CONFIGURATION:

VENDOR :   esp8266
MODEL  :   aos

this one
Components:

. examples
. src/infra/log
. src/infra/system
. src/infra/utils
. src/protocol/alcs
. src/protocol/coap/cloud
. src/protocol/coap/local
. src/protocol/http
. src/protocol/http2
. src/protocol/mqtt
. src/ref-impl/hal
. src/ref-impl/tls
. src/sdk-impl
. src/services/awss
. src/services/dev_bind
. src/services/dev_diagnosis
. src/services/http2_stream
. src/services/linkkit/cm
. src/services/linkkit/dev_reset
. src/services/linkkit/dm
. src/services/mdal/mal
. src/services/mdal/sal
. src/services/ota
. src/services/shadow
. src/services/subdev
. src/tools/linkkit_tsl_convert

this one

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
# make menuconfig
make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-top.mk:202：menuconfig] 错误 126

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
# make
BUILDING WITH EXISTING CONFIGURATION:

VENDOR :   esp8266
MODEL  :   aos

echo this one
this one
make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
# make clean

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
# make
BUILDING WITH EXISTING CONFIGURATION:

VENDOR :   esp8266
MODEL  :   aos

echo this one
this one
make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2

yl673091830, 2020-07-01:
你好，按照你提供的工程连接，提示如下错误，请问该如何解决？

下面是调试log：
Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c

make reconfig

SELECT A CONFIGURATION:

config.esp32.aos 4) config.mk3060.aos 7) config.ubuntu.x86

config.esp8266.aos 5) config.mk3080.aos 8) config.win7.mingw32

config.macos.x86 6) config.rhino.make 9) config.xboard.make
#? 2

SELECTED CONFIGURATION:

VENDOR : esp8266
MODEL : aos

CONFIGURE .............................. [examples]
CONFIGURE .............................. [src/infra/log]
CONFIGURE .............................. [src/infra/system]
CONFIGURE .............................. [src/infra/utils]
CONFIGURE .............................. [src/protocol/alcs]
CONFIGURE .............................. [src/protocol/coap/cloud]
CONFIGURE .............................. [src/protocol/coap/local]
CONFIGURE .............................. [src/protocol/http]
CONFIGURE .............................. [src/protocol/http2]
CONFIGURE .............................. [src/protocol/mqtt]
CONFIGURE .............................. [src/ref-impl/hal]
CONFIGURE .............................. [src/ref-impl/tls]
CONFIGURE .............................. [src/sdk-impl]
CONFIGURE .............................. [src/services/awss]
CONFIGURE .............................. [src/services/dev_bind]
CONFIGURE .............................. [src/services/dev_diagnosis]
CONFIGURE .............................. [src/services/http2_stream]
CONFIGURE .............................. [src/services/linkkit/cm]
CONFIGURE .............................. [src/services/linkkit/dev_reset]
CONFIGURE .............................. [src/services/linkkit/dm]
CONFIGURE .............................. [src/services/mdal/mal]
CONFIGURE .............................. [src/services/mdal/sal]
CONFIGURE .............................. [src/services/ota]
CONFIGURE .............................. [src/services/shadow]
CONFIGURE .............................. [src/services/subdev]
CONFIGURE .............................. [src/tools/linkkit_tsl_convert]

BUILDING WITH EXISTING CONFIGURATION:

VENDOR : esp8266
MODEL : aos

this one
Components:

. examples
. src/infra/log
. src/infra/system
. src/infra/utils
. src/protocol/alcs
. src/protocol/coap/cloud
. src/protocol/coap/local
. src/protocol/http
. src/protocol/http2
. src/protocol/mqtt
. src/ref-impl/hal
. src/ref-impl/tls
. src/sdk-impl
. src/services/awss
. src/services/dev_bind
. src/services/dev_diagnosis
. src/services/http2_stream
. src/services/linkkit/cm
. src/services/linkkit/dev_reset
. src/services/linkkit/dm
. src/services/mdal/mal
. src/services/mdal/sal
. src/services/ota
. src/services/shadow
. src/services/subdev
. src/tools/linkkit_tsl_convert

this one

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c

make menuconfig

make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-top.mk:202：menuconfig] 错误 126

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c

make

BUILDING WITH EXISTING CONFIGURATION:

VENDOR : esp8266
MODEL : aos

echo this one
this one
make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c

make clean

Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c

make

BUILDING WITH EXISTING CONFIGURATION:

VENDOR : esp8266
MODEL : aos

echo this one
this one
make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2

-------------------------------------------------------------------------------------------------------------------------------------------------

 
可以试一下这个工程：https://github.com/espressif/esp-ali-smartliving

ljy770, 2020-07-01:
@yl673091830 请说明一下使用的 esp-aliyun 版本和 ESP8266_RTOS_SDK 版本？
使用命令： git describe --always --tags --dirty

另外对天猫精灵说 “发现设备” 之后， 天猫精灵是否回复了“未发现设备”？

yl673091830, 2020-07-01:
1、是这个版本信息吗 v3.2-dirty
# git describe --always --tags --dirty
v3.2-dirty

2、天猫精灵搜索一会说“搜了一分钟都没找到智能设备，是否产品没有进入配网状态”，但是从调试log上看 ESP8266是在配网状态的

ljy770, 2020-07-01:
你给的是 ESP8266_RTOS_SDK 版本， esp-aliyun 版本呢？ 是用的 master 最新的吗？

yl673091830, 2020-07-01:
esp-aliyun 是最新版本的

yl673091830, 2020-07-01:
esp-aliyun 是用最新的版本

ljy770, 2020-07-01:
设备有没有烧录三元组呢？ 或者 make erase_flash 之后没有重新烧写？

yl673091830, 2020-07-01:
重新烧录了三元组。反复试了多次都不行，天猫精灵配网不成功，然后用手机一键配网就可以，应该不是三元组的问题

yl673091830, 2020-07-01:
用一键配网时，需要用手机扫描二维码才能配网成功，天猫精灵零配时没有这一步，是不是设备虽然在配网模式，但是没有发送三元组信息或者发送的信息有问题，导致天猫精灵找不到找不到设备

ljy770, 2020-07-01:
我们内部先确认一下给您答复。

需要看下设备是否已发出零配请求帧, 即 vender specific OUI 是 "D8-96-E0" 的 probe request.
如果发出了，为什么天猫精灵没有收到。

yl673091830, 2020-07-01:
好的，谢谢，等您答复~

另外，1、一键配网时需要扫描二维码才能配网成功，是不是这个版本不支持标准配网？2、后续会推出支持标准配网的版本吗？

ljy770, 2020-07-01:
不好意思，您说的标准配网是哪种？  我们有支持一键配网，天猫精灵配网，设备热点配网

yl673091830, 2020-07-01:
这是生活物联网平台的截图，必须选自定义配网，一键配网才能成功，默认的选官方标准配网，设备一直在切换信道，获取不到配网信息：图 1、

图 2

yuanjianmin, 2020-07-01:
> 
> 
> @Eavo 你提供的工程连接，按照步骤操作，提示如下错误，请问该如何解决？
> 
> 下面是调试log：
> Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
> # make reconfig
> 
> SELECT A CONFIGURATION:
> 
>     1. config.esp32.aos     4) config.mk3060.aos    7) config.ubuntu.x86
> 
>     2. config.esp8266.aos   5) config.mk3080.aos    8) config.win7.mingw32
> 
>     3. config.macos.x86     6) config.rhino.make    9) config.xboard.make
>        #? 2
> 
> 
> SELECTED CONFIGURATION:
> 
> VENDOR : esp8266
> MODEL : aos
> 
> CONFIGURE .............................. [examples]
> CONFIGURE .............................. [src/infra/log]
> CONFIGURE .............................. [src/infra/system]
> CONFIGURE .............................. [src/infra/utils]
> CONFIGURE .............................. [src/protocol/alcs]
> CONFIGURE .............................. [src/protocol/coap/cloud]
> CONFIGURE .............................. [src/protocol/coap/local]
> CONFIGURE .............................. [src/protocol/http]
> CONFIGURE .............................. [src/protocol/http2]
> CONFIGURE .............................. [src/protocol/mqtt]
> CONFIGURE .............................. [src/ref-impl/hal]
> CONFIGURE .............................. [src/ref-impl/tls]
> CONFIGURE .............................. [src/sdk-impl]
> CONFIGURE .............................. [src/services/awss]
> CONFIGURE .............................. [src/services/dev_bind]
> CONFIGURE .............................. [src/services/dev_diagnosis]
> CONFIGURE .............................. [src/services/http2_stream]
> CONFIGURE .............................. [src/services/linkkit/cm]
> CONFIGURE .............................. [src/services/linkkit/dev_reset]
> CONFIGURE .............................. [src/services/linkkit/dm]
> CONFIGURE .............................. [src/services/mdal/mal]
> CONFIGURE .............................. [src/services/mdal/sal]
> CONFIGURE .............................. [src/services/ota]
> CONFIGURE .............................. [src/services/shadow]
> CONFIGURE .............................. [src/services/subdev]
> CONFIGURE .............................. [src/tools/linkkit_tsl_convert]
> 
> BUILDING WITH EXISTING CONFIGURATION:
> 
> VENDOR : esp8266
> MODEL : aos
> 
> this one
> Components:
> 
> . examples
> . src/infra/log
> . src/infra/system
> . src/infra/utils
> . src/protocol/alcs
> . src/protocol/coap/cloud
> . src/protocol/coap/local
> . src/protocol/http
> . src/protocol/http2
> . src/protocol/mqtt
> . src/ref-impl/hal
> . src/ref-impl/tls
> . src/sdk-impl
> . src/services/awss
> . src/services/dev_bind
> . src/services/dev_diagnosis
> . src/services/http2_stream
> . src/services/linkkit/cm
> . src/services/linkkit/dev_reset
> . src/services/linkkit/dm
> . src/services/mdal/mal
> . src/services/mdal/sal
> . src/services/ota
> . src/services/shadow
> . src/services/subdev
> . src/tools/linkkit_tsl_convert
> 
> this one
> 
> Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
> # make menuconfig
> 
> make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-top.mk:202：menuconfig] 错误 126
> 
> Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
> # make
> 
> BUILDING WITH EXISTING CONFIGURATION:
> 
> VENDOR : esp8266
> MODEL : aos
> 
> echo this one
> this one
> make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
> make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2
> 
> Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
> # make clean
> 
> Administrator@Q8VVDYMB1BQOX2A MINGW32 /home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c
> # make
> 
> BUILDING WITH EXISTING CONFIGURATION:
> 
> VENDOR : esp8266
> MODEL : aos
> 
> echo this one
> this one
> make[1]: *** 没有规则可制作目标“E:/▒▒·▒▒▒▒/WiFi▒▒▒▒/▒▒▒▒▒▒▒/msys32/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/src/infra/utils/digest/utils_sha1.c”，由“/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_sha1.o” 需求。 停止。
> make: *** [/home/ESP8266_RTOS_SDK/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-submods.mk:5：sub-mods] 错误 2

你可以尝试一下make reconfig后选择esp8266(对应的是2)，然后不执行make menuconfig（看起来是你用的msys32 shell不支持make menuconfig）,直接执行make看看还会不会报错，另外清空的命令你可以执行make distclean

ljy770, 2020-07-01:
> 这是生活物联网平台的截图，必须选自定义配网，一键配网才能成功，默认的选官方标准配网，设备一直在切换信道，获取不到配网信息：图 1、 图 2

看不到图片，我在 ”人机交互“->”配网引导“ 页面没有看到官方标准配网，只有 默认配网方式：备选配网方式：零配方式（需设备支持）

ljy770, 2020-07-01:
> 好的，谢谢，等您答复~ 另外，1、一键配网时需要扫描二维码才能配网成功，是不是这个版本不支持标准配网？2、后续会推出支持标准配网的版本吗？

您有多的 8266 开板吗？ 建议试试其他的板子能否成功？ 是不是使用的 ESP-WROOM-02D？

yl673091830, 2020-07-01:
> > 这是生活物联网平台的截图，必须选自定义配网，一键配网才能成功，默认的选官方标准配网，设备一直在切换信道，获取不到配网信息：图 1、 图 2
> 
> 看不到图片，我在 ”人机交互“->”配网引导“ 页面没有看到官方标准配网，只有 默认配网方式：备选配网方式：零配方式（需设备支持）

在“https://living.aliyun.com/”新建产品之后，第一次进入“人机交互”->”配网引导“页面，是默认的“ 标准配网”，这个页面看不到 “默认配网方式：备选配网方式：零配方式（需设备支持）” 的选项，在右上角点击自定义配网才能看到选项。设定之后再次进入”配网引导”页面 还是上次设定的自定义配网方式，点击右上角的提示：”自定义配网体验不好？使用品类标准配网“，可以切回 “品类标准配网”，这里的“标准配网”不成功~

yl673091830, 2020-07-01:
> > 好的，谢谢，等您答复~ 另外，1、一键配网时需要扫描二维码才能配网成功，是不是这个版本不支持标准配网？2、后续会推出支持标准配网的版本吗？
> 
> 您有多的 8266 开板板吗？ 建议试试其他的板子能否成功？ 是不是使用的 ESP-WROOM-02D？

现在手上只这一块板子，我用的是 ESP-12S

ljy770, 2020-07-02:
请读取一下板子的 mac 地址？
在 esp_wifi_start 之后调用 esp_wifi_get_mac(WIFI_IF_STA, mac[6]);
或者查找 download 时候会打印 MAC 信息：
```
Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000 )...
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 80:7d:3a:5e:d7:xx
```

yl673091830, 2020-07-02:
> 请读取一下板子的 mac 地址？
> 在 esp_wifi_start 之后调用 esp_wifi_get_mac(WIFI_IF_STA, mac[6]);
> 或者查找 download 时候会打印 MAC 信息：
> 
> ```
> Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000 )...
> esptool.py v2.4.0
> Connecting....
> Chip is ESP8266EX
> Features: WiFi
> MAC: 80:7d:3a:5e:d7:xx
> ```

Chip is ESP8266EX
Features: WiFi
MAC: 48:3f:da:65:ee:05

ljy770, 2020-07-02:
请重启天猫精灵，重试下能否发现。
我们在调试中也发现偶尔有这个问题，一直不能发现某个 mac 地址的设备，改成其他mac 地址就可以。或者重启天猫精灵后可以发现了。您看下是否也是这样的情况。

ljy770, 2020-07-02:
> > > 这是生活物联网平台的截图，必须选自定义配网，一键配网才能成功，默认的选官方标准配网，设备一直在切换信道，获取不到配网信息：图 1、 图 2
> > 
> > 
> > 看不到图片，我在 ”人机交互“->”配网引导“ 页面没有看到官方标准配网，只有 默认配网方式：备选配网方式：零配方式（需设备支持）
> 
> 在“[https://living.aliyun.com/”](https://living.aliyun.com/%E2%80%9D) 新建产品之后，第一次进入“人机交互”->”配网引导“ 页面，是默认的“ 标准配网”，这个页面看不到 “默认配网方式：备选配网方式：零配方式（需设备支持）” 的选项，在右上角点击自定义配网才能看到选项。设定之后再次进入”配网引导”页面 还是上次设定的自定义配网方式，点击右上角的提示：”自定义配网体验不好？使用品类标准配网“，可以切回 “品类标准配网”，这里的“标准配网”用一键配网也不成功~

> > > 这是生活物联网平台的截图，必须选自定义配网，一键配网才能成功，默认的选官方标准配网，设备一直在切换信道，获取不到配网信息：图 1、 图 2
> > 
> > 
> > 看不到图片，我在 ”人机交互“->”配网引导“ 页面没有看到官方标准配网，只有 默认配网方式：备选配网方式：零配方式（需设备支持）
> 
> 在“[https://living.aliyun.com/”](https://living.aliyun.com/%E2%80%9D) 新建产品之后，第一次进入“人机交互”->”配网引导“ 页面，是默认的“ 标准配网”，这个页面看不到 “默认配网方式：备选配网方式：零配方式（需设备支持）” 的选项，在右上角点击自定义配网才能看到选项。设定之后再次进入”配网引导”页面 还是上次设定的自定义配网方式，点击右上角的提示：”自定义配网体验不好？使用品类标准配网“，可以切回 “品类标准配网”，这里的“标准配网”用一键配网也不成功~

>> 新建了一个 “空调” 产品， 还是没有看到“标准配网”， 或者你选择的产品是什么？才会出现这个配置

yl673091830, 2020-07-02:
>

我试着新建了“空调”产品，和你的描述一致。新建“电工照明”->"灯"，配网引导设置界面会有不同的选项

ljy770, 2020-07-02:
看到了，显示的标准配网方案：设备热点配网

热点配网这样配置, 在 conn_mgr task 之前添加：
```
    conn_mgr_set_sc_mode(CONN_SOFTAP_MODE);
    xTaskCreate((void (*)(void *))start_conn_mgr, "conn_mgr", 3072, NULL, 5, NULL);
```
但是这个新的标准配网方案，设备端可以拿到 ssid，但是流程上还有点问题，不能完全支持

yl673091830, 2020-07-05:
> 看到了，显示的标准配网方案：设备热点配网
> 
> 热点配网这样配置, 在 conn_mgr task 之前添加：
> 
> ```
>     conn_mgr_set_sc_mode(CONN_SOFTAP_MODE);
>     xTaskCreate((void (*)(void *))start_conn_mgr, "conn_mgr", 3072, NULL, 5, NULL);
> ```
> 
> 但是这个新的标准配网方案，设备端可以拿到 ssid，但是流程上还有点问题，不能完全支持

好的，谢谢

yl673091830, 2020-07-05:
> 请重启天猫精灵，重试下能否发现。
> 我们在调试中也发现偶尔有这个问题，一直不能发现某个 mac 地址的设备，改成其他mac 地址就可以。或者重启天猫精灵后可以发现了。您看下是否也是这样的情况。

是这个问题，谢谢