espressif/esp-aliyun #194: 配网时,域名解析失败

a273097203, 2020-07-11:
----------------------------- 以下请删除 -----------------------------

提交问题时，请先仔细阅读该段提示信息！！

为了更高效的解决 issue 中提交的问题，请按照以下 issue 模板进行提交，请注意以下信息：
1. 提交时请删除这段提示文字；
2. 提交时删除以下由 `//` 开头的文字，并对此做详细补充，如没有特殊信息，可写 `无`。

感谢配合。

----------------------------- 以上请删除 -----------------------------

## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：fed581b
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：8e74a471e

## 2. 问题描述

目前遇到在某一网络下,使用esp-aliyun 进行配网时出现域名解析失败的问题, 
现为了更好的描述问题,以及进行的对照测试,我做了以下定义: (使用esp demo 板 ,及测试均使用demo 工程)
     域名解析失败的网络, => [网络A]
     可以正常配网的网络, => [网络B]
     还使用了手机热点,     => [热点A]
     还使用了对照路由器  => [路由A] [路由B]
     使用了对照SDK/IDF   => [esp-idf v3.2] [esp-idf v3.3] [esp-aliyun v2.0] [esp-aliyun master] [alios-things v3.01] 

问题发现:
    使用 [esp-idf v3.2] [esp-aliyun v2.0] [路由A] [网络A] 进行配网时 出现域名解析失败问题.

对照测试:
  失败对照,失败问题相同,并且100% 出现
    1. [esp-idf v3.2] [esp-aliyun v2.0] [路由B] [网络A] -> 域名解析失败.
    2. [esp-idf v3.3] [esp-aliyun v2.0] [路由A] [网络A] -> 域名解析失败.
    3. [esp-idf v3.2] [esp-aliyun master] [路由A] [网络A] -> 域名解析失败.
    4. [esp-idf v3.3] [esp-aliyun master] [路由A] [网络A] -> 域名解析失败.

  成功对照
    5. [esp-idf v3.2] [esp-aliyun v2.0] [路由A] [网络B] -> 配网成功.
    6. [esp-idf v3.3] [esp-aliyun v2.0] [路由A] [网络B] -> 配网成功.
    7. [esp-idf v3.2] [esp-aliyun master] [路由A] [网络B] -> 配网成功.
    8. [esp-idf v3.3] [esp-aliyun master] [路由A] [网络B] -> 配网成功.

    9. [esp-idf v3.2] [esp-aliyun v2.0] [热点A] -> 配网成功.
    10. [esp-idf v3.3] [esp-aliyun v2.0] [热点A]-> 配网成功.
    11. [esp-idf v3.2] [esp-aliyun master] [热点A] -> 配网成功.
    12. [esp-idf v3.3] [esp-aliyun master] [热点A] -> 配网成功.

    13. [alios-things v3.01] [路由A] [网络A] -> 配网成功.

然后直接在 [网络A] 下 ping 域名 成功

### 2.1 复现步骤

1. 下载 [esp-idf v3.2]  [esp-aliyun v2.0] 
2. 使用examples\solutions\smart_light demo 工程 修改四元组, 
3. 烧录到esp32demo板
4. 在 [网络A] 下进行配网
5. 出现域名解析失败

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs

[20:40:04.481] [1;31m[crt] awss_config_press(180): enable awss
[20:40:04.502] [0m[1;33m[inf] zconfig_init(260): zconfig_init
[... 99 lines cut ...]
[20:40:30.947] [0m[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[20:40:31.039] [0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[20:40:31.135] [0m[1;33m[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
[20:40:31.283] [0m[1;33m[inf] awss_notify_dev_info(209): send notify success
[20:40:31.283] [0m[1;31m[err] _mqtt_connect(276): mqtt connect failed
[20:40:31.650] [0m[1;31m[err] _iotx_linkkit_master_connect(959): DM Start Failed
[20:40:31.650] [0m[1;32;40mlinkkit_main.494: IOT_Linkkit_Connect failed! retry after 5000 ms
[20:40:31.650] [0m
[20:40:31.650] 
[20:40:33.472] [1;33m[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
[20:40:33.493] [0m[1;33m[inf] awss_notify_dev_info(209): send notify success
[20:40:33.493] [0m
[20:40:35.780] [1;33m[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
[20:40:35.802] [0m[1;33m[inf] awss_notify_dev_info(209): send notify success

[配网失败log---1-2020-7-6.txt](https://github.com/espressif/esp-aliyun/files/4906228/log---1-2020-7-6.txt)
[配网失败---3-2020-7-6.txt](https://github.com/espressif/esp-aliyun/files/4906229/---3-2020-7-6.txt)
[配网失败log-启明---1--2020-7-10.txt](https://github.com/espressif/esp-aliyun/files/4906230/log-.---1--2020-7-10.txt)

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

配网手机 -> 华为 P10 plus
路由 -- > 华为 A1

ljy770, 2020-07-12:
请问下网络 A  和 网络 分别是什么样的网络？两者的区别？

a273097203, 2020-07-13:
> 请问下网络 A 和 网络 分别是什么样的网络？两者的区别？

你好, 

目前发现的 [网络A] 为租房集体网络,有终端交换机分接到各个租房 ,  对照的 [网络B] 为直接光纤到路由出来的网络, 

其他并未发现别的区别.

ljy770, 2020-07-13:
@a273097203 请抓取一下域名解析时候的空气包，看一下 DNS 是否回复。

a273097203, 2020-07-13:
> @a273097203 请抓取一下域名解析时候的空气包，对比成功和不成功时候的 DNS 包。

附件为配网失败与配网成功的log 在同一时间段,由使用[网络A] 进行配网的失败log ,与使用[热点]进行配网的成功log
[配网失败log---1-2020-7-6.txt](https://github.com/espressif/esp-aliyun/files/4911305/log---1-2020-7-6.txt)
[配网成功---2-2020-7-6.txt](https://github.com/espressif/esp-aliyun/files/4911306/---2-2020-7-6.txt)
[配网失败---3-2020-7-6.txt](https://github.com/espressif/esp-aliyun/files/4911307/---3-2020-7-6.txt)

a273097203, 2020-07-13:
你好,

附件为配网失败与配网成功的log 在同一时间段,由使用[网络A] 进行配网的失败log ,与使用[热点]进行配网的成功log

东莞市智纳家具有限公司
Dongguan Zhi Na Furniture Co., Ltd.
林瑜杰   研发部
Mobile:+86 15820843060
E-Mail:RDD01@dgzhina.com 
地址:东莞市厚街镇南环路新后街产业创新园H栋四楼H408A
H408A, four H, new industrial park, New South Street, houhuan Town, Houjie Town, Dongguan
 
发件人： ljy770
发送时间： 2020-07-13 16:30
收件人： espressif/esp-aliyun
抄送： a273097203; Mention
主题： Re: [espressif/esp-aliyun] 配网时,域名解析失败 (#194)
@a273097203 请抓取一下域名解析时候的空气包，看一下 DNS 是否回复。

ljy770, 2020-07-13:
@a273097203 不是设备 log. 是需要 [Omnipeek 抓包软件]和[抓包器]设置[设备和路由器mac地址为 filter ]进行抓包，为了看到包的内容，也要设置路由器加密方式为 无。

a273097203, 2020-07-13:
> @a273097203 不是设备 log. 是需要 [Omnipeek 抓包软件]和[抓包器]设置[设备和路由器mac地址为 filter ]进行抓包，为了看到包的内容，也要设置路由器加密方式为 无。

你好,非常抱歉,没有抓包设备可以进行相关的抓包,你看下是否还有别的方式或者提供一下详细的方法,谢谢

ljy770, 2020-07-13:
@a273097203 那只能打开LWIP里DNS的log, 在components/lwip/lwip/src/include/lwip/opt.h里将DNS_DEBUG置1，并且将components/lwip/port/esp32/include/lwipopts.h里LWIP_DEBUG也置1. 对比成功和不成功时候的设备 log.

yuanjianmin, 2020-07-13:
@a273097203 我看抓包文件配网成功---2-2020-7-6.txt里，连接的WIFI是WIFI_T1_TEST，可以解析域名连上云，两个失败的配网失败log---1-2020-7-6.txt和配网失败---3-2020-7-6.txt，连接的WIFI是LYJ_24G，请确认下。