espressif/esp-aliyun #195: 远程升级固件设备终端在升级固件大约4秒后wifi断开连接

ZorkHuang, 2020-07-12:
1. 开发环境
模组或开发板：
esp-aliyun 版本 ：v3.3(bb376c5)
ESP-IDF 版本：v4.01
2. 问题描述
在阿里云创建一个固件，然后验证固件的时候，云端下发了固件升级指令，终端收到升级智能后，开始执行升级固件操作，大约4秒钟后，终端的wifi断开了，导致升级失败；并且不能重连阿里云，必须重启设备；

2.1 复现步骤
在云端发起远程固件升级命令，设备端收到命令后开始执行升级，大约4秒钟后wifi断开连接，升级失败，日志如下：

[10:50:50.421]收←◆[1;33m[inf] _dm_fota_send_new_config_to_user(51): Send To User: {"version":"1.0.4"}
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 42
[0m[1;33m[inf] _iotx_linkkit_event_callback(245): Receive Message: {"version":"1.0.4"}
[0m[1;32;40muser_fota_event_handler.387: New Firmware Version: 1.0.4[0m
[0;32mI (66113) ota: Starting OTA...[0m
[0;32mI (66113) ota: W
[10:50:50.589]收←◆riting to partition subtype 17 at offset 0x1a0000[0m

[10:50:54.343]收←◆[0;32mI (70003) ota: esp_ota_begin succeeded[0m
[0;32mI (70003) ota: Please Wait. This may take time[0m

[10:50:55.432]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 920
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 102
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 275
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 748
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 652
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 371
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 6
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1017
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 383
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 640
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 760
[0m
[10:50:55.973]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 263
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 114
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 909
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 491
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 532
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 868
[0m
[10:50:56.360]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 155
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 222
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 801
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 599
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 424
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 976
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 47
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 1023
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 330
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 693
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 707
[... 140 lines cut ...]
[0m[1;31m[err] IOT_OTA_FetchYield(698): Fetch firmware failed
[0m[1;31m[err] wrapper_mqtt_publish(3099): mqtt client state is error,state = 3
[0m[1;31m[err] otamqtt_Publish(85): publish failed
[0m[1;31m[err] IOT_OTA_ReportProgress(550): Report progress failed
[0m[0;32mI (83633) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (83633) esp_image: segment 0: paddr=0x001a0020 vaddr=0x3f400020 size=0x34b34 (215860) map[0m
[0;31mE (83693) esp_image: invalid segment length 0xffffffff[0m
[0;31mE (83693) ota: esp_ota_end failed! err=0x1503. Image is invalid[0m
[1;31m[err] iotx_mc_keepalive(1855): network is disconnected!
[0m[1;33m[inf] iotx_cloud_conn_mqtt_event_handle(80): disconnected,fd = 0
[0m[1;33m[inf] _dm_client_event_cloud_disconnect_handle(174): IOTX_CM_EVENT_CLOUD_DISCONNECT
[0m[1;31m[err] _mqtt_cycle(1597): error occur rc=-27
[0m[1;33m[inf] _iotx_linkkit_event_callback(243): Receive Message Type: 1
[0m[1;32;40muser_disconnected_event_handler.129: Cloud Disconnected[0m
[1;33m[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[0m[1;33m[inf] wrapper_mqtt_release(2782): mqtt release!
[0m[1;31m[err] IOT_MQTT_Unsubscribe(721): params err
[0m[1;31m[err] IOT_MQTT_Unsubscribe(721): params err
[0m[1;31m[err] IOT_MQTT_Unsubscribe(721): params err
[0m[1;31m[err] IOT_MQTT_Unsubscribe(721): params err
[0m[1;31m[err] IOT_MQTT_Unsubscribe(721): params err
[0m[1;33m[inf] CoAPServer_deinit(172): CoAP Server deinit
[0m[1;33m[inf] CoAPServer_deinit(178): [1;33mWait Semaphore, will exit task[inf] CoAPServer_yield(89): 
[0mExit the CoAP daemon task, Post semphore
[0m[0;33mW (84013) link: [83337 ms] try linkkit[0m

ljy770, 2020-07-12:
请问 路由器型号？复现问题的概率？更换路由器是否还有问题？
有些路由器存在一段时间不广播 beacon 的情况，会导致掉线。

ZorkHuang, 2020-07-13:
有一些是TPLINK路由器，有一些是TPLINK的ap，基本上50%以上的设备都有这个问题；目前还未更换路由器，这个问题是由于远程升级固件，下载固件的时候才会出现，平时运行起不会出现这个问题；

ESP32启用了网口的，晚点我们试一试用网口看看会不会出现这个现象?

ljy770, 2020-07-14:
@ZorkHuang 确认一下，IDF 使用的是 加 tag 的版本？
对于出问题的路由器，可以做一下对比实验确认是否是 IDF 的问题：
1）对现在使用的 v4.0.1 版本，运行 examples/wifi/iperf 示例测试 TCP RX (设备做 tcp server)时候是否会断线？
2）使用早期的版本 v3.3 是否会断线？ https://github.com/espressif/esp-idf/releases/tag/v3.3