espressif/esp-aliyun #196: 更改cvs分区文件后天猫精灵配网无法绑定设备

1404360768, 2020-07-21:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：f2fb493
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：2fd6ea13
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

由于smart_light例子使用flash是2M的，我使用的模块是4M的，将分区域文件改为如下：
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
ota_0,    0,    ota_0,   0x10000, 0x100000
fctry,    data, nvs,     0x110000,0x100000
ota_1,    0,    ota_1,   0x210000,0x100000

然后在menuconfig串口设置里面将flash大小选择为4M后，天猫精灵找队友只能搜索设备，无法绑定。三元组以正确烧写。

### 2.1 复现步骤
见2

### 2.2 复现问题的代码

// 无

[log.txt](https://github.com/espressif/esp-aliyun/files/4950899/log.txt)

## 3. 调试 Logs

[inf] zconfig_callback_channel_locked(118): channel lock @ 13
I (21248) app main: IOTX_AWSS_LOCK_CHAN
[inf] zconfig_callback_over(136): zconfig done. ssid:wifi, mac:2076934c0c86
I (21267) app main: IOTX_AWSS_GOT_SSID_PASSWD
I (21275) awss: Close monitor mode
I (21377) app main: IOTX_AWSS_CONNECT_ROUTER
I (21380) awss: ssid: wifi, password: 123456789, channel: 13
I (21384) awss: Connecting AP
I (23793) wifi: state: 0 -> 2 (b0)
I (23799) wifi: state: 2 -> 3 (0)
I (23799) wifi: state: 2 -> 3 (0)I (23803) wifi: state: 3 -> 5 (10)
I (23805) wifi: pm start, type: 1
I (24302) awss: Connecting AP
I (24802) tcpip_adapter: sta ip: 192.168.123.236, mask: 255.255.255.0, gw: 192.168.123.1
I (24814) awss: AP connected
I (24816) app main: IOTX_AWSS_GOT_IP
I (24823) udp: success to establish udp, fd=54
I (24836) app main: IOTX_AWSS_SUC_NOTIFY
[inf] awss_notify_dev_info(205): topic:/sys/awss/event/connectap/notify
[inf] awss_notify_dev_info(209): send notify success
I (29211) conn_mgr: SNTP get time success

[inf] dm_client_open(37): CM Fd: 0
I (29226) app main: IOTX_CONN_CLOUD
[wrn] IOT_MQTT_Construct(468): Using default hostname: 'a1TJAgfPhyD.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[wrn] IOT_MQTT_Construct(475): Using default port: [443]
[wrn] IOT_MQTT_Construct(482): Using default client_id: a1TJAgfPhyD.LUqgeNBvCBUX8Slatqmn|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[wrn] IOT_MQTT_Construct(489): Using default username: LUqgeNBvCBUX8Slatqmn&a1TJAgfPhyD
[wrn] IOT_MQTT_Construct(497): Using default password: ******
[inf] iotx_mc_init(230): MQTT init success!
E (29486) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (29488) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (29495) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
E (30666) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (30668) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (30675) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
E (31857) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (31859) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (31866) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
[inf] _mqtt_connect(778): connect params: MQTTVersion=4, clientID=a1TJAgfPhyD.LUqgeNBvCBUX8Slatqmn|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|, keepAliveInterval=30, username=LUqgeNBvCBUX8Slatqmn&a1TJAgfPhyD
E (32906) iot_import_tls: HAL_SSL_Write, handle == NULL
[err] MQTTConnect(460): send connect packet failed
[err] _mqtt_connect(786): send connect packet failed, rc = -14
[err] IOT_MQTT_Construct(579): wrapper_mqtt_connect failed
[inf] iotx_mc_disconnect(2637): mqtt disconnect!
[inf] wrapper_mqtt_release(2782): mqtt release!
[wrn] IOT_MQTT_Construct(468): Using default hostname: 'a1TJAgfPhyD.iot-as-mqtt.cn-shanghai.aliyuncs.com'
[wrn] IOT_MQTT_Construct(475): Using default port: [443]
[wrn] IOT_MQTT_Construct(482): Using default client_id: a1TJAgfPhyD.LUqgeNBvCBUX8Slatqmn|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
[wrn] IOT_MQTT_Construct(489): Using default username: LUqgeNBvCBUX8Slatqmn&a1TJAgfPhyD
[wrn] IOT_MQTT_Construct(497): Using default password: ******
[inf] iotx_mc_init(230): MQTT init success!
E (33946) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (33948) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (33955) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
E (35139) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (35141) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (35148) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed
E (36346) esp-tls-wolfssl: wolfSSL_connect returned -0x1
E (36348) esp-tls-wolfssl: Failed to verify peer certificate , returned 28!
E (36356) esp-tls: Failed to open new connection
[err] wrapper_mqtt_connect(2702): TCP or TLS Connection failed

[log.txt](https://github.com/espressif/esp-aliyun/files/4950910/log.txt)

ljy770, 2020-07-30:
请问问题是否有解决？
三元组烧写地址是否改为您指定的 0x110000 位置？
更改三元组是否能连接成功？

1404360768, 2020-07-30:
> 请问问题是否有解决？三元组烧写地址是否替换您指定的0x110000位置？
> 更改三元组是否能连接成功？
已经解决，将fctry位置换到上面去了，解决后的分区如下：
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
fctry,    data, nvs,     0x10000,0x4000
ota_0,    0,    ota_0,   0x14000, 0x1F6000
ota_1,    0,    ota_1,   0x20A000,0x1F6000