espressif/esp-aliyun #198: gcc: 错误：unrecognized command line option ‘-mlongcalls’

colorgit, 2020-09-02:
大神，这个怎么破？

colorgit, 2020-09-02:
gcc: 错误：unrecognized command line option ‘-mlongcalls’
make[2]: *** [/home/esp-aliyun/make/component_wrapper.mk:286：src/flash_encrypt.o] 错误 1
make[1]: *** [/home/esp-aliyun/make/project.mk:517：component-esp8266-build] 错误 2
make[1]: *** [/home/esp-aliyun/make/project.mk:517：component-util-build] 错误 2
make[1]: *** [/home/esp-aliyun/make/project.mk:517：component-spi_flash-build] 错误 2

ljy770, 2020-09-03:
参考 [Get Started](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html) ， [tool chain](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html) 是否正确配置

colorgit, 2020-09-03:
谢谢大神关注回答。
配置了，ubuntu上，win7上用mingw32都试过了，就是编译通不过，用esp-aliyun的SDK。