espressif/esp-aliyun #200: 取消选择 MQTT DIRECT 后 MQTT连接报错

kust336, 2020-09-14:
## 1. 开发环境

- 模组或开发板：esp32 devkitv1
    esp-aliyun 版本（git rev-parse --short HEAD commit id）：v2.0
    esp-idf 版本（git rev-parse --short HEAD commit id）： v3.3

## 2. 问题描述
烧录后能正常配网; 配网后,报错:

[14:06:30.482]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_
[14:06:30.511]收←◆recv is 47
[0m[1;33m[inf] httpclient_common(570): close http channel
[0m[1;33m[inf] httpclient_close(502): client disconnected
[0m[1;31m[err] IOT_MQTT_Construct(435): ret = _iotx_preauth() = -1, abort
[0m
[14:06:32.619]收←◆
[14:06:32.655]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 42
[0m[1;33m[inf] httpclient_common(570): close http channel
[0m[1;33m[inf] httpclient_close(502): client disconnected
[0m[1;31m[err] IOT_MQTT_Construct(435): ret = _iotx_preauth() = -1, abort
[0m
[14:06:38.234]收←◆
[14:06:38.268]收←◆[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv(214): ret of _http_recv is 32
[0m[1;33m[inf] _http_recv
[14:06:38.297]收←◆(214): ret of _http_recv is 47
[0m[1;33m[inf] httpclient_common(570): close http channel
[0m[1;33m[inf] httpclient_close(502): client disconnected
[0m[1;31m[err] IOT_MQTT_Construct(435): ret = _iotx_preauth() = -1, abort
[0m
[14:06:38.769]收←◆[1;31m[err] _mqtt_connect(276): mqtt connect failed
[0m[0;31mE (86166) app main: IOTX_CONN_CLOUD_FAIL[0m
[1;31m[err] _iotx_linkkit_master_connect(962): DM Start Failed
[0m[1;32;40mlinkkit_thread.446: IOT_Linkkit_Connect Failed
[0m
[1;31m[err] IOT_MQTT_Destroy(616): handler is null
[0m[1;33m[inf] dm_client_open(37): CM Fd: 0
[0m[0;32mI (86186) app main: IOTX_CONN_CLOUD[0m

产品需要在国外使用,请帮忙处理下,谢谢

yuanjianmin, 2020-09-14:
请问是否按照https://github.com/espressif/esp-aliyun/tree/master/examples/solutions/smart_light#14%E5%9B%BD%E9%99%85%E7%AB%99%E8%AE%BE%E5%A4%87%E5%BC%80%E5%8F%91操作配置.

kust336, 2020-09-14:
> 请问是否按照https://github.com/espressif/esp-aliyun/tree/master/examples/solutions/smart_light#14%E5%9B%BD%E9%99%85%E7%AB%99%E8%AE%BE%E5%A4%87%E5%BC%80%E5%8F%91 操作配置.

是的APP注册时选用的是美国.  其它操作都是按这个上面做的.

yuanjianmin, 2020-09-16:
阿里全球激活中心更新了(https://help.aliyun.com/document_detail/167594.html?spm=5176.11065259.1996646101.searchclickresult.195e2d8f6mwgD9),所以该项目只能激活原先已经创建好的产品,无法激活新创建的国外产品.后期我们会推出支持该功能的sdk.感谢关注.

kust336, 2020-10-10:
你好！  请部问下激活功能有更新吗？官方是说要用git clone git@code.aliyun.com:living_platform/ali-smartliving-device-alios-things.git -b rel_1.6.2 才支持全球统一激活。

yuanjianmin, 2020-10-12:
> 你好！ 请部问下激活功能有更新吗？官方是说要用git clone [git@code.aliyun.com](mailto:git@code.aliyun.com):living_platform/ali-smartliving-device-alios-things.git -b rel_1.6.2 才支持全球统一激活。

激活功能是smartliving-1.6.0及以后的版本会支持,当前esp-aliyun项目无法更新sdk至1.6.0. 我们基于smartliving(生活物联网平台)开发了一套适配该平台的工程,框架结构与esp-aliyun类似,如需使用全球激活功能,请切换至项目(https://github.com/espressif/esp-ali-smartliving),分支master进行开发.感谢关注.

yuanjianmin, 2020-10-12:
> 阿里全球激活中心更新了(https://help.aliyun.com/document_detail/167594.html?spm=5176.11065259.1996646101.searchclickresult.195e2d8f6mwgD9) ,所以该项目只能激活原先已经创建好的产品,无法激活新创建的国外产品.后期我们会推出支持该功能的sdk.感谢关注.

支持全球激活的smart-living-1.6.0版本可在项目(https://github.com/espressif/esp-ali-smartliving) 获取.