espressif/esp-aliyun #202: esp8266编译报错:error: 'multiple definition of `os_htole16''

frPhoenix, 2020-12-06:
### 一、环境
- esp-aliyun 版本（git rev-parse --short HEAD commit id）：8b8eeada3

- ESP8266_RTOS_SDK 版本（git rev-parse --short HEAD commit id）： a768f1ce9

### 二、问题
`App "smart_light" version: v2.0-85-g8b8eead
LD build/smart_light.elf
/Users/phoenix/esp2/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /Users/phoenix/esp2/ESP8266_RTOS_SDK/components/esp8266/lib/libsmartconfig.a(sc_ieee80211.o): in function `os_htole16':
(.text.os_htole16+0x0): multiple definition of `os_htole16'; /Users/phoenix/esp2/ESP8266_RTOS_SDK/AliyunProject/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(os_misc.o):/Users/phoenix/esp2/ESP8266_RTOS_SDK/AliyunProject/esp-aliyun/iotkit-embedded/dev_bind/impl/os/os_misc.c:26: first defined here`

yuanjianmin, 2020-12-25:
请使用release/v3.3或者tag v3.3进行编译.

ljm996054544, 2022-06-01:
v3.3分支里的例程编译不通过是xtensa-lx106-elf的问题，重新安装可以了

ljm996054544, 2022-06-01:
ESP8266_RTOS_SDK 版本 v3.3
esp-aliyun 版本 master
` d:/app/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: E:/Desktop/git/ESP8266_RTOS_SDK/components/esp8266/lib\libsmartconfig.a(sc_ieee80211.o): in function `os_htole16':
(.text.os_htole16+0x0): multiple definition of `os_htole16'; E:/Desktop/git/ESP8266_RTOS_SDK_v3.3/esp-aliyun_master/examples/solutions/smart_light/build/esp-aliyun_master\libesp-aliyun_master.a(os_misc.o):E:/Desktop/git/ESP8266_RTOS_SDK_v3.3/esp-aliyun_master/iotkit-embedded/dev_bind/impl/os/os_misc.c:26: first defined here
d:/app/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: E:/Desktop/git/ESP8266_RTOS_SDK/components/esp8266/lib\libsmartconfig.a(sc_ieee80211.o): in function `os_le16toh':
(.text.os_le16toh+0x0): multiple definition of `os_le16toh'; E:/Desktop/git/ESP8266_RTOS_SDK_v3.3/esp-aliyun_master/examples/solutions/smart_light/build/esp-aliyun_master\libesp-aliyun_master.a(os_misc.o):E:/Desktop/git/ESP8266_RTOS_SDK_v3.3/esp-aliyun_master/iotkit-embedded/dev_bind/impl/os/os_misc.c:42: first defined here
d:/app/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: warning: alignment 1 of symbol `zero_mac' in E:/Desktop/git/ESP8266_RTOS_SDK/components/esp8266/lib\libsmartconfig.a(sc_ieee80211.o) is smaller than 4 in E:/Desktop/git/ESP8266_RTOS_SDK_v3.3/esp-aliyun_master/examples/solutions/smart_light/build/esp-aliyun_master\libesp-aliyun_master.a(zconfig_protocol.o)
collect2.exe: error: ld returned 1 exit status `