espressif/esp-aliyun #203: 使用APP进行smartconfig配网时，一直提示 esp-tls-wolfssl 错误，导致配网失败

whwxxstsyd, 2020-12-26:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

使用APP进行smartconfig配网时，最后一个步骤，一直报以下错误：

[0m[0;31mE (44207) esp-tls-wolfssl: wolfSSL_connect returned -0x1[0m
[0;31mE (44209) esp-tls-wolfssl: Failed to verify peer certificate 28![0m
[0;31mE (44216) esp-tls: Failed to open new connection[0m

导致配网失败，但是在出现该错误后，重启一下设备，又可以马上配网成功

### 2.1 复现步骤

### 2.2 复现问题的代码

## 3. 调试 Logs

[0;32mI (5788) factory_rst: Quick reboot timeout, clear reboot times[0m
[0;32mI (13163) auto_socket: ===================== control_func, dpID:6b, sta_data:0 =====================[0m
[0;32mI (13168) linkkit_example_solo: SocketPowerState Message ID: -1[0m
[0;32mI (13767) auto_socket: ===================== control_func, dpID:6b, sta_data:1 =====================[0m
[0;32mI (13772) linkkit_example_solo: SocketPowerState Message ID: -1[0m
[0;32mI (14254) auto_socket: ===================== control_func, dpID:6b, sta_data:0 =====================[0m
[0;32mI (14258) linkkit_example_solo: SocketPowerState Message ID: -1[0m
[0;32mI (15896) auto_socket: ================= heart_beat =================[0m
[0;32mI (15915) auto_socket: ================= heart_beat_func =================[0m
[0;32mI (20125) auto_socket: ================= wifi_mode_reset_func =================[0m
[0;31mE (20129) wrapper_kv: Set q_rt blob value[0m
[0;32mI (20182) auto_socket: ================= wifi_reset_func =================[0m
[0;32mI (20200) auto_socket: ================= wifi_mode_reset_func =================[0m
[0;31mE (20204) wrapper_kv: Set q_rt blob value[0m
[0;32mI (22195) auto_socket: ================= system_reboot =================[0m
G瑈PmH@HlFf鶊&韜#P@娼y堳i?J昑D諈?9?`HP@誫!KpFT餆%J訲??濉孴E詆{^pPY*謪?7Z1P@昰!	葽仵keZT?!孴ETg{餕Z*??7Z?P@昰!壻hFTo?!@Qe!圱ETg?餕桾E贃a?l?跉抝%丅ⅱJ浇验*5?)姅$?&iSI譜ΙS??2嵸*叺"ド彦亽r憗氀厺晛浇驯絽憰蒻诹j5R诖跉抝%丅ⅱJ浇验?降隶睍仮サ晛姾覀室毢诖j5R诖跉抝%丅ⅹJ浇验??iWV$$HH悙: 40MHz[0m
[0;32mI (53) boot: SPI Mode       : DOUT[0m
[0;32mI (61) boot: SPI Flash Size : 4MB[0m
[0;32mI (69) boot: Partition Table:[0m
[0;32mI (77) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (91) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (106) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (121) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (136) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (151) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
[0;32mI (167) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (182) boot: End of partition table[0m
[0;32mI (190) boot: No factory image, trying OTA 0[0m
[0;32mI (200) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x9dd04 (646404) map[0m
[0;32mI (501) esp_image: segment 1: paddr=0x000add1c vaddr=0x402add14 size=0x1e7b4 (124852) map[0m
[0;32mI (556) esp_image: segment 2: paddr=0x000cc4d8 vaddr=0x3ffe8000 size=0x00e84 (  3716) load[0m
[0;32mI (558) esp_image: segment 3: paddr=0x000cd364 vaddr=0x40100000 size=0x00a40 (  2624) load[0m
[0;32mI (571) esp_image: segment 4: paddr=0x000cddac vaddr=0x40100a40 size=0x058c0 ( 22720) load[0m
[0;32mI (598) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (625) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
[0;32mI (630) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
phy_version: 1156.0, 635e6fc, Feb 20 2020, 16:20:08, RTOS new
[0;32mI (702) phy_init: phy ver: 1156_0[0m
[0;32mI (705) reset_reason: RTC reset 1 wakeup 0 store 3, reason is 3[0m
....................................................

Product Key=a1yJAmJczNk.

Product Secret=SjPckzfmtWXNLy5x.

Device Name=auto_socket_test2.

[... 1093 lines cut ...]
[0m[1;33m[inf](18994) awss_notify_dev_info(236): coap send notify success

[0m[1;33m[inf](19104) awss_notify_dev_info(232): topic:/sys/device/info/notify
[0m[1;33m[inf](19113) awss_notify_dev_info(236): coap send notify success

[0m[1;33m[inf](19231) awss_update_token(72): update token

[0m[1;33m[inf](19264) awss_report_token_to_cloud(366): report token res:27

[0m[1;33m[inf](19413) awss_report_token_reply(117): awss_report_token_reply

[0m[1;33m[inf](19792) dm_msg_proc_thing_event_notify(488): _thing/event/notify

[0m[1;33m[inf](19933) _iotx_linkkit_event_callback(237): Receive Message Type: 46

[0m[1;33m[inf](19936) _iotx_linkkit_event_callback(239): Receive Message: {"devid":0,"payload":{"identifier":"awss.BindNotify","value":{"Operation":"Bind"}},"msgid":1572693069}

[0m[0;32mI (20587) linkkit_example_solo: Event notify Received, Devid: 0, Request: {"identifier":"awss.BindNotify","value":{"Operation":"Bind"}}[0m
[1;33m[inf](19986) dm_msg_request(212): DM Send Message, URI: /sys/a1yJAmJczNk/auto_socket_test2/_thing/event/notify_reply, Payload: {"id":"8","version":"1.0","data":{"identifier":"awss.BindNotify","value":{"Operation":"Bind"}},"method":"_thing.event.notify"}

[0m[1;33m[inf](20044) dm_client_publish(187): Publish Result: 0

[0m[0;32mI (20659) linkkit_example_solo: Post Property Message ID: 8[0m

## 4. 其他项

yuanjianmin, 2020-12-31:
请问一下您使用的esp-aliyun是哪个版本，具体commit id是多少？

yuanjianmin, 2020-12-31:
您这个问题应该是全球配网功能，该功能可能在esp-aliyun上是有问题的，建议您使用 https://github.com/espressif/esp-ali-smartliving master版本试一下。

whwxxstsyd, 2021-01-06:
目前问题已解决，经过测试，发现是个别线程分配内存空间过大导致，wolfSSL需要消耗一定的内存空间，可以看一下这篇文章

https://blog.csdn.net/espressif/article/details/102678075

closed 2021-01-14 by whwxxstsyd