espressif/esp-aliyun #209: Ubuntu 20: undefined reference to `pthread_create'

losingrose, 2021-06-09:
## 1. 开发环境
Ubuntu 20
- 模组或开发板：ESP8266
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：1259029
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：6c438bce
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7

## 2. 问题描述
make编译失败。最后报错找不到pthread_create

Generating esp8266.project.ld
LD build/smart_light.elf
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o):(.literal.HAL_ThreadCreate+0x4): undefined reference to `pthread_create'
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o):(.literal.HAL_ThreadDelete+0x0): undefined reference to `pthread_exit'
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o):(.literal.HAL_ThreadDelete+0x4): undefined reference to `pthread_cancel'
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o):(.literal.HAL_ThreadDelete+0x8): undefined reference to `pthread_join'
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o): In function `HAL_ThreadCreate':
/home/losingrose/esp/esp-aliyun/wrappers/wrapper_os.c:267: undefined reference to `pthread_create'
/home/losingrose/esp/esp-aliyun/wrappers/wrapper_os.c:267: undefined reference to `pthread_create'
/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/esp-aliyun/libesp-aliyun.a(wrapper_os.o): In function `HAL_ThreadDelete':
/home/losingrose/esp/esp-aliyun/wrappers/wrapper_os.c:267: undefined reference to `pthread_exit'
/home/losingrose/esp/esp-aliyun/wrappers/wrapper_os.c:267: undefined reference to `pthread_cancel'
/home/losingrose/esp/esp-aliyun/wrappers/wrapper_os.c:267: undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make: *** [/home/losingrose/esp/ESP8266_RTOS_SDK/make/project.mk:510：/home/losingrose/esp/esp-aliyun/examples/solutions/smart_light/build/smart_light.elf] 错误 1

ESP-YJM, 2021-06-10:
有没有先执行`make chip=esp8266 defconfig`,  执行这个会打开 CONFIG_ENABLE_PTHREAD。https://github.com/espressif/esp-aliyun/blob/master/examples/solutions/smart_light/sdkconfig_esp8266.defaults#L2

losingrose, 2021-06-10:
执行后OK了，感谢你的回复。