espressif/esp-aliyun #212: WIN10 VSCODE  IDF 4.3 Failed to resolve component 'esp-aliyun'

winvip8, 2021-08-16:
## 1. 开发环境
   WIN10 VSCODE  IDF 4.3

模组esp32 c3

## 2. 问题描述
运行idf.py menuconfig 或idf.py build时，提示组件不能解释问题

### 2.1 复现步骤

### 2.2 复现问题的代码

## 3. 调试 Logs

G:\PRJ\hardware\switch\ali\examples\solo\example_solo>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory g:\prj\hardware\switch\ali\examples\solo\example_solo\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 g:\prj\hardware\switch\ali\examples\solo\example_solo"...
-- Found Git: D:/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: D:/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at D:/esp32-c3/tools/cmake/build.cmake:189 (message):
  **Failed to resolve component 'esp-aliyun'.**
Call Stack (most recent call first):
  D:/esp32-c3/tools/cmake/build.cmake:215 (__build_resolve_and_add_req)
  D:/esp32-c3/tools/cmake/build.cmake:216 (__build_expand_requirements)
  D:/esp32-c3/tools/cmake/build.cmake:435 (__build_expand_requirements)
  D:/esp32-c3/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:11 (project)

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

XuBovey, 2021-09-07:
esp-aliyun 路径的问题。

路径的配置在demo的根目录下的CMakeLists.txt中如下：

`set (EXTRA_COMPONENT_DIRS "../../..")`

最简单的解决办法：
直接把esp-aliyun文件夹拷贝到demo的components文件夹下

winvip8, 2021-11-25:
好的，谢谢